arduino/ArduinoCore-zephyr #1: Wire and Wire1 are Swapped

mjs513, 2024-11-27:
**Describe the bug**
Looks like Wire and Wire1 are swapped, please check this thread for more details:
https://forum.arduino.cc/t/i2c-experiments-with-zephyr-on-giga/1326321

this post in particular: https://forum.arduino.cc/t/i2c-experiments-with-zephyr-on-giga/1326321/6

**Target board + cli verbose compilation output**
Arduino GigaR1

**Full verbose** compilation output, ideally with `arduino-cli` invocation or from IDE 2.3.3+
[Output.txt](https://github.com/user-attachments/files/17936410/Output.txt)

**Output of Serial Monitor**
"Error initializing communication with IMU"

**Output of readelf**
You can find the loaction of the elf file by compiling in Verbose mode and looking near the end of the compilation output (after `Linking everything together..`)
Paste (or attach) the output of `arm-none-eabi-readelf -a $your_sketch_elf_file`

**Optional: attach the elf file**
[LIS3DML_icm20649_bolderflight_libs.ino.zip](https://github.com/user-attachments/files/17936337/LIS3DML_icm20649_bolderflight_libs.ino.zip)

**Optional: attach the sketch**

**Additional context**
Runs under Giga1 Mbed no issues.

mjs513, 2024-11-28:
@KurtE were digging some more into the way Zephyr works (see the referenced thread) and it appears that only one Wire port may be defined even though I don't see defined in the variants for the giga.

mjs513, 2024-11-28:
Downloaded the current updates in the repository and Wire now seems to defined correctly.  However when I use the I2CScanner sketch and change to Wire1 I get an error message:

>Compilation error: 'Wire1' was not declared in this scope; did you mean 'Wire'?

So kind of confirms only one Wire object is defined

mjs513, 2024-11-29:
Just a quick update.  Got it working by adding this to the giga m7 overlay file

```
&i2c1 {
	status = "disabled";
	pinctrl-0 = <&i2c1_scl_pb8 &i2c1_sda_pb9>;
	pinctrl-names = "default";
	clock-frequency = <I2C_BITRATE_FAST>;
};

&i2c2 {
	status = "disabled";
	pinctrl-0 = <&i2c2_scl_ph4 &i2c2_sda_pb11>;
	pinctrl-names = "default";
	clock-frequency = <I2C_BITRATE_FAST>;
};
```
and
```
i2cs = <&i2c2>, <&i2c4>, <&i2c1>;
```
Then running a modified Scanner (PJRC) was able to see wire and wire1
```
Scanning Wire...
Device found at address 0x1C  (LIS3MDL)
Device found at address 0x68  (DS1307,DS3231,MPU6050,MPU9050,MPU9250,ITG3200,ITG3701,LSM9DS0,L3G4200D)
done

Scanning Wire1...
Device found at address 0x60  (MPL3115,MCP4725,MCP4728,TEA5767,Si5351)
Device found at address 0x68  (DS1307,DS3231,MPU6050,MPU9050,MPU9250,ITG3200,ITG3701,LSM9DS0,L3G4200D)
Device found at address 0x76  (MS5607,MS5611,MS5637,BMP280)
done
```
@KurtE mentioned you might want to update your scanner for more than 1 wire object.   Also might be convenient to define WIRE_INTERFACES_COUNT in Wire after the MACRO is called in Wire.h

Heres the sketch I am using for reference:
[GIGA_SCANNER.zip](https://github.com/user-attachments/files/17961882/GIGA_SCANNER.zip)

facchinm, 2024-12-09:
Hi @mjs513 ,
thanks for spotting! Do you mind providing a PR for the dts to preserve attribution 🙂 ? Thx!

closed 2024-12-09 by mjs513

mjs513, 2024-12-09:
Yep we added a couple things the overlay for SPI based on @KurtE stuff:

```
&spi1 {
	status = "okay";
	pinctrl-0 = <&spi1_nss_pa4 &spi1_sck_pb3
		     &spi1_miso_pg9 &spi1_mosi_pd7>;
	pinctrl-names = "default";
};

&spi5 {
	status = "okay";
	pinctrl-0 = <&spi5_nss_pk1 &spi5_sck_ph6
		     &spi5_miso_pj11 &spi5_mosi_pj10>;
	pinctrl-names = "default";
};
```

and

`		spis = <&spi1>, <&spi5>;
`

think the diff is we specified spi5 instead of just saying ok

facchinm, 2024-12-09:
I'll check the SDFat lib, let's move to another issue so we can properly keep track there