arduino/ArduinoCore-zephyr #4: SdFat failing using SPI or SPI1 (after updating overlay).

mjs513, 2024-12-09:
@KurtE and myself tried running a couple of sdfat test cases that work when using Giga Mbed.

What happens they compile no issue and load no issued using zephyr.  Howver, immediately after upload it hangs.  Did try looking as serial1 output but no output.

Using spi:
[SDFat_Giga_zephyr-241209a.zip](https://github.com/user-attachments/files/18062898/SDFat_Giga_zephyr-241209a.zip)
[SDFat_compile.txt](https://github.com/user-attachments/files/18062899/SDFat_compile.txt)

Using SPI1 we used a method to specify our own SPI driver just in case but the same thing occurs:
[sketch_dec6a.ino.zip](https://github.com/user-attachments/files/18062912/sketch_dec6a.ino.zip)

Overlay was updated to specify:
```
&spi5 {
	status = "okay";
	pinctrl-0 = <&spi5_nss_pk1 &spi5_sck_ph6
		     &spi5_miso_pj11 &spi5_mosi_pj10>;
	pinctrl-names = "default";
};
```

Not sure what else you need.

facchinm, 2024-12-09:
**Debugging journal**

How to debug this issue:
* with UART connected -> we see the shell, type `sketch`, get 
      ```[00:00:01.528,000] <err> llext: Failed to copy regions, ret -12                                             
      Failed to load sketch, rc -12```                                                                             
  * Reason: the sketch is too big to fit in LLEXT HEAP (the one that will be used when interacted via shell)
  * Solution: temporarily reduce `CONFIG_HEAP_MEM_POOL_SIZE` to 2048, increase `CONFIG_LLEXT_HEAP_SIZE` to 64
* next error:
      ```[00:00:03.332,000] <err> llext: Undefined symbol with no entry in symbol table isalpha, offset 13200, link 0
      [00:00:03.343,000] <err> llext: Failed to link, ret -61                                                     
      Failed to load sketch, rc -61```                                                  
  * Reason: symbol not yet exported in `llext_exports.c`
  * Solution: add `isaplha` and `isupper` to that file

mjs513, 2024-12-09:
Going to agree with @KurtE - learning a lot now with using zephyr

KurtE, 2024-12-09:
Sorry, I am new to this system.  So when you say:
```
with UART connected -> we see the shell, type sketch, get
[00:00:01.528,000] <err> llext: Failed to copy regions, ret -12                                              Failed to load sketch, rc -12

Reason: the sketch is too big to fit in LLEXT HEAP (the one that will be used when interacted via shell)
Solution: temporarily reduce CONFIG_HEAP_MEM_POOL_SIZE to 2048, increase CONFIG_LLEXT_HEAP_SIZE to 64
```
How do you do this?  
Can you do this within the UART?  
Or do you need to edit some file and change it?  Which one?  autoconf.h?  in the directory like:
/home/kurte/github/ArduinoCore-zephyr/variants/arduino_giga_r1_m7/llext-edk/include/zephyr/include/generated

Note: I see those defines in several files
```

~/github/ArduinoCore-zephyr/loader/prj.conf:
   21  CONFIG_LLEXT=y
   22  CONFIG_LLEXT_LOG_LEVEL_ERR=y
   23: CONFIG_LLEXT_HEAP_SIZE=32
   24  CONFIG_LLEXT_SHELL=y
   25  CONFIG_LLEXT_STORAGE_WRITABLE=y

~/github/ArduinoCore-zephyr/loader/boards/arduino_nano_33_ble_sense.conf:
   36  CONFIG_SHELL_STACK_SIZE=2048
   37  CONFIG_HEAP_MEM_POOL_SIZE=16384
   38: CONFIG_LLEXT_HEAP_SIZE=96
   39  CONFIG_MAIN_STACK_SIZE=16384
   40  

~/github/ArduinoCore-zephyr/loader/boards/arduino_portenta_h7_m7.conf:
   23  CONFIG_SHELL_STACK_SIZE=32768
   24  CONFIG_MAIN_STACK_SIZE=32768
   25: CONFIG_LLEXT_HEAP_SIZE=128
   26  
   27  #CONFIG_ADC=y

~/github/ArduinoCore-zephyr/variants/arduino_giga_r1_m7/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h:
  414  #define CONFIG_LLEXT 1
  415  #define CONFIG_LLEXT_TYPE_ELF_OBJECT 1
  416: #define CONFIG_LLEXT_HEAP_SIZE 32
  417  #define CONFIG_LLEXT_SHELL 1
  418  #define CONFIG_LLEXT_SHELL_MAX_SIZE 8192

~/github/ArduinoCore-zephyr/variants/arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h:
  777  #define CONFIG_LLEXT 1
  778  #define CONFIG_LLEXT_TYPE_ELF_OBJECT 1
  779: #define CONFIG_LLEXT_HEAP_SIZE 96
  780  #define CONFIG_LLEXT_SHELL 1
  781  #define CONFIG_LLEXT_SHELL_MAX_SIZE 8192

~/github/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h:
  567  #define CONFIG_LLEXT 1
  568  #define CONFIG_LLEXT_TYPE_ELF_OBJECT 1
  569: #define CONFIG_LLEXT_HEAP_SIZE 128
  570  #define CONFIG_LLEXT_SHELL 1
  571  #define CONFIG_LLEXT_SHELL_MAX_SIZE 8192

~/github/ArduinoCore-zephyr/variants/ek_ra8d1/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h:
  325  #define CONFIG_LLEXT 1
  326  #define CONFIG_LLEXT_TYPE_ELF_OBJECT 1
  327: #define CONFIG_LLEXT_HEAP_SIZE 32
  328  #define CONFIG_LLEXT_SHELL 1
  329  #define CONFIG_LLEXT_SHELL_MAX_SIZE 8192

~/github/ArduinoCore-zephyr/variants/frdm_mcxn947_mcxn947_cpu0/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h:
  507  #define CONFIG_LLEXT 1
  508  #define CONFIG_LLEXT_TYPE_ELF_OBJECT 1
  509: #define CONFIG_LLEXT_HEAP_SIZE 32
  510  #define CONFIG_LLEXT_SHELL 1
  511  #define CONFIG_LLEXT_SHELL_MAX_SIZE 8192

~/github/zephyr/doc/services/llext/config.rst:
   12  data. The following option controls this allocation.
   13  
   14: :kconfig:option:`CONFIG_LLEXT_HEAP_SIZE`
   15  
   16          Size of the LLEXT heap in kilobytes.

~/github/zephyr/samples/subsys/llext/edk/app/prj.conf:
   12  CONFIG_ZBUS_CHANNEL_NAME=y
   13  
   14: CONFIG_LLEXT_HEAP_SIZE=32
   15  
   16  # Uncomment to disable kext1.

~/github/zephyr/samples/subsys/llext/modules/prj.conf:
   11  CONFIG_LLEXT=y
   12  CONFIG_LLEXT_LOG_LEVEL_DBG=y
   13: CONFIG_LLEXT_HEAP_SIZE=64
   14  CONFIG_LLEXT_TYPE_ELF_RELOCATABLE=y
   15  

~/github/zephyr/samples/subsys/llext/shell_loader/prj.conf:
    9  CONFIG_LLEXT=y
   10  CONFIG_LLEXT_LOG_LEVEL_DBG=y
   11: CONFIG_LLEXT_HEAP_SIZE=8
   12  CONFIG_LLEXT_SHELL=y
   13  

~/github/zephyr/subsys/llext/llext_mem.c:
   26  #endif
   27  
   28: K_HEAP_DEFINE(llext_heap, CONFIG_LLEXT_HEAP_SIZE * 1024);
   29  
   30  /*

~/github/zephyr/tests/subsys/llext/simple/prj.conf:
    3  CONFIG_LOG=y
    4  CONFIG_LLEXT=y
    5: CONFIG_LLEXT_HEAP_SIZE=32
    6  CONFIG_LLEXT_LOG_LEVEL_DBG=y
    7  

~/github/ArduinoCore-zephyr/build/zephyr/.config:
 1317  # CONFIG_LLEXT_TYPE_ELF_RELOCATABLE is not set
 1318  # CONFIG_LLEXT_TYPE_ELF_SHAREDLIB is not set
 1319: CONFIG_LLEXT_HEAP_SIZE=32
 1320  CONFIG_LLEXT_SHELL=y
 1321  CONFIG_LLEXT_SHELL_MAX_SIZE=8192
```
I am assuming if I change one of these files, I probably at least have to do the
./extras/build.sh

Do I then need to then do the burn bootloader and then rebuild and program the GIGA?

Thanks, I know that this is probably a pain to explain, but then others hopefully will learn as well.
Thanks again

mjs513, 2024-12-09:
@facchinm 

Just tried your updates in the PR, still getting 
`[00:00:13.561,000] [1;31m<err> llext: Failed to copy regions, ret -12[0m
Failed to load sketch, rc -12`
even with just trying the QuickStart.ino.  Good news is that is the only error I am seeing now that you explained how to use the debugger.

KurtE, 2024-12-09:
Quick update: I also tried out using your Pull Request code.  I do think some things are working better, I am still having some issues, but learning.

Couple questions and observations:
With current code, the config structure starts off uninitialized (maybe zeros)
```
private:
  const struct device *spi_dev;
  struct spi_config config;

```'
So if you do something like:
```
SPI.begin();
SPI.transfer(10);

```
Nothing is transferred. Whereas I believe most systems are setup to defualt to 4mhz, 8 bit, MSBFIRST.
Which my quick test on the mbed version output/  The Zephyr version did not output anything. 

One hack/fix was in SPI.begin: 
have it do something like:
```
beginTransaction(SPISettings());
endTransaction();
```
Or you can unwind the code there.

Side note: on both windows and linux, just before linking I get a message like:
`Bootloader file specified but missing: /home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/bootloaders/zephyr-arduino_giga_r1_m7.bin
`
Not sure if it is important or not.  The directory bootloaders does not exist. The only  place I notice that directory referenced is in platform.txt

facchinm, 2024-12-10:
Answering in order:
> temporarily reduce CONFIG_HEAP_MEM_POOL_SIZE to 2048, increase CONFIG_LLEXT_HEAP_SIZE to 64

This requires modifying the loader's .conf for the given board, a run of `./extra/build.sh` and a Burn Bootloader to become effective
In general, every modification in `loader` folder requires this procedure 🙂 

> SPI does not start with a sensible default

Good catch, preparing a PR for this too

> Bootloader missing warning

Not an issue, this will be needed to create a production-ready hex file with loader+sketch in a single flashable `.hex` file, something for tomorrow 🙂

mjs513, 2024-12-10:
>temporarily reduce CONFIG_HEAP_MEM_POOL_SIZE to 2048, increase CONFIG_LLEXT_HEAP_SIZE to 64

A bit confused since for the giga_r1 conf  those configure parameters are not in the `arduino_giga_r1_m7.conf` file at all.   Believe @KurtE showed the parameters for the different boards.  Based on his list (shown in previous message) only the nanosense 33 ble has both parameters defined.  While the pjrf.conf file has:

```
CONFIG_HEAP_MEM_POOL_SIZE=32768
CONFIG_MAIN_STACK_SIZE=32768

CONFIG_ARDUINO_API=n

CONFIG_SHELL=y
CONFIG_SHELL_STACK_SIZE=2048
CONFIG_SHELL_CMD_BUFF_SIZE=2048
CONFIG_SHELL_LOG_LEVEL_DBG=n

CONFIG_LLEXT=y
CONFIG_LLEXT_LOG_LEVEL_ERR=y
CONFIG_LLEXT_HEAP_SIZE=32
CONFIG_LLEXT_SHELL=y
CONFIG_LLEXT_STORAGE_WRITABLE=y
CONFIG_LLEXT_EXPORT_DEVICES=y
```

so really not sure what you mean by increase mem pool size and reduce heap size when they are not defined in the conf for the giga r1. 

Am I missing something here.  PS the Potentia only has heap size defined as 128?

mjs513, 2024-12-10:
@facchinm 

went ahead and added
```
CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_LLEXT_HEAP_SIZE=64
```
No more hangs.  Great news but what will the permanent fix be?

Good news is that SPI works but SPI1 works.  Probably have to add:
```
&spi5 {
	status = "okay";
	pinctrl-0 = <&spi5_nss_pk1 &spi5_sck_ph6
		     &spi5_miso_pj11 &spi5_mosi_pj10>;
	pinctrl-names = "default";
};
```

or something like it.

KurtE, 2024-12-10:
I also added the above, and getting some better luck...
@mjs513
```
However, if I switch the cspin to pin 10 (no change to sketch) it still works to dump the drive contents even though the sketch specified the CS PIN as 6. Happens if I change the CSPIN in the sketch to other pins. PIN10 is always act as a CSPIN as well as the pin selected.
```
Quick note: with one of the SDFat examples I tried (SDInfo), I found that it appeared to use the
```
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(16))
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16))
#endif  // HAS_SDIO_CLASS

```
ENABLE_DEDICATED_SPI section... Not sure why.  
But I edited that section with some deliberate garbage and the compiler errored.

With the display stuff not sure what is going on.  I have it setup for SPI1, currently, may try it on SPI 
in a minute.

The real simple TFT sketch, runs on it. 
[simple_tft-241210a.zip](https://github.com/user-attachments/files/18086581/simple_tft-241210a.zip)

However one where I stripped out stuff from my MBED Giga display code, does not load:
[simple_ili9341_GIGA_zephyr-241210a.zip](https://github.com/user-attachments/files/18086598/simple_ili9341_GIGA_zephyr-241210a.zip)
Note: the monitor setup to Serial1, did catch an issue with missing code, which I resolved.
[image]

But once resolved and I try to load the sketch, the monitor window does not show anything
Nor respond, and the Serial object (/dev/ttyACM0) is gone...

mjs513, 2024-12-12:
@KurtE - @facchinm 

Was doing a bunch more reading on spi api and overlays.  Looks like if the CS Pin is not defined in the overlay it won't hardwire it in the driver so a the format for spi1 for example would be:
```C++
&spi5 {
	status = "okay";
	pinctrl-0 = <&spi5_sck_ph6
		     &spi5_miso_pj11 &spi5_mosi_pj10>;
	pinctrl-names = "default";
};
```

Note the nss pin is no longer defined.  I tried the SDInfo example in sdfat with a CSpin defined as 4 and it worked as expected:
```
BLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 4
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 144 ms

Card type: SDXC
sdSpecVer: 5.00
HighSpeedMode: false

Manufacturer ID: 0X3
OEM ID: SD
Product: SC64G
Revision: 8.0
Serial number: 0XCE8B50DD
Manufacturing date: 12/2019
CID HEX: 035344534336344780CE8B50DD013C49

cardSize: 63864.57 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros
CSD HEX: 400E00325B590001DBD37F800A4040DF

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,124702720
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is exFAT
sectorsPerCluster: 256
fatStartSector:    49152
dataStartSector:   65536
clusterCount:      486992
freeClusterCount:  486865

type any character to start
```
now if I repeat the run with swapping the jumper to pin 10 (previously hard coded):
```
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_ACMD41 = 0X17
SD errorData = 0X80

type any character to start
```

Still have to try SPI but assume it will work.  This was also something that @KurtE and myself discussed earlier.

KurtE, 2024-12-12:
I was going to reply, I had better luck without it being defined, As you mentioned, the SPI library does not use the idea of hardware CS support.

KurtE, 2024-12-12:
Question: would you prefer more generic questions on Device tree usage DTS/Overlay/Sketch??? here or new Issue or forum?

Things like: Status of OKAY versus disabled?  What does OKAY mean?  Device object created and initiated?  In particular, if multiple OKAY objects exist who use the same objects in a phandle-array?  More specific refer to the same IO pin.
Like pin 13 could be GPIO, SPI, or PWM or ...

Sorry last time I played with Device trees, was back in the early Beagle Bone Black days... And I did not spend a lot of time on it, as I had migrated to experimenting with ODroid boards...

mjs513, 2024-12-12:
@KurtE - @facchinm 

Quick update to previous post.  To avoid having to recompile the core I switched to a MPU-9250 using the Bolderflight MPU-9250 library.  Tested in same way as sd fat.  

CSPin selected in sketch = pin4:
Ran no issues - good data seen in serial monitor.

Swapped jumper to pin10 from pin4 reran sketch received:
`Error initializing communication with IMU`

So I would say its probably a good fix to just remove nss pin from list for spiX pins.

KurtE, 2024-12-12:
@facchinm @mjs513 
Quick update on the SPI ILI9341 library porting and GIGA dying, including the Monitor window dead...

I have a version of the GraphicTest running... Still some issues in the code, like the DeadBeef test page is not correct..
But that is secondary.

The main issue is/was, there appears to be some form of problem with object constructors.   Not sure if there is enough 
memory or stack space?  Or Timing...  The code was like:
```
ILI9341_GIGA_n::ILI9341_GIGA_n(uint8_t cs, uint8_t dc, uint8_t rst, uint8_t mosi,
                         uint8_t sclk, uint8_t miso) {
  _cs = cs;
  _dc = dc;
  _rst = rst;
  _mosi = mosi;
  _sclk = sclk;
  _miso = miso;
  _width = WIDTH;
  _height = HEIGHT;

  rotation = 0;
  cursor_y = cursor_x = 0;
  textsize_x = textsize_y = 1;
  textcolor = textbgcolor = 0xFFFF;
  wrap = true;
  font = NULL;
  gfxFont = NULL;
  setClipRect();
  setOrigin();

  // Added to see how much impact actually using non hardware CS pin might be
  //_cspinmask = 0;
  //_csport = NULL;

#ifdef ENABLE_ILI9341_FRAMEBUFFER
  _pfbtft = NULL;
  _use_fbtft = 0; // Are we in frame buffer mode?
  _we_allocated_buffer = NULL;
#endif
}
```
I now have it like:
```
ILI9341_GIGA_n::ILI9341_GIGA_n(uint8_t cs_pin, uint8_t dc_pin, uint8_t rst_pin) :
    _cs(cs_pin), _dc(dc_pin), _rst(rst_pin) 
{
}
```
And I initialized all of the member variables.  Some probably wrong... but.
I did this by trying to set the constructor to constepxr,
However I could not do this without changing Print::Print to constexpr as the class is derived from print.

I also could not leave the constructor as constexpr as it left a linker error (both for zephyr as well as MBED)

But it now runs. 

Secondary or tertiary comment for this.  With the current setup.  This as well as 2.3.4 IDE, I have finding I have to more often 
than before define forward references in the sketch file.  Like in the graphics test sketch I had to add:
```
// forward references:
extern unsigned long testFillScreen();
extern unsigned long testText();
extern unsigned long testLines(uint16_t color);
extern unsigned long testFastLines(uint16_t color1, uint16_t color2);
extern unsigned long testRects(uint16_t color);
extern unsigned long testFilledRects(uint16_t color1, uint16_t color2);
extern unsigned long testFilledCircles(uint8_t radius, uint16_t color);
extern unsigned long testCircles(uint8_t radius, uint16_t color);
extern unsigned long testTriangles();
extern unsigned long testFilledTriangles();
extern unsigned long testRoundRects();
extern unsigned long testFilledRoundRects();
```
The library and example sketch are up on my github project:
https://github.com/KurtE/Arduino_GIGA-stuff/tree/main/libraries/ILI9341_GIGA_zephyr

mjs513, 2024-12-13:
@KurtE - @facchinm 

As a quick and dirty test I put together a simple datalogger using the SdFat lib an the MPU9250 with the giga - used cspin 7 for the sdcard and cspin 4 for the 9250.  Everything seems to be working at least with 2 spi devices on the same bus.
```
Before SD Begin
Setup done

Menu Options:
	s - Start Logging data (Restarting logger will append records to existing log)
	x - Stop Logging data
	d - Dump Log
	h - Menu

Logging Data!!!

Stopped Logging Data!!!

Dumping Log!!!
1	0	0.15	-0.56	-10.08	0.01	0.02	-0.03	0.00	0.00	0.00	25.87
1	0	0.12	-0.57	-10.15	0.01	0.02	-0.03	0.00	0.00	0.00	25.86
1	0	0.14	-0.58	-10.14	0.01	0.02	-0.03	0.00	0.00	0.00	25.88
1	0	0.14	-0.64	-10.16	0.01	0.02	-0.03	0.00	0.00	0.00	25.87
1	1	0.12	-0.58	-10.10	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.87
1	0	0.15	-0.57	-10.09	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.88
1	0	0.16	-0.61	-10.18	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.87
1	0	0.13	-0.60	-10.13	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.86
1	0	0.11	-0.57	-10.17	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.87
1	0	0.15	-0.58	-10.16	0.01	0.02	-0.03	-17.69	30.98	-18.13	25.88
1	1	0.10	-0.59	-10.23	0.01	0.02	-0.03	-17.69	31.69	-17.44	25.87

```
as nothing fancy just a test of 2 devices on the same bus.

Going to repeat what @KurtE  said
>Secondary or tertiary comment for this. With the current setup. This as well as 2.3.4 IDE, I have finding I have to more often than before define forward references in the sketch file. Like in the graphics test sketch I had to add:

For this sketch I had to add:
```
void logData();
void stopLogging();
void dumpLog();
void menu();
```

Not sure if you are planning on changing this so it matches up with Arduino so you don't have to forward declare references.

KurtE, 2024-12-30:
@facchinm @mjs513 - I think most of the stuff mentioned in this thread has been fixed?

Except maybe the configuration of sketches .conf file settings, which is or should be it's own issue?

Wondering if maybe this one should be closed?

mjs513, 2024-12-30:
@KurtE - @facchinm 

Was just waiting to recommend closure until PR was actually merged.

As for the file settings, yeah that is still an open issue and probably should be its own issue.

mjs513, 2025-01-29:
See issue https://github.com/arduino/ArduinoCore-zephyr/issues/24.

SPI seems to work with SDFAT however SPI1 doe not.  With LA attached not seeing and signals on MISO, MOSI or CLK.  Seems to hang.

KurtE, 2025-01-29:
@facchinm @mjs513  - yep it looks like my ILI9341 and the like code is not working on SPI1 as well.

facchinm, 2025-01-29:
Did it work before the latest mass rebuild?

mjs513, 2025-01-29:
> Did it work before the latest mass rebuild?
yes it did for both sdfat and @KurtE's ILI9341 test sketch.   Note we did do a rebuild since we are from from the sources.

KurtE, 2025-01-29:
Sorry was off doing some other fun things, like go to grocery store. 

So far I have tracked the SPI1 crash/hang on GIGA to somewhere within SPI::transfer,  will add some more debug stuff

KurtE, 2025-01-30:
Not sure if best to continue here or to create new issue?

In the file <zeprhyr>/drivers/spi/spi_ll_stm32.c.
in the transceive function:
It gets to:
```
	printk("\tCall ll_func_enable_int_tx_empty\n");
	ll_func_enable_int_tx_empty(spi);

	printk("\tCall spi_context_wait_for_completion\n");
	ret = spi_context_wait_for_completion(&data->ctx);

```
```
uart:~$ sketch
[00:00:07.543,000] <inf> usb_cdc_acm: Device suspended
Enter transceive(0x805e48c 0x24013734 0x2400d618 0x2400d628)
        Call LL_SPI_StartMasterTransfer
        returned - Wait while !
        IsActiveMasterTransfer
        Call spi_stm32_cs_control
        Call ll_func_enable_int_errors
        Call ll_func_enable_int_tx_empty

```
the function: ll_func_enable_int_tx_empty
turns on an interrupt bit:
```
static inline void ll_func_enable_int_tx_empty(SPI_TypeDef *spi)
{
#if DT_HAS_COMPAT_STATUS_OKAY(st_stm32h7_spi)
	LL_SPI_EnableIT_TXP(spi);
#else
	LL_SPI_EnableIT_TXE(spi);
#endif /* st_stm32h7_spi */
}
```
And I am guessing the interrupt triggers and maybe not handled?  As we never come back to do the next print

KurtE, 2025-01-30:
@mjs513 @facchinm and ...

Another piece to the puzzle.  SPI object does not have same issue... 
Next up look again how both of them are defined...

And still wondering what changed...

KurtE, 2025-01-31:
Another piece of the puzzle,
When it is computing the clocks:
For SPI (spi1) it use 240mhz, but for SPI1(spi5) it use 120mhz, so maybe spi5 is setup to be controlled by the
M4...

Not sure where that is controlled yet...

KurtE, 2025-01-31:
Maybe the dtsh output might help:
https://blog.golioth.io/dtsh-a-devicetree-viewer-for-zephyr/
Here is a treeview of my current GIGA setup
```
                                       Also Known As               Bus          Binding
                                       ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
/                                      …                           …            …
├── chosen                             …                           …            …
├── aliases                            …                           …            …
├── soc                                …                           …            …
│   ├── interrupt-controller@e000e100  nvic                        …            arm,v7m-nvic [↗]
│   ├── timer@e000e010                 systick                     …            arm,armv7m-systick [↗]
│   ├── flash-controller@52002000      flash                       …            st,stm32-flash-controller [↗]
│   │   └── flash@8000000              flash0                      …            st,stm32-nv-flash [↗]
│   │       └── partitions             …                           …            fixed-partitions [↗]
│   │           ├── partition@0        bootloader, boot_partition  …            Each child node of the fixed-partitions node represents… [↗]
│   │           ├── partition@40000    image-0, slot0_partition    …            Each child node of the fixed-partitions node represents… [↗]
│   │           └── partition@e0000    user_sketch                 …            Each child node of the fixed-partitions node represents… [↗]
│   ├── rcc@58024400                   rcc                         …            st,stm32h7-rcc [↗]
│   │   └── reset-controller           rctl                        …            st,stm32-rcc-rctl [↗]
│   ├── interrupt-controller@58000000  exti                        …            st,stm32-exti [↗]
│   ├── pin-controller@58020000        pinctrl                     …            st,stm32-pinctrl [↗]
│   │   ├── gpio@58020000              gpioa                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58020400              gpiob                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58020800              gpioc                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58020C00              gpiod                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58021000              gpioe                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58021400              gpiof                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58021800              gpiog                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58021C00              gpioh                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58022000              gpioi                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58022400              gpioj                       …            st,stm32-gpio [↗]
│   │   ├── gpio@58022800              gpiok                       …            st,stm32-gpio [↗]
│   │   ├── adc1_inp16_pa0             adc1_inp16_pa0              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp0_pa0_c            adc1_inp0_pa0_c             …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp1_pa1_c            adc1_inp1_pa1_c             …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp18_pa4             adc1_inp18_pa4              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp19_pa5             adc1_inp19_pa5              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp9_pb0              adc1_inp9_pb0               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp5_pb1              adc1_inp5_pb1               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp10_pc0             adc1_inp10_pc0              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp12_pc2             adc1_inp12_pc2              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp13_pc3             adc1_inp13_pc3              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp4_pc4              adc1_inp4_pc4               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc1_inp8_pc5              adc1_inp8_pc5               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc3_inp0_pc2_c            adc3_inp0_pc2_c             …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── adc3_inp1_pc3_c            adc3_inp1_pc3_c             …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── dac1_out1_pa4              dac1_out1_pa4               …            This binding gives a base representation of the STM32 pins configuration [↗]
[... cut ...]
│   └── led_2                          blue_led                    …            GPIO LED child node [↗]
├── gpio_keys                          …                           …            gpio-keys [↗]
│   └── button_0                       user_button, sw0            …            GPIO KEYS child node [↗]
├── sdram@c0000000                     sdram1                      …            zephyr,memory-region [↗]
├── gpio@deadbeef                      gpioz                       …            vnd,gpio [↗]
└── zephyr,user                        …                           …            …
```

mjs513, 2025-02-01:
@KurtE  -  @facchinm 

Looking at the devicetree I noticed that spi5 is on timer1 (think thats whats it telling me.

```
│   │   ├── spi1_miso_pg9              spi1_miso_pg9               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── spi5_miso_pj11             spi5_miso_pj11              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── spi1_mosi_pd7              spi1_mosi_pd7               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── spi5_mosi_pj10             spi5_mosi_pj10              …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── spi1_sck_pb3               spi1_sck_pb3                …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── spi5_sck_ph6               spi5_sck_ph6                …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── tim1_ch3_pj9               tim1_ch3_pj9                …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── tim1_ch2_pj11              tim1_ch2_pj11               …            This binding gives a base representation of the STM32 pins configuration [↗]
│   │   ├── tim1_ch1_pk1               tim1_ch1_pk1                …            This binding gives a base representation of the STM32 pins configuration [↗]
```

In the camera PR timer1 prescaler was changed from 4 to 0:
[image]

as well as associated PWM:
[image]

Wonder if this is affecting SPI1 on the giga board

mjs513, 2025-04-26:
Going to close this for now

closed 2025-04-26 by mjs513