arduino/ArduinoCore-zephyr #5: SDFat library doesn't work

facchinm, 2024-12-09:
**Describe the bug**
SDFat examples (like QuickStart) compile but don't run

**Target board + cli verbose compilation output**
**Full verbose** compilation output, ideally with `arduino-cli` invocation or from IDE 2.3.3+
```
**Output of Serial Monitor**
```

**Output of readelf**
You can find the loaction of the elf file by compiling in Verbose mode and looking near the end of the compilation output (after `Linking everything together..`)
Paste (or attach) the output of `arm-none-eabi-readelf -a $your_sketch_elf_file`

**Optional: attach the elf file**

**Optional: attach the sketch**

**Additional context**
Add any other context about the problem here.

mjs513, 2024-12-09:
just posted [this ](https://github.com/arduino/ArduinoCore-zephyr/issues/4)as well where I am seeing no output on the serial monitor but the error you are seeing sounds reasonble.

facchinm, 2024-12-09:
How to debug this issue:
* with UART connected -> we see the shell, type `sketch`, get ```[00:00:01.528,000] <err> llext: Failed to copy regions, ret -12                                             
Failed to load sketch, rc -12                                                                               
  * Reason: the sketch is too big to fit in LLEXT HEAP (the one that will be used when interacted via shell)
  * Solution: temporarily reduce `CONFIG_HEAP_MEM_POOL_SIZE` to 2048, increase `CONFIG_LLEXT_HEAP_SIZE` to 64
* next error:
  * ```[00:00:03.332,000] <err> llext: Undefined symbol with no entry in symbol table isalpha, offset 13200, link 0
[00:00:03.343,000] <err> llext: Failed to link, ret -61                                                     
Failed to load sketch, rc -61                                                                               
```
  * Reason: symbol not yet exported in `llext_exports.c`
  * Solution: add `isaplha` and `isupper` to that file

facchinm, 2024-12-09:
@mjs513 ouch, too fast 😄 Closing this one in favour of #4

closed 2024-12-09 by facchinm

mjs513, 2024-12-09:
Sorry - been at this for the last week or so :)  But looks like you got the fix -  thank you.