arduino/ArduinoCore-zephyr #8: Is it possible not to use llext-edk?

Pillar1989, 2024-12-10:
I am very happy to see the collection of arduino and zephyr.
I just opened this core and had a deep look at its operation principle.  Arduino application runs on the zephyr rtos system as a module. The technology is still exist many problems. 
https://docs.zephyrproject.org/latest/services/llext/load.html#troubleshooting
 is it possible to not use llext technology?  arduino main runs as a main application.

facchinm, 2024-12-10:
Hi @Pillar1989 ,
the original project from which we fork from had this exact scope originally, but of course it comes with its own shortcomings (you'll be able to create a zephyr project using Arduino libraries and language, but inside the zephyr toolchain).
We strive to keep things compatible but it's not something we consider mandatory at the moment.
On the other hand, the llext subproject is advancing very fast so hang tight 🙂

Pillar1989, 2024-12-10:
When will it be released? @facchinm

closed 2025-05-12 by Pillar1989