arduino/ArduinoCore-zephyr #15: Sketch is too big to fit in LLEXT HEAP

mjs513, 2024-12-15:
When testing SDFat and Sparkfun BMI270 libraries following error was found (see https://github.com/arduino/ArduinoCore-zephyr/issues/4#issuecomment-2528211591).  @facchinm identified the temp fix and issue as:

- with UART connected -> we see the shell, type sketch, get

`[00:00:01.528,000] <err> llext: Failed to copy regions, ret -12                                              Failed to load sketch, rc -12`

-- Reason: the sketch is too big to fit in LLEXT HEAP (the one that will be used when interacted via shell)

-- Solution: temporarily reduce CONFIG_HEAP_MEM_POOL_SIZE to 2048, increase CONFIG_LLEXT_HEAP_SIZE to 64

However when making the change to CONFIG_LLEXT_HEAP_SIZE to 64 error still occurred when attempting to run a Sparkfun BMI270 test sketch.  Increasing to 128 resolved that issue.  In addition looking at the Portenia H7 conf followwing change was made to the Giga conf:
```
CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_SHELL_STACK_SIZE=32768
CONFIG_MAIN_STACK_SIZE=32768
CONFIG_LLEXT_HEAP_SIZE=128
```

This probably needs to be changed in the Config before a beta release

KurtE, 2024-12-15:
Note: The settings that @mjs513 were also needed to get my SD to ILI9341 picture viewer to work.  At times
it would limp along at 96 with only BMP files supported.  However with the 128 it works with both BMP and JPEG (JPEGDEC library), but still fails if I also try to enable PNG files (PNGDEC library).

mjs513, 2025-01-29:
This seems to be resolved for now with the changes made.  Noted that when the production versions come out these may change.

closed 2025-01-29 by mjs513