arduino/ArduinoCore-zephyr #20: Arduino Giga R1 WiFi Not Working.

mjs513, 2024-12-23:
Using the Wifi scan network example:
```
/*
 This example  prints the board's MAC address, and
 scans for available WiFi networks using the GIGA R1 WiFi board.
 Every ten seconds, it scans again. It doesn't actually
 connect to any network, so no encryption scheme is specified.

 Circuit:
 * GIGA R1 WiFi

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 21 June 2012
 by Tom Igoe and Jaymes Dec
 modified 3 March 2023
 by Karl Söderby
 */

#include <SPI.h>
#include <WiFi.h>

void printMacAddress(byte mac[]);
void listNetworks();
void printMacAddress(byte mac[]);

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // print your MAC address:
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  printMacAddress(mac);
}

void loop() {
  // scan for existing networks:
  Serial.println("Scanning available networks...");
  listNetworks();
  delay(10000);
}

void listNetworks() {
  // scan for nearby networks:
  Serial.println("** Scan Networks **");
  int numSsid = WiFi.scanNetworks();
  if (numSsid == -1) {
    Serial.println("Couldn't get a WiFi connection");
    while (true);
  }

  // print the list of networks seen:
  Serial.print("number of available networks:");
  Serial.println(numSsid);

  // print the network number and name for each network found:
  for (int thisNet = 0; thisNet < numSsid; thisNet++) {
    Serial.print(thisNet);
    Serial.print(") ");
    Serial.print(WiFi.SSID(thisNet));
    Serial.print("\tSignal: ");
    Serial.print(WiFi.RSSI(thisNet));
    Serial.print(" dBm");
    Serial.print("\tEncryption: ");
    printEncryptionType(WiFi.encryptionType(thisNet));
  }
}

void printEncryptionType(int thisType) {
  // read the encryption type and print out the name:
  switch (thisType) {
    case ENC_TYPE_WEP:
      Serial.println("WEP");
      break;
    case ENC_TYPE_TKIP:
      Serial.println("WPA");
      break;
    case ENC_TYPE_CCMP:
      Serial.println("WPA2");
      break;
    case ENC_TYPE_NONE:
      Serial.println("None");
      break;
    case ENC_TYPE_AUTO:
      Serial.println("Auto");
      break;
    case ENC_TYPE_UNKNOWN:
    default:
      Serial.println("Unknown");
      break;
  }
}

void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}
```
Ran into issues on initial run.

1.  llext_exports.c - several symbols were not defined so added the following:
```
EXPORT_SYMBOL(atoi);
EXPORT_SYMBOL(islower);
EXPORT_SYMBOL(memcpy);
EXPORT_SYMBOL(memset);
EXPORT_SYMBOL(strtok);
EXPORT_SYMBOL(strtol);
```
2.  Keep having to put giga into boot mode to load the sketch.

Once they were added and recompiled core received following error::
```
[m
[1;32muart:~$ [msketch
[00:00:13.562,000] [1;33m<wrn> usb_device: USB device support already enabled[0m
Communication with WiFi module failed!
```
then realized the PR does not setup wifi for the giga.

mjs513, 2024-12-28:
If I update conf to include `CONFIG_WIFI=y` I get several linker errors
```
[233/238] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/my_new_zephyr_folder/ArduinoCore-zephyr/build/zephyr/zephyr_pre0.map
: && ccache /home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/my_new_zephyr_folder/ArduinoCore-zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/options/liblib__posix__options.a  zephyr/subsys/llext/libsubsys__llext.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/dma/libdrivers__dma.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/memc/libdrivers__memc.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/pwm/libdrivers__pwm.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  zephyr/kernel/libkernel.a  -Wl,--no-whole-archive  zephyr/libsyscall_weakdefs.a  -L/home/my_new_zephyr_folder/ArduinoCore-zephyr/build/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfpu=fpv5-d16  -mfloat-abi=hard  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_DOUBLE_PRINTF_SCANF -L"/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m+dp/hard" -lc -lgcc && cd /home/my_new_zephyr_folder/ArduinoCore-zephyr/build/zephyr && /usr/bin/cmake -E true
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(llext_exports.c.obj):(._llext_const_symbol.static.__llext_sym_net_mgmt_NET_REQUEST_WIFI_IFACE_STATUS_+0x4): undefined reference to `net_mgmt_NET_REQUEST_WIFI_IFACE_STATUS'
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(llext_exports.c.obj):(._llext_const_symbol.static.__llext_sym_net_mgmt_NET_REQUEST_WIFI_CONNECT_+0x4): undefined reference to `net_mgmt_NET_REQUEST_WIFI_CONNECT'
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(llext_exports.c.obj):(._llext_const_symbol.static.__llext_sym_net_if_get_wifi_sta_+0x4): undefined reference to `net_if_get_wifi_sta'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/my_new_zephyr_folder/ArduinoCore-zephyr/build
```

Not `net_mgmt_NET_REQUEST_WIFI_CONNECT` is defined in the llext_exports.c file.  WOuld normally just add but...

mjs513, 2025-01-02:
@facchinm - @KurtE

Decided to try and get Wifi working on the Giga since I found this: https://github.com/zephyrproject-rtos/zephyr/issues/77012 which gave me a hint on how to proceed. 

Modified giga config to add:
```

CONFIG_NETWORKING=y
CONFIG_WIFI=y
CONFIG_WIFI_AIROC=y
CONFIG_WIFI_OFFLOAD=y
CONFIG_NET_OFFLOAD=y

CONFIG_CYW4343W=y
CONFIG_CYW4343W_MURATA_1DX=y

#CONFIG_CPP=y
#CONFIG_GLIBCXX_LIBCPP=y
```
and in the overlay added:
```
&sdmmc1 {
	pinctrl-0 = <&sdmmc1_d0_pc8 &sdmmc1_d1_pc9
		&sdmmc1_d2_pc10 &sdmmc1_d3_pc11
		&sdmmc1_d4_pb8 &sdmmc1_ck_pc12
		&sdmmc1_cmd_pd2>;
	pinctrl-names = "default";
	status = "okay";

	//clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00010000>,
	//		<&rcc STM32_SRC_PLL2_R SDMMC_SEL(1)>;

	/* Wifi configuration */
	airoc-wifi {
		status = "okay";
		compatible = "infineon,airoc-wifi", "infineon,airoc-wifi-sdio";

		/* Wi-Fi control gpios */
		wifi-reg-on-gpios    = <&gpiob 10 GPIO_ACTIVE_HIGH>;
		wifi-host-wake-gpios = <&gpioi 8 GPIO_ACTIVE_HIGH>;
	};

};
```
still experimenting with that - not sure about the clock or the compaitible set up yet.

On first try got undefined symbols similar to what I saw with bluetooth for
```
//FORCE_EXPORT_SYM(net_if_ipv4_maddr_add);
//FORCE_EXPORT_SYM(net_if_ipv4_maddr_join);
```
so commented them out as a test.

however received 2 core compile errors that I cant figure out:
```
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/wifi/libdrivers__wifi.a(airoc_wifi.c.obj): in function `airoc_init':
/home/my_new_zephyr_folder/zephyr/drivers/wifi/infineon/airoc_wifi.c:780: undefined reference to `airoc_wifi_init_primary'
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(llext_exports.c.obj):(._llext_const_symbol.static.__llext_sym___cxa_pure_virtual_+0x4): undefined reference to `__cxa_pure_virtual'
```

the __cxa_pure_virtual @KurtE already identified as an issue.

KurtE, 2025-01-02:
As for: airoc_wifi_init_primary

Looking at the CmakeLists.txt file in the directory I see:
```
zephyr_library_sources_ifdef(CONFIG_AIROC_WIFI_BUS_SDIO airoc_whd_hal_sdio.c)
zephyr_library_sources_ifdef(CONFIG_AIROC_WIFI_BUS_SPI airoc_whd_hal_spi.c)
```
Guessing you need to add either CONFIG_AIROC_WIFI_BUS_SDIO or CONFIG_AIROC_WIFI_BUS_SPI 
to the giga config file:  Looking at schematic I believe the SDIO version.

mjs513, 2025-01-02:
I did try adding in CONFIG_AIROC_WIFI_BUS_SDIO but the build complains about 
>error: AIROC_WIFI_BUS_SDIO (defined at drivers/wifi/infineon/Kconfig.airoc:21) is assigned in a
configuration file, but is not directly user-configurable (has no prompt). It gets its value
indirectly from other symbols. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_AIROC_WIFI_BUS_SDIO and/or look up
AIROC_WIFI_BUS_SDIO in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

and looks like it enabled by default: see https://docs.zephyrproject.org/latest/kconfig.html and 
```
ONFIG_AIROC_WIFI_BUS_SDIO[](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_AIROC_WIFI_BUS_SDIO)
No prompt - not directly user assignable.

Enable SDIO bus support.

Type
bool
Dependencies
n
Defaults
y
Selects
[CONFIG_SDHC](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SDHC)
[CONFIG_SDIO_STACK](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SDIO_STACK)
```

facchinm, 2025-01-07:
WiFi support for STM32 needs an SDIO driver which is not yet upstramed; you can follow the discussion in zephyr here https://github.com/zephyrproject-rtos/zephyr/discussions/77079#discussioncomment-10367964  .
Closing the issue for the time being to avoid confusion

closed 2025-01-07 by facchinm

mjs513, 2025-01-07:
@facchinm  - thanks for the info.  Will keep an eye out.  Guess estimate that will be completed by end of year is out the window

rlinus, 2025-09-18:
I got wifi working by adding the following lines to prj.conf:

`
# Enable kernel heap so k_malloc / k_aligned_alloc are provided
CONFIG_HEAP_MEM_POOL_SIZE=131072
# 128 KiB is a safe starting point for WHD; adjust if tight

# (Optional but often needed with Wi-Fi)
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096

# it must be CONFIG_WIFI_INIT_PRIORITY > CONFIG_SDHC_INIT_PRIORITY (=80), which is not true by default
CONFIG_WIFI_INIT_PRIORITY=90
`