arduino/ArduinoCore-zephyr #23: setClock does not appear to do anything on the Arduino Giga

mjs513, 2024-12-25:
Using a MPU9250 as my I2c test device I tested using SCL/SDA and SCL1/SDA1.  Seems as though i2c clock speed is hardcoded someplace.

SCL/SDA - i2c2
In the overlay default clock is set to fast or 400khz

Setting the clock to 100Khz using setClock(100000) or setClock(i2c_standard)  shows an actual clock of 400khz
[image]

Setting it 400khz shows same clock
[image]

SCL1/SDA1 -  i2c4
No default clock is set and seeing 400khz when I do a setclock of 100khz.
[image]

mjs513, 2024-12-26:
To resolve the setClock issue I implemented this piece of code for set Clock in wire.cpp
```
void arduino::ZephyrI2C::setClock(uint32_t freq) {
    uint8_t speed = 	I2C_SPEED_STANDARD;
  if(freq == 100000) {
    speed = I2C_SPEED_STANDARD;
  } else if(freq == 400000) {
    speed = I2C_SPEED_FAST;
  } else if(freq == 1000000) {
    speed = I2C_SPEED_FAST_PLUS;
  } else {
    speed = I2C_SPEED_STANDARD;
  }
	uint32_t i2c_cfg = I2C_MODE_CONTROLLER |
					I2C_SPEED_SET(speed);

	if (i2c_configure(i2c_dev, i2c_cfg)) {
		//Serial.println("Failed to configure i2c interface.");
	}
}
```
also had to add this to llext_exports.c
```
FORCE_EXPORT_SYM(__aeabi_ldivmod);
```
seems to be working for 100khz and 400khz.  Tried 1Mhz but my LA read 111khz??  Tested on Wire and Wire1 with the following sketch:
```

facchinm, 2025-01-08:
Hi Mike, thanks for the investigation! Can I ask you to prepare a PR for the `setClock` implementation? As usual, to preserve proper attribution 🙂 Thank you!

closed 2025-01-09 by facchinm