arduino/ArduinoCore-zephyr #28: GIGA: C++ Zephyr Hello World Sample produces multilple errors.

mjs513, 2025-01-01:
**Describe the bug**
Running the C++ Hello World Zephyr example: [Hello C++ world](https://docs.zephyrproject.org/latest/samples/cpp/hello_world/README.html#hello_cpp_world)Print "Hello World" to the console in C++.  But modified to use setup and loop yields multiple errors when compiling.

**Target board + cli verbose compilation output**
Target Board: Arduino GIGA R1

Verbose output - too large to paste.
[Hello_world_cpp_verbose.txt](https://github.com/user-attachments/files/18284186/Hello_world_cpp_verbose.txt)

**Mandatory: attach the sketch**
```
#include <iostream>

void setup()
{
  Serial.begin(9600);
	std::cout << "Hello, C++ world! " << CONFIG_BOARD << std::endl;
}

void loop(){}
```

**Additional context**
Tried using CONFIG_CPP=y in giga config file but get errors whether using it or not.

facchinm, 2025-01-07:
This is one of the known issues (see https://github.com/arduino/ArduinoCore-zephyr?tab=readme-ov-file#-upcoming-features ->  Fix corner cases with std:: includes (like <iterator>) ) .
It looks like somehow in the last ld invocation the information about `-fnoexception` gets lost, so the system tries to link nonexisting stuff.
Working on a fix 🙂

mjs513, 2025-01-07:
Just incorporated https://github.com/arduino/ArduinoCore-zephyr/pull/17/files/e653548709ed2d97f68b9cff64cf4778285e4b4f..ed1193730c31d50925f23e304bdaa413f18c6c91 change and gave it a try and fixed this issue:

```
[1;32muart:~$ [msketch
Create semaphore 0x2400ec1c
Create semaphore 0x2400ec00
[00:00:06.712,000] [1;33m<wrn> usb_device: USB device support already enabled[0m
setup: Hello World!
coop_thread_entry: Hello World!
setup: Hello World!
coop_thread_entry: Hello World!
setup: Hello World!
```
probably should test with the following commented out as well:
```
CONFIG_CPP=y
CONFIG_STD_CPP17=y
CONFIG_GLIBCXX_LIBCPP=y
```

facchinm, 2025-01-07:
@mjs513 I think you are referring to the other issue 🙂 This one is a bit more convoluted

mjs513, 2025-03-31:
Retesting with 0.3.0 shows a different error now:
```
  [1;32muart:~$ [msketch
  [00:01:32.233,000] [1;31m<err> llext: Undefined symbol with no entry in symbol table _ZNSt8ios_base4InitC1Ev, offset 16, link section 12[0m
  [00:01:32.233,000] [1;31m<err> elf: Could not find symbol _ZNSt8ios_base4InitC1Ev![0m
  [00:01:32.233,000] [1;31m<err> llext: Failed to link, ret -61[0m
  Failed to load sketch, rc -61
```
So guess with have to leave open

mjs513, 2025-05-15:
as May 15  error shown in Mar 31 post still present

closed 2025-05-15 by mjs513