arduino/ArduinoCore-zephyr #33: Tools->Mode->Debug appears to be stuck on

KurtE, 2025-01-16:
I tried it out for a while on my first GIGA board.  To see what all I could do with the
debug monitor, with a working sketch.

It now appears to be permanently in that mode.  That is no matter what zephyr sketch I install, it will always
just show a prompt at the uart:~$ 
And wait for me to type sketch<return>

Note: I have downloaded many versions of the firmware to this board.  I have run it on both Windows and Ubuntu and I have down clean downloads, bootstrap and builds of the ArduinoCore-zephyr code base.

Also my other Giga, does not require me to enter something, at the prompt, unless there is a problem.

KurtE, 2025-03-06:
@facchinm @pillo79 - Note: I am still having this issue.  That maybe on both of my GIGA boards, I have to have debug monitor open and type: sketch to get it up and running, regardless of the state of Debug/Standard

Any ideas on how to reset this?

pillo79, 2025-03-06:
Will publish a major update soon on this!

mjs513, 2025-05-15:
@pillo79 - @facchinm 

this having this issue as of May 15.  Any status - its really getting annoying.  I do have a second Giga which has the same issue.

KurtE, 2025-05-15:
@pillo79 @mjs513 - Yes that is the case.
Note: this happens on my older GIGA, and I think that is true with @mjs513 as well.  The one I purchased later
does not have this issue.

So not sure if this has anything to do with layout and the like, or if there is some subtle hardware difference between
the earlier GIGA's,  like I don't know what, maybe timing, maybe fuse settings, ...

mjs513, 2025-05-15:
@pillo79 @mjs513 - Yes that is the case (ditto)

On the older Giga going to walk you through the process
1.  Burn bootloader - no com port enumerated.
2. Loaded blink by putting it into boot mode and loading it via standard and sketch does not run
[image]
3.  repeating by running in debug  but no Com port again
[image]

typing sketch in debug window - I get com port and sketch runs
[image]

closed 2025-09-29 by KurtE