arduino/ArduinoCore-zephyr #38: GIGA: analogWrite to DAC pins (A12, A13) does not appear to be implemented.

KurtE, 2025-01-18:
At one point I thought I saw some DAC code in place, but now I don't see it:

Tried simple sketch: 
```
int sensorPin = A0;   // select the input pin for the potentiometer
int sensorValue = 0;  // variable to store the value coming from the sensor
int sensorValuePrev = -1;

int DACPin = A12;
void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  sensorValue = analogRead(sensorPin);
  if (sensorValue != sensorValuePrev) {
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    analogWrite(DACPin, sensorValue);
    Serial.println(sensorValue);
    sensorValuePrev = sensorValue;
  }
  delay(100);
}
```
Which faulted the processor.  As A12 was not found in the list of PWM pins. This is covered in issue #19 
Note: I pushed up a fix for the fault as part of my current PR: #32

facchinm, 2025-01-22:
At the moment `analogWrite` only handles PWM, will work on a patch for DAC very soon 🙂

KurtE, 2025-01-22:
@facchinm - I could be missing something, but I don't believe that there is any support at least for the
giga for PWM, except for the one Camera pin and maybe it works for those on the same timer. 

I was playing with this awhile back and more data in the thread:
https://github.com/arduino/ArduinoCore-zephyr/issues/19#issuecomment-2552366603
I tried several different things, but found lots of issues and stuff I was unsure how to solve.

First issue, the table:
```
		pwm-pin-gpios =	    <&gpioj 9 0>,
					<&arduino_header 6 0>,
				    <&arduino_header 5 0>,
				    <&arduino_header 17 0>,
				    <&arduino_header 12 0>,
				    <&arduino_header 2 0>,
				    <&arduino_header 3 0>,
				    <&arduino_header 9 0>,
				    <&arduino_header 10 0>;

```
First one is your camera one, 
But the others using Arduino_header pins, 0-5 are A0-A5,
So, the pins in this list correspond to: (D57, D0, A5, D11, D6, A2, A3, D3, D4)
Which I don't believe is what you want.  I believe D2-D13, which I see by tables as well
as: https://docs.arduino.cc/tutorials/giga-r1-wifi/cheat-sheet/#pwm-pins

PR #46 (2025-01-24): Fix analogWrite()
Implement DAC and fix the dtoverlay for Giga

Fixes #38 and #19
closed 2025-01-24 by facchinm