arduino/ArduinoCore-zephyr #44: State of the builds?  Should we be running build_all.sh?

KurtE, 2025-01-23:
Sorry quick question(s)?

For the most part I have been building only for the GIGA board. 

But this morning, I thought I would try to run the build_all.sh script as to see the state of the builds. 

And several of the builds are failing:
From the script:
```
./extra/build.sh arduino_giga_r1//m7 arduino_giga_r1_m7
./extra/build.sh arduino_nano_33_ble//sense arduino_nano_33_ble_sense
./extra/build.sh arduino_portenta_h7//m7 arduino_portenta_h7
./extra/build.sh ek_ra8d1 ek_ra8d1
./extra/build.sh frdm_mcxn947/mcxn947/cpu0 frdm_mcxn947_mcxn947_cpu0
./extra/build.sh frdm_rw612 frdm_rw612
```
GIGA builds, but Nano and Portenta fail.

Did not copy down what failed on NANO but on Portenta failed with undefined:
` undefined reference to `__cxa_pure_virtual
`

I was curious about these other two boards, as I do have a Portenta H7 (currently on HAT carrier but soon will have breakout as well)
Nano have V1 of BLE (not sure if you need sense?) but I have a V2 Sense arriving today. 

Don't know what the others are, so probably won't worry about them.  Other WIP boards?

Main questions:

1) Should we continue with emphasis mainly on GIGA? 
2) Is there some place to check to know current state of each of the boards.  Like I saw zephyr branch for GIGA display shield.  What has been pulled into the main branch? 

Thanks

KurtE, 2025-01-23:
Quick update, was able to get Portenta and Nano to build by adding these two lines:
```
CONFIG_CPP=y
CONFIG_STD_CPP17=y
```
To their .conf files

Question 3) When we create PRs, should it include only the real files we changed or should it also include all of the files that were changed after you do the build.sh scripts?

facchinm, 2025-01-24:
Fixed by #47

closed 2025-01-24 by facchinm