arduino/ArduinoCore-zephyr #49: Nano 33 Sense v2:  I2c Scanner running for Wire only acting strangely

mjs513, 2025-01-24:
The only device I have on wire is a MPU9250.  

With the know issue that Wire1 is not defined yet ran the i2c scanner for Wire only and its returning all devices in the scan list as attached:
```
Scanning Wire...
Device found at address 0x02  (AS3935)
Device found at address 0x1D  (LSM303D,LSM9DS0,ADXL345,MMA7455L,LSM9DS1,LIS3DSH)
Device found at address 0x20  (MCP23017,MCP23008,PCF8574,FXAS21002,SoilMoisture)
Device found at address 0x22  (MCP23017,MCP23008,PCF8574)
Device found at address 0x24  (MCP23017,MCP23008,PCF8574,ADAU1966,HM01B0)
Device found at address 0x27  (MCP23017,MCP23008,PCF8574,LCD16x2,DigoleDisplay)
Device found at address 0x29  (TSL2561,VL6180,TSL2561,TSL2591,BNO055,CAP1188)
Device found at address 0x2C  (MCP44XX ePot)
Device found at address 0x2E  (MCP44XX ePot)
Device found at address 0x30  (Si7210)
Device found at address 0x32  (Si7210)
Device found at address 0x34  (MAX11612,MAX11613)
Device found at address 0x36  (unknown chip)
Device found at address 0x38  (RA8875,FT6206,MAX98390)
Device found at address 0x3A  (MLX90632)
Device found at address 0x3C  (SSD1306,DigisparkOLED)
Device found at address 0x3E  (unknown chip)
Device found at address 0x40  (PCA9685,Si7021,MS8607)
Device found at address 0x42  (PCA9685)
Device found at address 0x44  (PCA9685, SHT3X, ADAU1966)
Device found at address 0x47  (PCA9685)
Device found at address 0x4A  (ADS1115,Qwiic Keypad,CS42448)
Device found at address 0x4D  (unknown chip)
Device found at address 0x4F  (unknown chip)
Device found at address 0x51  (EEPROM)
Device found at address 0x53  (ADXL345,EEPROM)
Device found at address 0x55  (EEPROM)
Device found at address 0x57  (EEPROM)
Device found at address 0x59  (unknown chip)
Device found at address 0x5B  (unknown chip)
Device found at address 0x5D  (unknown chip)
Device found at address 0x5F  (unknown chip)
Device found at address 0x61  (MCP4725,AtlasEzoDO)
Device found at address 0x63  (MCP4725,AtlasEzoPH)
Device found at address 0x65  (unknown chip)
Device found at address 0x67  (unknown chip)
Device found at address 0x6A  (LSM9DS1)
Device found at address 0x6C  (unknown chip)
Device found at address 0x6E  (unknown chip)
Device found at address 0x70  (HT16K33,TCA9548A)
Device found at address 0x72  (HT16K33)
Device found at address 0x74  (unknown chip)
Device found at address 0x76  (MS5607,MS5611,MS5637,BMP280)
Device found at address 0x78  (unknown chip)
Device found at address 0x7A  (unknown chip)
Device found at address 0x7C  (FRAM_ID)
Device found at address 0x7E  (unknown chip)
done
```

sketch used:
```
// Wire Scanner - scans for I2C devices on all Wire ports
//
// This Wire library adapted is for Teensy boards
//
//   https://github.com/PaulStoffregen/Wire/
//
// Adapted from I2C Scanner originally published on Arduino Playground
// see comments below for link and credits for original authors.
#include <Arduino.h>
#include <Wire.h>

#define WIRE_INTERFACES_COUNT 1
void printKnownChips(byte address);
[... 179 lines cut ...]
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//
```

mjs513, 2025-01-24:
Out of curiosity I loaded up a MPU9250 sensor sketch using the bolderflight library and seems to still work. But don't understand the scanner.  Wonder what would happen if I put a second device on wire.  Ok later - need coffee.

KurtE, 2025-01-24:
I tried to copy in the i2c0 and i2c1 configuration stuff nothing helped.

I then looked at the Wire library, and noticed that the begin method does not do anything but initialize the right buffer.  
Any real init appears to be done on setting the clock:

So I changed your (@mjs513) sketch like:
```
// Wire Scanner - scans for I2C devices on all Wire ports
//
// This Wire library adapted is for Teensy boards
//
//   https://github.com/PaulStoffregen/Wire/
//
// Adapted from I2C Scanner originally published on Arduino Playground
// see comments below for link and credits for original authors.
#include <Arduino.h>
#include <Wire.h>

#define WIRE_INTERFACES_COUNT 2
void printKnownChips(byte address);
void scan(TwoWire &myport);

void setup() {
#if WIRE_INTERFACES_COUNT >= 1
  Wire.begin();
  Wire.setClock(I2C_SPEED_FAST);
#endif
#if WIRE_INTERFACES_COUNT >= 2
  Wire1.begin();
  Wire1.setClock(I2C_SPEED_FAST);
#endif
#if WIRE_INTERFACES_COUNT >= 3
  Wire2.begin();
#endif
#if WIRE_INTERFACES_COUNT >= 4
  Wire3.begin();
#endif
  Serial.begin(9600);
  while (!Serial);   // Wait for Arduino Serial Monitor
  Serial.println(F("\nI2C Scanner"));
}

void loop() {
  Serial.println();
#if WIRE_INTERFACES_COUNT >= 1
  Serial.println(F("Scanning Wire..."));
  scan(Wire);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 2
  Serial.println(F("Scanning Wire1..."));
  scan(Wire1);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 3
  Serial.println(F("Scanning Wire2..."));
  scan(Wire2);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 4
  Serial.println(F("Scanning Wire3..."));
  scan(Wire3);
  delay(500);
#endif
  delay(5000);           // wait 5 seconds for next scan
}

void scan(TwoWire &myport) {
  int nDevices = 0;
  for (int address = 1; address < 127; address++) {
[... 159 lines cut ...]
I2C Scanner

Scanning Wire...
No I2C devices found

Scanning Wire1...
Device found at address 0x10  (CS4272)
Device found at address 0x23  (MCP23017,MCP23008,PCF8574)
Device found at address 0x33  (MAX11614,MAX11615,Si7210,MLX90640,MLX90641)
Device found at address 0x39  (TSL2561, APDS9960)
Device found at address 0x4E  (unknown chip)
Device found at address 0x5C  (unknown chip)
Device found at address 0x68  (DS1307,DS3231,MPU6050,MPU9050,BMI270,ITG3200,ITG3701,LSM9DS0,L3G4200D)
Device found at address 0x7D  (unknown chip)
done

Scanning Wire...
No I2C devices found
```

@facchinm and ... 
Wonder if maybe the begin should set the clock?

Secondary note: I tried setting clock to STANDARD and sort of random stuff returned, like groups of every possible ID.
I will hook up LA to the Wire pins and see what speeds are being used

KurtE, 2025-01-24:
Quick update on it, I ran the MBED version and runs a lot quicker.
results:

```
Scanning Wire...
No I2C devices found

Scanning Wire1...
Device found at address 0x10  (CS4272)
Device found at address 0x39  (TSL2561, APDS9960)
Device found at address 0x44  (PCA9685, SHT3X, ADAU1966)
Device found at address 0x5C  (unknown chip)
Device found at address 0x68  (DS1307,DS3231,MPU6050,MPU9050,BMI270,ITG3200,ITG3701,LSM9DS0,L3G4200D)
done

```
More experimenting to do.

mjs513, 2025-01-24:
NIce @KurtE.  

Looking at it now and going to try getting wire1 to work.

As to your question on setting clock - just looked and agree with you probably should setClock to standard in wire begin as a default.

Think I know the reason we have to setclock for sense - for the giga we have a default speed set up in the overlay but for the sense don't believe they do that.

Not sure why standard is not working.

mjs513, 2025-01-25:
Just updated the device list in the sketch along with pinmode on wire1:
```
// Wire Scanner - scans for I2C devices on all Wire ports
//
// This Wire library adapted is for Teensy boards
//
//   https://github.com/PaulStoffregen/Wire/
//
// Adapted from I2C Scanner originally published on Arduino Playground
// see comments below for link and credits for original authors.
#include <Arduino.h>
#include <Wire.h>

#define WIRE_INTERFACES_COUNT 2
void printKnownChips(byte address);
void scan(TwoWire &myport);

void setup() {
  pinMode(31, OUTPUT);
#if WIRE_INTERFACES_COUNT >= 1
  Wire.begin();
  Wire.setClock(400000);
#endif
#if WIRE_INTERFACES_COUNT >= 2
  Wire1.begin();
  Wire1.setClock(400000);
#endif
#if WIRE_INTERFACES_COUNT >= 3
  Wire2.begin();
#endif
#if WIRE_INTERFACES_COUNT >= 4
  Wire3.begin();
#endif
  Serial.begin(9600);
  while (!Serial);   // Wait for Arduino Serial Monitor
  Serial.println(F("\nI2C Scanner"));
}

void loop() {
  Serial.println();
#if WIRE_INTERFACES_COUNT >= 1
  Serial.println(F("Scanning Wire..."));
  scan(Wire);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 2
  Serial.println(F("Scanning Wire1..."));
  scan(Wire1);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 3
  Serial.println(F("Scanning Wire2..."));
  scan(Wire2);
  delay(500);
#endif
#if WIRE_INTERFACES_COUNT >= 4
  Serial.println(F("Scanning Wire3..."));
  scan(Wire3);
  delay(500);
#endif
  delay(5000);           // wait 5 seconds for next scan
}

void scan(TwoWire &myport) {
  int nDevices = 0;
  for (int address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    myport.beginTransmission(address);
[... 128 lines cut ...]
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//
```

PR #50 (2025-01-29): BLE 33 - .overlay updates, enable Wire1, pins match MBED, change analogRead to 12 bit
Resolves: #49 #48

Note: @mjs513 has a version of the wire scanner that works with this.  
however Wire object takes a while to scan unless external PU resistors are connected to SCL/SDA pins.  
The sketch also turns on the Wire1 PU resistors and enables the Wire1 objects, using D32 and D33 below.
These pins are shown on the schematics for the different versions of the Nano 33 BLE boards.

Also @mjs513 found an interesting web page:
https://github.com/saltyJeff/zephyr_arduino_nano_33_ble/blob/main/src/init_ar [... cut]
closed 2025-01-29 by facchinm