arduino/ArduinoCore-zephyr #56: Nano 33 ble SPI fails after BMI270 Connected

mjs513, 2025-01-27:
**Describe the bug**
Using a sparkfun BMI270 sensor connected via spi hangs after device connected message.  Library used: Sparkfun BMI270.

**Target board + cli verbose compilation output**
**Full verbose** compilation output, ideally with `arduino-cli` invocation or from IDE 2.3.3+
Issues without the full verbose output will be discarded as invalid.
```
loading library from d:\Users\Merli\Documents\Arduino\libraries\Arduino-EasyTransfer-master: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\ArduinoCore-API-master: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\FastAhrs: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\Lidar-Lite-v3-3d-Scanner: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\M5Stack: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\esp32-camera-master: invalid library: no header files found
FQBN: arduino:zephyr:nano33ble
Using board 'nano33ble' from platform in folder: C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0
Using core 'arduino' from platform in folder: C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0

Detecting libraries used...
C:\Users\Merli\AppData\Local\Arduino15\packages\zephyr\tools\arm-zephyr-eabi\0.16.8/bin/arm-zephyr-eabi-g++ -g -Os -std=c++17 -c -w -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DNRF52840_XXAA -DKERNEL -DK_HEAP_MEM_POOL_SIZE=32768 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/deprecated -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/deprecated-avr-comp -iprefixC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense @C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/includes.txt C:\Users\Merli\AppData\Local\arduino\sketches\A1124A403BACF5E53EF339AB8AE1E885\sketch\Example02_BasicReadingsSPI.ino.cpp -o nul
Alternatives for SPI.h: [SPI]
ResolveLibrary(SPI.h)
  -> candidates: [SPI]
C:\Users\Merli\AppData\Local\Arduino15\packages\zephyr\tools\arm-zephyr-eabi\0.16.8/bin/arm-zephyr-eabi-g++ -g -Os -std=c++17 -c -w -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DNRF52840_XXAA -DKERNEL -DK_HEAP_MEM_POOL_SIZE=32768 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\libraries\SPI -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/deprecated -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/deprecated-avr-comp -iprefixC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense @C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/includes.txt C:\Users\Merli\AppData\Local\arduino\sketches\A1124A403BACF5E53EF339AB8AE1E885\sketch\Example02_BasicReadingsSPI.ino.cpp -o nul
Alternatives for SparkFun_BMI270_Arduino_Library.h: [SparkFun BMI270 Arduino Library@1.0.3]
ResolveLibrary(SparkFun_BMI270_Arduino_Library.h)
  -> candidates: [SparkFun BMI270 Arduino Library@1.0.3]
C:\Users\Merli\AppData\Local\Arduino15\packages\zephyr\tools\arm-zephyr-eabi\0.16.8/bin/arm-zephyr-eabi-g++ -g -Os -std=c++17 -c -w -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DNRF52840_XXAA -DKERNEL -DK_HEAP_MEM_POOL_SIZE=32768 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\variants\arduino_nano_33_ble_sense -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\libraries\SPI -Id:\Users\Merli\Documents\Arduino\libraries\SparkFun_BMI270_Arduino_Library\src -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/deprecated -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.1.0\cores\arduino/api/depreca [... cut ...]
    Serial.print("\t");

    // Print rotation data
    Serial.print("Rotation in deg/sec");
    Serial.print("\t");
    Serial.print("X: ");
    Serial.print(imu.data.gyroX, 3);
    Serial.print("\t");
    Serial.print("Y: ");
    Serial.print(imu.data.gyroY, 3);
    Serial.print("\t");
    Serial.print("Z: ");
    Serial.println(imu.data.gyroZ, 3);

    // Print 50x per second
    delay(20);
}
```

**Additional context**
Could be stack size - will adjust with increased stack size to see if it works

mjs513, 2025-01-27:
if I try to adjust stack to the following:
```
##CONFIG_SHELL_STACK_SIZE=2048
##CONFIG_HEAP_MEM_POOL_SIZE=16384
##CONFIG_LLEXT_HEAP_SIZE=96
##CONFIG_MAIN_STACK_SIZE=16384

CONFIG_HEAP_MEM_POOL_SIZE=2048
CONFIG_SHELL_STACK_SIZE=16384
CONFIG_MAIN_STACK_SIZE=16384
CONFIG_LLEXT_HEAP_SIZE=128
```
just see gibberish after crash:

[image]

mjs513, 2025-01-28:
Related to question @KurtE raised
https://github.com/arduino/ArduinoCore-zephyr/issues/52#issue-2811616671

KurtE, 2025-01-28:
On the External Wire pins, I tried a couple of devices on SPI object.
Simple Sparkfun QWIIC button, which works. 

And a QWIIC display: https://www.smart-prototyping.com/Zio-Qwiic-OLED-Display-1_5inch-128x128
Using the Adafruit_SSD1327 library, Running their ssd1327_test sketch, changed I2C address to 0x3C

Also changed the start part of setup to paus before doing things:
```
void next(const char *msg = nullptr) {

  if (msg)Serial.println(msg);
  Serial.println("Press any key to continue");
  while (Serial.read() == -1);
  while (Serial.read() != -1);
}

void setup()   {                
  Serial.begin(115200);
  while (! Serial) delay(100);
  Serial.println("SSD1327 OLED test");
  next("Before Wire.begin");
  Wire.begin();
  Wire.setClock(400000);

  next("Before display.begin");
  if ( ! display.begin(0x3C) ) {
     Serial.println("Unable to initialize OLED");
     while (1) yield();
  }
  next("Before clear");
  
  display.clearDisplay();
  display.display();

  // draw many lines
  testdrawline();
  display.display();
  delay(1000);
  display.clearDisplay();
```

Runs fine on MBED.  Fails on zephyr, no output to Serial or Serial1 and I lose Serial...
Investigating.

mjs513, 2025-01-28:
@KurtE - @facchinm 

Verified adding CONFIG_FPU=y to the ble sense config file resolves the issues with using the BMI270 on SPI.  Also verified with the MPU9250 which would not initialize/run in previous test.

closed 2025-01-29 by mjs513