arduino/ArduinoCore-zephyr #57: NANO:  using floating point crashes the processor.

KurtE, 2025-01-28:
We have been running into some SPI and Wire issues.  So while I was testing out some SPI,
I found that I could run a simple SPI sketch which worked.  I also was able to output
to a simple ST7735 display.  So then wondered if maybe an issue with floating point.

So I tried a simple sketch:
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {}

}

uint32_t loop_count = 0;
uint32_t loop_increment = 63;
void loop() {
  Serial.print(loop_count);
  Serial.println();
  loop_count++;
  Serial.print(dl);

  delay(250);
}

```
Runs fine, now add in any floating point
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {}
}

uint32_t loop_count = 0;
uint32_t loop_increment = 63;
void loop() {
  Serial.print(loop_count);
  Serial.print(" ");
  float f = loop_count * 2.0;
  int dl = f;
  Serial.print(" ");
  Serial.print(dl);
  Serial.println();
  loop_count++;

  delay(250);
}
```
The board completely crashes.  Any hint of floating point appears to crash.

Note: first attempt was to simply print out f like: Serial.print(f, 3); 
But wanted to rule out floating point print issue...

mjs513, 2025-01-28:
As we brainstormed earlier in the day added in 
```
CONFIG_FPU=y

```
now your posted sketch is running:
```
47  94
48  96
49  98
50  100
51  102
52  104
53  106
54  108
55  110
```
also seems to have cleared up issues we were having with:
https://github.com/arduino/ArduinoCore-zephyr/issues/56
and
https://github.com/arduino/ArduinoCore-zephyr/issues/53

KurtE, 2025-01-28:
Also Arduino_HS300x example sketch ReadSensorImperial.ino is now running and showing Temp and Humidty.

KurtE, 2025-01-28:
Wonder if we should create new PR for this or add the change to one of our two open ones.

facchinm, 2025-01-29:
Fixed by #50

closed 2025-01-29 by facchinm