arduino/ArduinoCore-zephyr #58: Debugging issues - With the merge of all of the recent changes

KurtE, 2025-01-30:
After we merged all of the most recent Pull requests and the appears to be issues with using the debug monitor or console...

On the GIGA for example, SPI1 is no longer, So I am trying to debug that:

I have USB to serial adapter connected to Serial1 as before.  I am running in Standard mode:
When I run the sketch, the sketch simply dies, and all I see in the monitor is:
`uart:~$
` 
It also does not allow me to enter anything.

Currently I am debugging this by adding printk statements into the code to find out where it is dying.
I know that it is within the STM32 SPI code as now my current debug outputs show:
```
[00:00:00.018,000] <inf> usb_cdc_acm: Device suspended
[00:00:00.243,000] <inf> usb_cdc_acm: Device configured
Before SPI 0x2401396c begin
        After
        Before beginSPITransaction(5000000)
        WC:ef   OutputToSPI(239)        ZephyrSPI::ZephyrSPI before spi_transceive: 0x24013974 0x24035f50 0x24035f60

Enter spi_ll_stm32.c transceive before context_lock     before spi_stm32_configure
        before spi_context_buffers_setup: 8
        After buffers setup
        before LL_SPI_Enable
        After LL_SPI_Enable
        before LL_SPI_StartMasterTransfer
        after LL_SPI_StartMasterTransfer
uart:~$
```
This is with my newer GIGA connected up to my Windows machine...  I tried once yesterday with compiling with mode: Debug and it would not allow me to enter anything like: sketch

If I try running this with my original GIGA board, which is currently connected up to my Ubuntu machine,  I have an additional problem, that for some reason, this GIGA acts like I always build for debug.  It originally respected the setting for debug versus standard, but at some point, it always started to act like in debug and I have to enter: sketch 
to get it to run...  And keyboard input is not working...  And I have burned the loader many different times, loaded many sketches, also loaded sketches built for MBED... But when I reload zephyr bootloader and sketch still have to enter: sketch.
Suggestion?

Side note: I am in the process of making sure I have a clean install on my Ubuntu machine, so I deleted, the: ArduinoCore-zephyr directory, as well as zephyr and modules:

I git cloned (in my case): 
```
git clone https://www.github.com/kurte/ArduinoCore-zephyr
cd ArduinoCore-zephyr
./extra/bootstrap.sh
```
At this point is it OK to simply do the: ./extra/build_all.sh
or do we first need to do:
```
source ./venv/bin/activate
west update
deactivate
```
My guess is yes, as if I look at: zephyr project and do a: git branch
It shows: (might be typos as typing this in from windows machine)
```
* (HEAD detached at refs/heads/manifest-rev)
  manifest-rev
```  
My windows machine, where yesterday I did the west update, it shows:
```
kurte@KurtXPS8950:~/git/zephyr$ git branch
* (HEAD detached at 33bc8a018ec)
  manifest-rev
  plin_4.0.99
kurte@KurtXPS8950:~/git/zephyr$
```

Note: I will now do update again on windows machine to see if maybe something changed overnight.
It did...
```
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/zephyr$ git branch
* arduino
  nano33_overlay
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/zephyr$

Keeping fingers crossed maybe things will just work again

KurtE, 2025-01-30:
Quick update:
Full reinstall and build on Ubuntu and: 
Serial1 input for monitor appears to be working.
Although still have to enter:sketch even though I compiled standard.
SPI1 still crashing.

Current rebuild on windows machine (in WSL) and still not seeing any input from Putty window... Was working before, still looks
like RX/TX pins of connector are going to IO pins 0, 1 and GND... Will debug more later...

mjs513, 2025-01-30:
Just to confirm am seeing the same thing on the nano and giga

KurtE, 2025-02-03:
@mjs513 @facchinm @pillo79 

With the current stuff, I am having a lot of problems trying to debug stuff on the NANO

I have a USB to serial adapter hooked up to TX/RX pins connected to terminal program (Currently using a version
of PuTTy on my Ubuntu machine same on WIndows).

When I boot sometimes I will see in putty window: the uart:~$ type prompt.
And a few usb_cdc_acm: message Not always here.
In the kernel code I am not seeing any of the printk message here nor USB...

Now if I build with mode: Debug,  a prompt shows up in the IDE2 Serial monitor window.  I can type a few commands
like: help,  and the like. 
When I enter: sketch
Then the sketch appears to maybe totally hang.  No output on USB or Serial port.

(Back in mode; normal)
Note: if in my sketch I add printk("I am in my sketch")
This shows up in the Putty window.  Likewise if I added call Serial1.begin(115200);.
Serial1.println("xyz") also shows up in Putty...

Suggestions?

For now I will try to add code to initialize the SPI1 devices and use some other print statements and the like
to see if it worked...

mjs513, 2025-02-04:
@Kurte @facchinm @pillo79 

Know this is redundant but I can confirm this is happening on a Nano 33 sense v2 and and nano 33 BLE

KurtE, 2025-05-16:
Quick update, since I have not tried it in a couple of months.

I built the current stuff for the NANO - I am sync with the latest up ArduinoCore-zephyr - including the updated build.sh

When I build a sketch for mode=standard.
All debug outputs go to the Serial1 object (output to my kitty window)

If I build with mode=debug,  
The debug prompt and output go the the USB serial port.  That is I have to type in sketch in the USB Serial monitor to do
anything. and then my printk's go out to USB Serial.

Simple test I ran was to use blink, but added simple loop counter:
```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
int loop_count;
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  printk("%u\n", loop_count++);
  delay(1000);                      // wait for a second
}
```