arduino/ArduinoCore-zephyr #59: Sketch upload fails on Nano 33 BLE Sense rev2

timberwolf-toao, 2025-01-30:
Wanted to try the Zephyr Core on a Arduino Nano 33 BLE sense rev2.

Sketch upload fails with "SAM-BA operation failed" though bootloader installation went thru without any issues.
Tried with Arduino IDE 2.3.4 on MacOS and Windows 10, issue stays the same.
Even tried Zephyr 0.1.0 and 0.2.0, no difference.
Used sketch doesn't seem to matter, even an stub one with emtpy setup() and loop() functions produces the same error.

After upload failure the port is also gone and can only be recovered by double clicking reset.
If this is somethin obvious it totally eludes me, sorry for that.
Only thing that comes to my mind is the previous usage of this board for OpenMV, so it got flashed with the updated bootloader and softdevice previously.

Any hints?

Upload output:
```
Sketch uses 1556 bytes (0%) of program storage space. Maximum is 786432 bytes.
Global variables use 1364 bytes (0%) of dynamic memory, leaving 522260 bytes for local variables. Maximum is 523624 bytes.
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.001 seconds
Write 6816 bytes to flash (2 pages)

[                              ] 0% (0/2 pages)
SAM-BA operation failed
Failed uploading: uploading error: exit status 1

```
Bootloader installation output:
```
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.000 seconds
Write 338164 bytes to flash (83 pages)

[                              ] 0% (0/83 pages)
[=                             ] 3% (3/83 pages)
[=                             ] 4% (4/83 pages)
[=                             ] 6% (5/83 pages)
[==                            ] 7% (6/83 pages)
[==                            ] 8% (7/83 pages)
[==                            ] 9% (8/83 pages)
[===                           ] 10% (9/83 pages)
[===                           ] 12% (10/83 pages)
[===                           ] 13% (11/83 pages)
[====                          ] 14% (12/83 pages)
[====                          ] 15% (13/83 pages)
[=====                         ] 16% (14/83 pages)
[=====                         ] 18% (15/83 pages)
[=====                         ] 19% (16/83 pages)
[======                        ] 20% (17/83 pages)
[======                        ] 21% (18/83 pages)
[======                        ] 22% (19/83 pages)
[=======                       ] 24% (20/83 pages)
[=======                       ] 25% (21/83 pages)
[=======                       ] 26% (22/83 pages)
[========                      ] 27% (23/83 pages)
[========                      ] 28% (24/83 pages)
[=========                     ] 30% (25/83 pages)
[=========                     ] 31% (26/83 pages)
[=========                     ] 32% (27/83 pages)
[==========                    ] 33% (28/83 pages)
[==========                    ] 34% (29/83 pages)
[==========                    ] 36% (30/83 pages)
[===========                   ] 37% (31/83 pages)
[===========                   ] 38% (32/83 pages)
[===========                   ] 39% (33/83 pages)
[============                  ] 40% (34/83 pages)
[============                  ] 42% (35/83 pages)
[=============                 ] 43% (36/83 pages)
[=============                 ] 44% (37/83 pages)
[=============                 ] 45% (38/83 pages)
[==============                ] 46% (39/83 pages)
[==============                ] 48% (40/83 pages)
[==============                ] 49% (41/83 pages)
[===============               ] 50% (42/83 pages)
[===============               ] 51% (43/83 pages)
[===============               ] 53% (44/83 pages)
[================              ] 54% (45/83 pages)
[================              ] 55% (46/83 pages)
[================              ] 56% (47/83 pages)
[=================             ] 57% (48/83 pages)
[=================             ] 59% (49/83 pages)
[==================            ] 60% (50/83 pages)
[==================            ] 61% (51/83 pages)
[==================            ] 62% (52/83 pages)
[===================           ] 63% (53/83 pages)
[===================           ] 65% (54/83 pages)
[===================           ] 66% (55/83 pages)
[====================          ] 67% (56/83 pages)
[====================          ] 68% (57/83 pages)
[====================          ] 69% (58/83 pages)
[=====================         ] 71% (59/83 pages)
[=====================         ] 72% (60/83 pages)
[======================        ] 73% (61/83 pages)
[======================        ] 74% (62/83 pages)
[======================        ] 75% (63/83 pages)
[=======================       ] 77% (64/83 pages)
[=======================       ] 78% (65/83 pages)
[=======================       ] 79% (66/83 pages)
[========================      ] 80% (67/83 pages)
[========================      ] 81% (68/83 pages)
[========================      ] 83% (69/83 pages)
[=========================     ] 84% (70/83 pages)
[=========================     ] 85% (71/83 pages)
[==========================    ] 86% (72/83 pages)
[==========================    ] 87% (73/83 pages)
[==========================    ] 89% (74/83 pages)
[===========================   ] 90% (75/83 pages)
[===========================   ] 91% (76/83 pages)
[===========================   ] 92% (77/83 pages)
[============================  ] 93% (78/83 pages)
[============================  ] 95% (79/83 pages)
[============================  ] 96% (80/83 pages)
[============================= ] 97% (81/83 pages)
[============================= ] 98% (82/83 pages)
[==============================] 100% (83/83 pages)
Done in 13.763 seconds

```

KurtE, 2025-01-30:
Try double click again on NANO.
Then check to see what Serial port exists: Tools->Port and choose it (changes between double click and normal mode... at least for me).

Now try to upload a simple sketch like the example blink...

timberwolf-toao, 2025-01-30:
The Nano only provides a serial port in recovery mode (double click, breathing LED), no other devices providing any ports are on the whole system.
With that port the upload of a sketch fails but bootloader installation works.
After the upload error of a sketch or a normal reset, it doesn't provide any port or other USB device as far as I could recognize.
Uploading the bootloader even repeatedly always is succesful.

KurtE, 2025-01-30:
After you programmed the bootloader, did you do as I mentioned, to double click on the NANO
and make sure you have then the that port selected and then try to download a sketch?
Again, suggest real simple sketch like blink:

Note: this is what the Readme file says:

>🏗️ First Use
To get started with your board:
>
>Put the board in bootloader mode by double-clicking the RESET button.
Run the Burn Bootloader option from the IDE/CLI.
Note that due to limitations in the Arduino IDE, you may need to select any programmer from the Programmers menu.
Once the bootloader is installed, you can load your first sketch by placing the board into bootloader mode again.
Note
>
>After the initial setup, future sketches will be loaded automatically without needing to reset the board.

Note: that last part, is somewhat true.  If your sketch crashes, it will very likely take down the USB communications, and as such
you will need to do the double click again.

Also if you install a different version of the zephyr install, Or rebuild from sources... You will again need to reburn the boot loader.

timberwolf-toao, 2025-01-30:
Short answer: Yes, it is definetly the right port.
Long answer: USB VID&PID match, no other port on the system and it doesn't upload a single page of sketch code, so pretty irrelevant what sketch I actually use.

Without digging further though the code of this repo I'm pretty sure for now that the issue lies within the new bootloader I have on that nano, see [here](https://docs.arduino.cc/tutorials/nano-33-ble-sense-rev2/getting-started-omv/). I guess the code in this repo doesn't account for the changed adresses and/or the new bootloader doesn't want to write to the adresses the code in this repo is instructing it to do.

timberwolf-toao, 2025-01-31:
Yes, that was it. The "new"bootloader for OpenMV & Softdevice usage isn't compatible with the Zephyr Core in the current stage.
Reverted to the old bootloader via mbed package and SWD.

Maybe worth a short note in the Readme?