arduino/ArduinoCore-zephyr #60: Nano 33 BLE:  micros() function does not work

mjs513, 2025-01-30:
While playing with sdfat library found sketch was hanging on test for micro().  Running the following simple test sketch shows micros() always returns 0:
```
unsigned long time; 

void setup() { 
   Serial.begin(9600); 
} 

void loop() { 
   Serial.print("Time:");
   time = micros(); //prints time since program started
   Serial.println(time); // wait a second so as not to send massive amounts of data
   delay(1000); 
}

RETURNS:
Time:0
Time:0
Time:0
Time:0
Time:0
Time:0
```
will look more later but off to do other things.  Think @facchinm did a pr on the giga for time functions.

KurtE, 2025-01-30:
Quick update, which shows the issue:
Here is a slightly modified version of the test program I had for testing millis/micros on giga:
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {}
  Serial.println("\nTest millis and Micros");
  Serial.println(micros());

}

void loop() {
  unsigned long start_us = micros();
  unsigned long start_ms = millis();
  delay(1000); 

  unsigned long end_us = micros();
  unsigned long end_ms = millis();

  unsigned long delta_us = end_us - start_us;
  unsigned long delta_ms = end_ms - start_ms;

  Serial.print("US: ");
  Serial.print(start_us);
  Serial.print(" ");
  Serial.print(end_us);
  Serial.print(" ");
  Serial.print(delta_us);
  Serial.print(" KC32: ");
  Serial.print(k_cycle_get_32());
  Serial.print(" KC64: ");
  Serial.print(k_cycle_get_64());
  

  Serial.print("\tMS: ");
  Serial.print(start_ms);
  Serial.print(" ");
  Serial.print(end_ms);
  Serial.print(" ");
  Serial.println(delta_ms);

  if (Serial.available()) {
    Serial.println("Paused");
    while (Serial.read() != -1) {}
    while (Serial.read() == -1) {}
    while (Serial.read() != -1) {}
  }
}
```
Here is the output:
```
US: 0 0 0 KC32: 3052268 KC64: 0	MS: 92146 93146 1000
US: 0 0 0 KC32: 3085163 KC64: 0	MS: 93149 94149 1000
US: 0 0 0 KC32: 3118029 KC64: 0	MS: 94152 95152 1000
US: 0 0 0 KC32: 3150942 KC64: 0	MS: 95157 96157 1000
US: 0 0 0 KC32: 3183851 KC64: 0	MS: 96161 97161 1000
US: 0 0 0 KC32: 3216773 KC64: 0	MS: 97166 98166 1000
US: 0 0 0 KC32: 3249646 KC64: 0	MS: 98170 99170 1000
US: 0 0 0 KC32: 3282576 KC64: 0	MS: 99174 100174 1000
US: 0 0 0 KC32: 3315481 KC64: 0	MS: 100178 101178 1000
US: 0 0 0 KC32: 3348402 KC64: 0	MS: 101183 102183 1000
US: 0 0 0 KC32: 3381322 KC64: 0	MS: 102187 103187 1000
US: 0 0 0 KC32: 3414203 KC64: 0	MS: 103191 104192 1001
US: 0 0 0 KC32: 3447108 KC64: 0	MS: 104195 105195 1000
US: 0 0 0 KC32: 3480012 KC64: 0	MS: 105199 106199 1000
US: 0 0 0 KC32: 3512936 KC64: 0	MS: 106204 107204 1000
US: 0 0 0 KC32: 3545865 KC64: 0	MS: 107209 108209 1000
```

KurtE, 2025-01-30:
Quick update:
If we look at the kernel function:
```
static inline uint64_t k_cycle_get_64(void)
{
	if (!IS_ENABLED(CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER)) {
		__ASSERT(0, "64-bit cycle counter not enabled on this platform. "
			    "See CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER");
		return 0;
	}

	return arch_k_cycle_get_64();
}
```
And searching for who has this set:

Searching 54762 files for "#define CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER 1"

```
D:\github\ArduinoCore-zephyr\variants\arduino_giga_r1_m7\llext-edk\include\zephyr\include\generated\zephyr\autoconf.h:
  381  #define CONFIG_SPI_STM32 1
  382  #define CONFIG_SPI_STM32_USE_HW_SS 1
  383: #define CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER 1
  384  #define CONFIG_SYSTEM_CLOCK_INIT_PRIORITY 0
  385  #define CONFIG_TICKLESS_CAPABLE 1

D:\github\ArduinoCore-zephyr\variants\arduino_portenta_h7\llext-edk\include\zephyr\include\generated\zephyr\autoconf.h:
  446  #define CONFIG_SPI_LOG_LEVEL_DEFAULT 1
  447  #define CONFIG_SPI_LOG_LEVEL 3
  448: #define CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER 1
  449  #define CONFIG_SYSTEM_CLOCK_INIT_PRIORITY 0
  450  #define CONFIG_TICKLESS_CAPABLE 1

D:\github\ArduinoCore-zephyr\variants\ek_ra8d1\llext-edk\include\zephyr\include\generated\zephyr\autoconf.h:
  290  #define CONFIG_SPI_B_RA_DTC 1
  291  #define CONFIG_SPI_B_USE_HW_SS 1
  292: #define CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER 1
  293  #define CONFIG_SYSTEM_CLOCK_INIT_PRIORITY 0
  294  #define CONFIG_TICKLESS_CAPABLE 1

D:\github\ArduinoCore-zephyr\variants\frdm_mcxn947_mcxn947_cpu0\llext-edk\include\zephyr\include\generated\zephyr\autoconf.h:
  365  #define CONFIG_SPI_LOG_LEVEL 3
  366  #define CONFIG_SPI_MCUX_LPSPI 1
  367: #define CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER 1
  368  #define CONFIG_SYSTEM_CLOCK_INIT_PRIORITY 0
  369  #define CONFIG_TICKLESS_CAPABLE 1

4 matches across 4 files
```
So Giga/Portenta have it and NANO does not...

So I see a couple ways to fix this:  
One way is to break the rollover on boards that don't have 64 bit timer...

The other is, to see if we have some form of rollover interrupt on the 32 bit timer, that we can have something increment another
32 bit variable, and you then get both the values build it into 64 bits...

PR #62 (2025-03-05): Fix micros() for nano without breaking giga
Fixes #60 

The rollover fix for the giga used the 64 bit timer instead of 32 bits.  The nano does not have a 64 bit timer, so you need to use the 32 bit instead.

So I know check to see if the processor supports 64 bits if so use it else fall back to 32 bit.

@mjs513 you might want to try this out to see if it fixes the issue for you
closed 2025-03-05 by facchinm