arduino/ArduinoCore-zephyr #80: Support Portenta C33

facchinm, 2025-03-04:
WIP branch: https://github.com/arduino/ArduinoCore-zephyr/tree/fully_linked
Stopping issues: https://github.com/zephyrproject-rtos/mbedtls/pull/68

KurtE, 2025-03-04:
@facchinm - Should we be trying this branch out?
Also have you looked at some of the current PRs, to maybe pull into this branch?

thanks
Kurt

KurtE, 2025-03-04:
I know I am living on the wild side... But on my secondary machine, I did clone this this branch down and also did clean install of zephyr and modules. 

I tried the build_all.sh it failed on the build: ./extra/build.sh ek_ra8d1 ek_ra8d1
Also on the C33 as this issue mentions.

I tried a compile a couple of simple sketches for GIGA/Portenta R7...
And had compiler errors.  For example blink on Portenta:
```
10124 |         fn(DT_N_S_zephyr_user, digital_pin_gpios, 1, __VA_ARGS__) DT_DEBRACKET_INTERNAL sep \
      |         ^~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:5273:33: note: in expansion of macro 'DT_N_S_zephyr_user_P_digital_pin_gpios_FOREACH_PROP_ELEM_SEP_VARGS'
 5273 | #define DT_CAT4(a1, a2, a3, a4) a1 ## a2 ## a3 ## a4
      |                                 ^~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:3410:9: note: in expansion of macro 'DT_CAT4'
 3410 |         DT_CAT4(node_id, _P_, prop, _FOREACH_PROP_ELEM_SEP_VARGS)(      \
      |         ^~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/cores/arduino/Arduino.h:48:9: note: in expansion of macro 'DT_FOREACH_PROP_ELEM_SEP_VARGS'
   48 |         DT_FOREACH_PROP_ELEM_SEP_VARGS(DT_PATH(zephyr_user), digital_pin_gpios,                    \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/sys/util_internal.h:104:26: note: in expansion of macro 'UTIL_PRIMITIVE_CAT'
  104 | #define UTIL_CAT(a, ...) UTIL_PRIMITIVE_CAT(a, __VA_ARGS__)
      |                          ^~~~~~~~~~~~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:5247:9: note: in expansion of macro 'UTIL_CAT'
 5247 |         UTIL_CAT(DT_ROOT, MACRO_MAP_CAT(DT_S_PREFIX, __VA_ARGS__))
      |         ^~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:5247:18: note: in expansion of macro 'DT_ROOT'
 5247 |         UTIL_CAT(DT_ROOT, MACRO_MAP_CAT(DT_S_PREFIX, __VA_ARGS__))
      |                  ^~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:140:22: note: in expansion of macro 'DT_PATH_INTERNAL'
  140 | #define DT_PATH(...) DT_PATH_INTERNAL(__VA_ARGS__)
      |                      ^~~~~~~~~~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/cores/arduino/Arduino.h:48:40: note: in expansion of macro 'DT_PATH'
   48 |         DT_FOREACH_PROP_ELEM_SEP_VARGS(DT_PATH(zephyr_user), digital_pin_gpios,                    \
      |                                        ^~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/cores/arduino/Arduino.h:61:9: note: in expansion of macro 'DIGITAL_PIN_GPIOS_FIND_PIN'
   61 |         DIGITAL_PIN_GPIOS_FIND_PIN(                                                              \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved202524-31905-a9dw7t.359ss/Blink/Blink.ino:35:16: note: in expansion of macro 'LED_BUILTIN'
   35 |   digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
      |                ^~~~~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/generated/zephyr/devicetree_generated.h:2155:95: error: pasting "/* 0x58022400 */" and "U" does not give a valid preprocessing token
 2155 | #define DT_N_S_soc_S_pin_controller_58020000_S_gpio_58022400_REG_IDX_0_VAL_ADDRESS 1476535296 /* 0x58022400 */
      |                                                                                               ^~~~~~~~~~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h:102:23: note: in definition of macro '__UINT32_C'
  102 | #define __UINT32_C(c) c ## U
      |                       ^
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:2438:9: note: in expansion of macro 'DT_U32_C'
 2438 |         DT_U32_C(DT_REG_ADDR_BY_IDX_RAW(node_id, idx))
      |         ^~~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:5273:33: note: in expansion of macro 'DT_N_S_soc_S_pin_controller_58020000_S_gpio_58022400_REG_IDX_0_VAL_ADDRESS'
 5273 | #define DT_CAT4(a1, a2, a3, a4) a1 ## a2 ## a3 ## a4
      |                                 ^~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:2415:9: note: in expansion of macro 'DT_CAT4'
 2415 |         DT_CAT4(node_id, _REG_IDX_, idx, _VAL_ADDRESS)
      |         ^~~~~~~
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/devicetree.h:2438:18: note: in expansion of macro 'DT_REG_ADDR_BY_IDX_RAW'
 2438 |         DT_U32_C(DT_REG_ADDR_BY_IDX_RAW(node_id, idx))
[... cut ...]
/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/cores/arduino/Arduino.h:61:9: note: in expansion of macro 'DIGITAL_PIN_GPIOS_FIND_PIN'
   61 |         DIGITAL_PIN_GPIOS_FIND_PIN(                                                              \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved202524-31905-a9dw7t.359ss/Blink/Blink.ino:35:16: note: in expansion of macro 'LED_BUILTIN'
   35 |   digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
      |                ^~~~~~~~~~~
exit status 1

Compilation error: exit status 1
```

KurtE, 2025-03-05:
@facchinm @pillo79 - Sorry I am sure you already know this and probably even know the fix to the above:

But it appears like, changes to the file: devicetree_generated.h is causing these problems:

There are lines like:
```
#define DT_N_S_cpus_S_cpu_0_S_mpu_e000ed90_P_reg {3758157200 /* 0xe000ed90 */, 64 /* 0x40 */}
```
Which are causing major problems for a lot of the macros that do things like: DT_FOREACH_PROP_ELEM_SEP_VARGS

Before (i.e. current release this line looked like:
```
#define DT_N_S_cpus_S_cpu_0_S_mpu_e000ed90_P_reg {3758157200, 64}
```

facchinm, 2025-03-05:
@KurtE wow, that was fast 🙂 We are cleaning up the repo a bit to move main to this branch, but before that I'm going to merge some of your PRs 😉

pillo79, 2025-03-12:
Regarding the [macro comments build issue above](https://github.com/arduino/ArduinoCore-zephyr/issues/80#issuecomment-2699193343) - after lots of experimenting, I think I have identified the issue and fixed it in 1eade1086f73a61d215bf989d00d4e68c67edf12. We will drop the Zephyr change that removes comments in the next rebase so it's one less to think about :-)

KurtE, 2025-03-12:
@pillo79 - Thanks, 
Let us know when you have something for us to play with :) 
And will the Arduino zepythr be updated to use the released version of zephyr? Or use the Zephyr release directly or maybe pull in some of the prs that were pulled in after it was released, like the activate/deactivate...

Thanks again

KurtE, 2025-03-14:
@mjs513 @pillo79 @facchinm and all,

Yesterday (and this morning) I thought I would do a quick test to see if I could simply say to use the released 4.1...So I updated
the west.yml file, to point at it:
```
  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v4.1.0
      import:
        name-allowlist:
```
I tried doing a: ./extra/build.sh
And got a bunch of errors.

So I looked at what was different on the Arduino zephyr project:
https://github.com/zephyrproject-rtos/zephyr/compare/main...arduino:zephyr:arduino_core_merge_4.1.0-rc2
And I did the first two updates locally:

[image]

Those allowed most of the build for giga and H7 to complete, however I believe I have a mismatch now of which SDK version should be used:
```
[454/454] Linking CXX executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      281888 B       384 KB     71.69%
             RAM:      387316 B       512 KB     73.87%
          EXTMEM:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:         16 KB        32 KB     50.00%
           SRAM4:          0 GB        64 KB      0.00%
          SDRAM1:          0 GB         8 MB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/kurte/github/ArduinoCore-zephyr/build/zephyr/zephyr.elf for board: arduino_portenta_h7
-- west build: running target llext-edk
[1/1] Generating llext-edk.tar.xz
-- Found Python3: /home/kurte/github/ArduinoCore-zephyr/venv/bin/python (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/kurte/.cache/zephyr
Exporting provides.ld
./extra/build.sh: line 36: /ssd/zephyr-sdk-0.16.8//arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf: No such file or directory
./extra/build.sh: line 37: /ssd/zephyr-sdk-0.16.8//arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf: No such file or directory
./extra/build.sh: line 38: /ssd/zephyr-sdk-0.16.8//arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf: No such file or directory
./extra/build.sh: line 39: /ssd/zephyr-sdk-0.16.8//arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf: No such file or directory
./extra/build.sh: line 40: /ssd/zephyr-sdk-0.16.8//arm-zephyr-eabi/bin/arm-zephyr-eabi-readelf: No such file or directory
cat: loader/boards/arduino_portenta_h7.overlay: No such file or directory
kurte@kurte-750-435st:~/github/ArduinoCore-zephyr$ 
```
I should probably double check I am not seeing these messages with the other builds.

Or maybe this is covered in one of the other updates made in the rc2 merge branch.

KurtE, 2025-03-15:
Quick update: The error shown above was because I lost where I exported the location of the zephyr SDK.... 
Added it back.  Note: I have 0.17,0

I then tried pulling in all of the changes that were made on the beta2 branch.  
I did this, by creating a new branch from the 4.1.0 release.  I then went into github desktop, and went back to the 
[arduino_core_merge_4.1.0-rc2](https://github.com/arduino/zephyr/tree/arduino_core_merge_4.1.0-rc2) branch
and choose all of the changes that looked like they were done locally, and then choose the cherry pick command and then
applied them to my new branch, which I pushed up: https://github.com/KurtE/zephyr/tree/arduino_core_merge_4.1.0-kje

The build went through up to the C33 build and failed due to the: #define BV problem.

I was then able to build simple sketch like blink for GIGA or portenta H7, but both fail to run:
```
uart:~$ sketch
Invalid sketch header
uart:~$

```
on the off chance it is the SDK version that is an issue, I will try with the older one.

Other than that, Not sure what else to try.

KurtE, 2025-03-15:
Another update, It is now working on My Ubuntu machine... Have not tried much yet, but will.
Problem was the post build loader app was not in sync with the loader\main.c updates.  I was able to build the
app, by installing go, and then cd to the directory in extra... and run the build.sh.  Then copy the build app to the right spot in your arduino15 directory...

Currently seeing, if I can now reproduce this on my WIndows machine.  Building under WSL, and then copy the stuff up to
Windows directories.  The post build runs now on windows.

Rough instructions on what I am trying now on Windows machine:
In WSL:
```
(Create new root directory and CD to it)
 git clone https://www.github.com/kurte/ArduinoCore-zephyr 
cd ArduinoCore-zephyr/
git remote add upstream https://www.github.com/arduino/ArduinoCore-zephyr 
 git fetch --all 
 git checkout fully_linked
./extra/bootstrap.sh
cd extra/post_build_tool/
./build.sh  (Assumes you have go installed)

cd (your rootdir)/zephyr
git remote add kurte https://github.com/kurte/zephyr
git fetch --all
git checkout arduino_core_merge_4.1.0-kje

cd (rootdir)/ArduinoCore-zephyr
./extra/build_all.sh

```
The build is in process...
It did error out on the: ./extra/build.sh ek_ra8d1 ek_ra8d1
Will look at that later.

Next up will copy the build plus the tools and see if Arduino is at all happy.

Will update this once I know more

KurtE, 2025-03-16:
Sorry, I know that I am jumping the gun here, but thought it would be interesting to try to integrate up to 
the current release of Zephyr.  Mainly curious to see if any of the updated camera objects are working better or not.

Not sure if better to update previous comment or create a new one.

**Update:**
First on my Ubuntu machine - the GIGA does upload new sketch and runs... After I entered sketch in the
debug window... Did a modified version of blink... 

The debug monitor also shows the normal: usb_cdc_acm: Device suspended ... and then configured.

On the Ubuntu machine, when I try the same thing on the Portenta H7, 
The debug monitor outputs different stuff like:
```
*** Booting Zephyr OS build v3.7.0-12117-ge968bc26beb4 ***
uart:~$
```
It looks like the sketch is started, However, there is no USB Serial port created: /dev/ttyACMx 

On my Windows machine, I was playing around with the Portenta branch, trying to integrate our changes for the current PRs, It builds, but it does not run.

Different error message in Debug window:
```
uart:~$ sketch
[00:00:23.863,000] <err> llext: Undefined symbol with no entry in symbol table __device_dts_ord_102, offset 1012, link section 11
[00:00:23.875,000] <err> llext: Failed to link, ret -61
Failed to load sketch, rc -61
uart:~$
```

One thing I noticed on my Window's machine building the loader, at the end of the build is a different message:
```
[485/485] Linking CXX executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      318596 B       384 KB     81.02%
             RAM:      358260 B       512 KB     68.33%
          EXTMEM:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:         16 KB        32 KB     50.00%
           SRAM4:          0 GB        64 KB      0.00%
          SDRAM1:          4 MB         8 MB     50.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/kurte/zephyr_2.1/ArduinoCore-zephyr/build/zephyr/zephyr.elf for board: arduino_portenta_h7
-- west build: running target llext-edk
[1/1] Generating llext-edk.tar.xz
-- Found Python3: /home/kurte/zephyr_2.1/ArduinoCore-zephyr/venv/bin/python (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/kurte/.cache/zephyr
Exporting provides.ld
cat: loader/boards/arduino_portenta_h7.overlay: No such file or directory
kurte@KurtXPS8950:~/zephyr_2.1/ArduinoCore-zephyr$
```
Looks like while generating the provides.ld file, which I believe is used in the new binding stuff... it can not find the
overlay file.  Note: the overlay file in this case is:
`-- Found devicetree overlay: /home/kurte/zephyr_2.1/ArduinoCore-zephyr/loader/boards/arduino_portenta_h7_m7.overlay
`
(I then went and verified that I was seeing the same message on the Ubuntu build)

I tried renaming this to what the build.sh file was looking for:
`cat build/zephyr/zephyr.map | grep __device_dts_ord | grep -v rodata | grep -v llext_const_symbol |  awk -F' ' '{print "PROVIDE("$2" = "$1");"}'  >> variants/$variant/provides.ld
`
But then the main build code fails.  Tried changing the build command to be the _h7_m7
`./extra/build.sh arduino_portenta_h7//m7 arduino_portenta_h7_m7
`
But then it does not find the variant...   

Last attempt - I copied the overlay file to what the cat was looking for. 
Build went through.  I flashed the loader, I then reloaded the program and now it looks like it might be running.
However again there is no USB Serial port. 
back to same messages  as the Ubuntu build.
```
*** Booting Zephyr OS build v3.7.0-12117-ge968bc26beb4 ***
uart:~$ sketch
Failed to load sketch, rc 1
uart:~$

```
Will check to see if I the GIGA is working sufficiently to try out the different Cameras... Need to resetup the GIGA on my Dev machine with debug setup as it still appears to be always in debug mode. (Need to enter sketch in debug window, to run)
Side note:
On GIGA I see the menu item was renamed: from mode to Debug.  

Side Note sure what the difference is between: Debug "Debug" and Debug "Standard"

Now back to playing.

mjs513, 2025-03-17:
@KurtE 

I synched up with your repor/branch using the fully linked plus prs branch and did an build_all.  Seems to be failing on the Portenta C33:
```
/home/my_new_zephyr_folder/modules/crypto/mbedtls/library/x509_crt.c:1750:25: error: expected identifier or '(' before string constant
 1750 | #define BC              "18"
      |                         ^~~~
/home/my_new_zephyr_folder/modules/hal/renesas/drivers/ra/fsp/src/bsp/cmsis/Device/RENESAS/Include/R7FA6M5BH.h:9225:27: note: in expansion of macro 'BC'
 9225 |             __IOM uint8_t BC   : 3;    /*!< [2..0] Bit Counter
                  */
      |                           ^~
[578/616] Building C object modules/mbedtls/CMakeFil..._folder/modules/crypto/mbedtls/library/ssl_tls.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/my_new_zephyr_folder/ArduinoCore-zephyr/build
```
all other boards seem to be building so will test with the giga and h7 lite to see what we get.

mjs513, 2025-03-18:
@KurtE - @pillo79 - @facchinm 

Ok redid the whole thing using @KurtE's fully_linked repo.  Using sdk-17.0 and all changes identified in earlier posts. 
1.  Copy and pasted arduino_portenta_h7_m7.overlay to arduino_portenta_h7.overlay to clear up the cat error.
2.  Now using sdk-17.0 to clear up the build errors on not finding readelf in zephry-16.8
3.  Rebuilt post_build_tool using go.

Now for GIGA
1.  Sketches load and run in standard mode without losing Serial.
2.  In DEBUG mode typing sketch in Serial monitor does nothing, ie. sketch does not run

For H7
1.  After uploading firmware - it loses serial monitor so bigh issue.  Think @KurtE mentioned still will print to Serial1
2.  Debug same think no serial monitor.

For Nano 33 ble sense
1.  Serial works as before no issues
2.  When running WireScanner noticed that you change board names 
`ARDUINO_NANO_BLE` changed to `ARDUINO_ARDUINO_NANO33BLE`
not sure if that is intentional or not
3.  Same thing for the GIGA
`arduino_giga_r1` went to `arduino_giga`

KurtE, 2025-03-23:
@facchinm @pillo79 (@mjs513)

Not sure if comments should go here?  Or new issue or issue up on zephyr...

As you know, I am playing with the PR #85 have some fixes for H8, decided to play with C33.  Some of it appears to work.
Thought it might be fun to flesh out some more of the stuff, like maybe pin compatibility with MBED version/C33 documentation, 
So updated the pin table:

```
	zephyr,user {
		digital-pin-gpios = <&ioport1 5 GPIO_ACTIVE_HIGH>,
				    <&ioport1 6 GPIO_ACTIVE_HIGH>,
				    <&ioport1 11 GPIO_ACTIVE_HIGH>,
				    <&ioport3 3 GPIO_ACTIVE_HIGH>,
				    <&ioport4 1 GPIO_ACTIVE_HIGH>,
				    <&ioport2 10 GPIO_ACTIVE_HIGH>,
				    <&ioport6 1 GPIO_ACTIVE_HIGH>,
				    <&ioport4 2 GPIO_ACTIVE_HIGH>,
				    <&ioport9 0 GPIO_ACTIVE_HIGH>,
				    <&ioport2 4 GPIO_ACTIVE_HIGH>,
				    <&ioport3 15 GPIO_ACTIVE_HIGH>,
				    <&ioport4 7 GPIO_ACTIVE_HIGH>,
				    <&ioport4 8 GPIO_ACTIVE_HIGH>,
				    <&ioport1 10 GPIO_ACTIVE_HIGH>,
				    <&ioport6 2 GPIO_ACTIVE_HIGH>,

				    <&ioport0 6 GPIO_ACTIVE_HIGH>,
				    <&ioport0 5 GPIO_ACTIVE_HIGH>,
				    <&ioport0 4 GPIO_ACTIVE_HIGH>,
				    <&ioport0 2 GPIO_ACTIVE_HIGH>,
				    <&ioport0 1 GPIO_ACTIVE_HIGH>,
				    <&ioport0 15 GPIO_ACTIVE_HIGH>,
				    <&ioport0 14 GPIO_ACTIVE_HIGH>,
				    <&ioport0 0 GPIO_ACTIVE_HIGH>,

				    <&ioport6 5 GPIO_ACTIVE_HIGH>,
				    <&ioport6 8 GPIO_ACTIVE_HIGH>,
				    <&ioport3 11 GPIO_ACTIVE_HIGH>,
				    <&ioport6 0 GPIO_ACTIVE_HIGH>,

				    <&ioport0 9 GPIO_ACTIVE_HIGH>,
				    <&ioport4 9 GPIO_ACTIVE_HIGH>,
				    <&ioport5 5 GPIO_ACTIVE_HIGH>,
				    <&ioport7 6 GPIO_ACTIVE_HIGH>,
				    <&ioport7 7 GPIO_ACTIVE_HIGH>,
				    <&ioport7 8 GPIO_ACTIVE_HIGH>,
				    <&ioport8 2 GPIO_ACTIVE_HIGH>,

					<&ioport1 7 GPIO_ACTIVE_HIGH>,	// LEDR
					<&ioport4 0 GPIO_ACTIVE_HIGH>,
					<&ioport8 0 GPIO_ACTIVE_HIGH>,

  					// I2C pins
  					<&ioport5 11 GPIO_ACTIVE_HIGH>, /*   D37    |     SDA1     */
  					<&ioport5 12 GPIO_ACTIVE_HIGH>, /*   D38    |     SCL1     */
  					<&ioport3 2 GPIO_ACTIVE_HIGH>, /*   D39    |     SDA2    */
  					<&ioport3 1 GPIO_ACTIVE_HIGH>, /*   D40    |     SCL2    */

  					// CAN pins
  					<&ioport2 2 GPIO_ACTIVE_HIGH>, /*   D41    |   CAN RX    */
  					<&ioport2 3 GPIO_ACTIVE_HIGH>, /*   D42    |   CAN TX    */
  					<&ioport6 10 GPIO_ACTIVE_HIGH>, /*   D43    |   CAN1 RX   */
  					<&ioport6 9 GPIO_ACTIVE_HIGH>, /*   D44    |   CAN1 TX   */

  					// SPI pins
  					<&ioport1 0 GPIO_ACTIVE_HIGH>, /*   D45    |   MISO1  */
  					<&ioport1 1 GPIO_ACTIVE_HIGH>, /*   D46    |   MOSI1  */
  					<&ioport1 2 GPIO_ACTIVE_HIGH>, /*   D47    |   SCLK1  */
  					<&ioport1 3 GPIO_ACTIVE_HIGH>, /*   D48    |   CS1    */

[... 99 lines cut ...]
```
	sci5_default: sci5_default {
		group1 {
			/* tx rx */
			psels = <RA_PSEL(RA_PSEL_SCI_5, 8, 5)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 13)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 8)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 0)>;
		};
	};

	sci6_default: sci6_default {
		group1 {
			/* tx rx */
			psels = <RA_PSEL(RA_PSEL_SCI_5, 8, 5)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 13)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 8)>,
				<RA_PSEL(RA_PSEL_SCI_5, 5, 0)>;
		};
	};
```
Notice the psels of these two objects are the same. will play some, but I believe that 6 object was simply copied from 5 
and needs to have these pins edited, to probably match:

[image]

mjs513, 2025-03-23:
Synched up with @KurtE's branch with his PR changes and testing on H7
1.  Using his branch threads seem to be working:  thread_create, thread_define, mutexes as well as semaphores.
2.  Obviously Serial is working with his change to arduino_portenta_h7_stm32h747xx_m7.dts
3.  I2C is working as well - Ran Scanner.
4.  neither the GC2145 nor the OV7670 camera is working on the H7 anymore - have to investigate more since they were before.  Keep getting cant set video format?

facchinm, 2025-03-28:
C33 merged and available in released 0.3.0, will tackle @KurtE 's report and wifi support (based on https://github.com/zephyrproject-rtos/zephyr/pull/87685) in the next few days.
Thanks everyone for the great support and patience 🙂

KurtE, 2025-03-30:
@facchinm @pillo79 and ...

I pushed up new branch on my fork with C32 pin table stuff as well as a Draft PR.  So far not getting any of the additional SCI based Serial ports working, they hang in open...  Investigating. 

Would like to get an SPI working to the external pins, but not sure how doable that is now?  The one that is implemented (have not tried to use yet, connects to ESP32, I believe. 

I am not seeing any implementation of SPI using SCI in the lower levels, (Zephyr or Modules).  I believe the chip supports two hardware ports. 

With my excel document, I thought about adding the information from The RA6M5 User's Manual (about page 546...) that shows the
different functions for each of the IO pins.  But was having difficulties with this document (excel did not want to read some of these tables.   So I wrote a quick and dirty sketch that reads the pin information from the MBED version and outputs information, which I then added to my Excel document (https://github.com/KurtE/Arduino_GIGA-stuff/blob/main/Documents/Arduino_GIGA_R1_pins.xlsx)

It's not perfect, but, It helps me at least get ideas... Ths page starts off like:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/kurte/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/kurte/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Pin Number | port/pin | HD Connect | Function | HAT Pin | Usages | MID Carrier | Index Table[0] | Index Table[1] | Index Table[2] | Index Table[3] | Index Table[4] | Index Table[5]
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
D00 | P105 | J2:59 |   | RPI07 |   | J15:25, CAM:8 | PWM[1] | INTERRUPT[0] | SCI:TX_MOSI_SDA[8] |   |   |  
D01 | P106 | J2:61 |   | RPI22 |   | J15:27 | PWM[8] | SCI:SCK[8] |   |   |   |  
D02 | P111 | J2:63 |   | RPI26 |   | J15:29 | PWM[3] | INTERRUPT[4] | SCI:SCK[2] | SCI:SCK[9] | SCK[0] |  
D03 | P303 | J2:65 |   | RPI31 |   | J15:31 | PWM[7] | SCI:CTS_RTS_SS[9] |   |   |   |  
D04 | P401 | J2:67 |   | RPI33 |   | J15:33 | CAN_TX[0] | PWM[6] | INTERRUPT[5] | SDA[0] | SCI:CTS_RTS_SS[4] | SCI:TX_MOSI_SDA[7]
D05 | P210 | J2:60 |   | RPI36 |   | J15:35 | PWM_AGT[5] |   |   |   |   |  
D06 | P601 | J2:62 |   | RPI37 |   | J15:37 | PWM[6] | SCI:RX_MISO_SCL[9] |   |   |   |  
D07 | P402 | J2:35 | CS0 |   | CS0 | J15:10 | CAN_RX[0] | INTERRUPT[4] | SCI:CTS_RTS_SS[4] | SCI:RX_MISO_SCL[7] |   |  
D08 | P900 | J2:41 | MOSI0 |   | MOSI0 | J15:16 | SCI:TX_MOSI_SDA[4] |   |   |   |   |  
D09 | P204 | J2:37 | SCK0 |   | SCK0 | J15:12 | PWM[4] | SCI:SCK[4] | SCK[0] |   |   |  
D10 | P315 | J2:39 | MISO0 |   | MISO0 | J15:14 | SCI:RX_MISO_SCL[4] |   |   |   |   |  
D11 | P407 | J1:44 | SDA0 | RPI27 | SDA | J15:26,CAM:4 | PWM[6] | SDA[0] | SCI:CTS_RTS_SS[4] |   |   |  
D12 | P408 | J1:46 | SCL0 | RPI28 | SCL | J15:28,CAM:12 | PWM[6] | INTERRUPT[7] | SCL[0] | SCI:RX_MISO_SCL[3] | SCI:CTS_RTS_SS[4] |  
D13 | P110 | J1:35 | RX0 | RPI29 | SCI9/UART1 RX | J14:08 | CAN_RX[1] | PWM[1] | INTERRUPT[3] | SCI:CTS_RTS_SS[2] | SCI:RX_MISO_SCL[9] | MISO[0]
D14 | P602 | J1:33 | TX0 | RPI32 | SCI9/UART1 TX | J14:10 | PWM[7] | SCI:TX_MOSI_SDA[9] |   |   |   |  
D15 | P006 | J2:73 | A0 | PH01 |   | J15:09 | ANALOG[6] | INTERRUPT[11] |   |   |   |  
</body>

</html>
Wondering if there are enough pins exported, to use the hardware SPI0, especially since even if the CSI version was implemented, it looks pretty limited.  Like it suggests only 8 bit transfer mode 

Back to playing

facchinm, 2025-03-31:
Hi guys, the current `main` branch is synced with `arduino_core_merge_post_4.1` branch of the zephyr tree, which is turn updates the renesas hal to get rid of the BC issue.
I suggest you to run a `west update` after checking out `main` to sync all the dependencies, and to install via board manager version 0.3.0 to pull the latest `post_build_tool` (avoiding in this way the need to compile it with go).
Again, thanks a lot for all your work!
The plan for the next version is:
* enable `fully_linked` for all the boards
* support wifi on C33 via https://github.com/zephyrproject-rtos/zephyr/pull/87685
* merge @KurtE 's PR on Portenta H7
* refactor the core to get rid of the artifacts

KurtE, 2025-03-31:
Thanks @facchinm 

Sounds good.  At some point it might help to post, or ??? about the state of some of the things that have gone into the builds.

Examples: 

fully_linked - What does that really mean? I can guess parts of it, but does it require me to first load the bootloader or 
is that all part of it?  If I build again without fully-linked, do I need to reload the bootloader...

GIGA: Some of your zephyr forks, mention display shield... Is the shield working?  Same software libraries as MBED or ???

WiFi/BT: supported in... using ...

thanks again.

mjs513, 2025-03-31:
pretty much have the same questions as @KurtE.

facchinm, 2025-03-31:
I agree with you guys, the release notes are a bit lacking at the moment, but we'll try to get better 🙂 
I'm going to provide a status update in the wiki, hopefully it will be complete enough to answer some questions.

facchinm, 2025-03-31:
Here we go https://github.com/arduino/ArduinoCore-zephyr/wiki/%5B260331%5D-Project-status-%E2%80%90-release-0.3.0 . Hopefully since it's a wiki we can use the compatibility matrix to keep track of the various activities (feel free to modify the page of course 🙂 )

KurtE, 2025-03-31:
Thanks @facchinm 

Helps a lot.

A quick answer to myself about could I define another SPI object for SPI0...  

> Wondering if there are enough pins exported, to use the hardware SPI0, especially since even if the CSI version was implemented, it looks pretty limited. Like it suggests only 8 bit transfer mode
Back to playing

I don't think it would work very well:

MISO0 - D13 (could use), D41 (Used for CAN, most breakouts appear to only show after going through CAN ... D109 - Ethernet stuff
MOSI0 - D42 (Can), D110 Ethernet
SOCI0 - D2 OK, D9 OK, D111 Ethernet