arduino/ArduinoCore-zephyr #84: Portenta H7 using 4.1.0-rce

mjs513, 2025-03-19:
**Describe the bug**
Portenta H7 compiles and loads seemly after updating firmware but USB is lost after upload and lost at firmware upload.  However Sketch (BLINK) still runs!!!

**Target board + cli verbose compilation output**
```
loading library from d:\Users\Merli\Documents\Arduino\libraries\Arduino-EasyTransfer-master: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\ArduinoCore-API-master: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\FastAhrs: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\Lidar-Lite-v3-3d-Scanner: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\M5Stack: invalid library: no header files foundloading library from d:\Users\Merli\Documents\Arduino\libraries\esp32-camera-master: invalid library: no header files found
FQBN: arduino:zephyr:portentah7
Using board 'portentah7' from platform in folder: C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0
Using core 'arduino' from platform in folder: C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0

Detecting libraries used...
C:\Users\Merli\AppData\Local\Arduino15\packages\zephyr\tools\arm-zephyr-eabi\0.16.8/bin/arm-zephyr-eabi-g++ -g -Os -std=c++17 -c -w -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DHSE_VALUE=25000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=16384 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32H747xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -DCORE_CM7 -mcpu=cortex-m7 -mfloat-abi=softfp -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_PORTENTA_H7 -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino/api/deprecated -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino/api/deprecated-avr-comp -iprefixC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7 @C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/includes.txt C:\Users\Merli\AppData\Local\arduino\sketches\5A5C1C75825DB83AB3C3239B9E9BE89D\sketch\Blink.ino.cpp -o nul
Generating function prototypes...
C:\Users\Merli\AppData\Local\Arduino15\packages\zephyr\tools\arm-zephyr-eabi\0.16.8/bin/arm-zephyr-eabi-g++ -g -Os -std=c++17 -c -w -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DHSE_VALUE=25000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=16384 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32H747xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h -imacrosC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -DCORE_CM7 -mcpu=cortex-m7 -mfloat-abi=softfp -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_PORTENTA_H7 -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7 -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino/api/deprecated -IC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\cores\arduino/api/deprecated-avr-comp -iprefixC:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7 @C:\Users\Merli\AppData\Local\Arduino15\packages\arduino\hardware\zephyr\0.2.0\variants\arduino_portenta_h7/includes.txt C:\Users\Merli\AppData\Local\arduino\sketches\5A5C1C75825DB83AB3C3239B9E9BE89D\sketch\Blink.ino.cpp -o C:\Users\Merli\AppData\Local\Temp\4005199808\sketch_merged.cpp
C:\Users\Merli\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\Merli\AppData\Local\Temp\4005199808\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\zephyr\\tools\\arm-zephyr-eabi\\0.16.8/bin/arm-zephyr-eabi-g++" -g -Os -std=c++17 -c -DLL_EXTENSION_BUILD -DCONFIG_ARDUINO_API_SERIAL_BUFFER_SIZE=256 -DHSE_VALUE=25000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=16384 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32H747xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 "-imacrosC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\variants\\arduino_portenta_h7/llext-edk/include/zephyr/include/generated/zephyr/autoconf.h" "-imacrosC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\variants\\arduino_portenta_h7/llext-edk/include/zephyr/include/zephyr/toolchain/zephyr_stdint.h" -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -fdata-sections -ffunction-sections -fno-unwind-tables -fno-strict-aliasing -fno-printf-return-value -fno-common -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop --specs=picolibc.specs -D_POSIX_THREADS -mlong-calls -nodefaultlibs -DCORE_CM7 -MMD -mcpu=cortex-m7 -mfloat-abi=softfp -mfpu=fpv5-d16 -DARDUINO=10607 -DARDUINO_PORTENTA_H7 -DARDUINO_ARCH_ZEPHYR -DARDUINO_ARCH_ZEPHYR -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\cores\\arduino" "-IC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\variants\\arduino_portenta_h7" "-IC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\cores\\arduino/api/deprecated" "-IC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\Merli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\zephyr\\0.2.0\\variants\\arduino_portenta_h7" "@C:\\Users\\Merli\\AppData\\Local [... cut ...]
Download	[                         ]   0%            0 bytes
Download	[==========               ]  40%         4096 bytes
Download	[====================     ]  80%         8192 bytes
Download	[=========================] 100%        10196 bytes
Download done.
File downloaded successfully
Submitting leave request...
Transitioning to dfuMANIFEST state

```

**Output of Serial Monitor**
NONE - portenta not recognized by windows PC

**Optional: attach the sketch**
Use blink

**Additional context**
Add any other context about the problem here.

KurtE, 2025-03-20:
Some possible background information.  @mjs513 and I have been trying to figure out what is different.  Between the GIGA and the Portenta H7 and what changed between the stuff up till the zephyr 4.1.  

We know that the GIGA uses USB_FS for communications to the pc, where the Portenta uses USB_HS object for the communications. 
One thing we have noticed is that with the 4.1 build, the Portenta now has two CDC_ACM objects in the tree, where as earlier it only had one, and likewise the GIGA only has one in both the older and newer configurations.  Here is a snippit of the dtsh output
showing the two: tree --format NKYC > /mnt/d/temp/dtsh_H7_v4.1.html
[image]

I included the giga and H7 dtsh outputs here in case anyone wants to take a look

[dtsh_html_outputs.zip](https://github.com/user-attachments/files/19367935/dtsh_html_outputs.zip)

Now trying to figure out whare the 2nd one (board_cdc_acm_uart) came from and if it matters.

KurtE, 2025-03-22:
@facchinm @pillo79 @mjs513 - I believe I have a fix for the USB issue here.
As I showed in the posting two days ago.  The Portenta H7 added a new object to the usb@40040000 object, the
board_cdc_acm_uart.

I found that object was defined in zephyr: <../boards/common/usb/cdc_acm_serial.dtsi

I looked for who included this file and found that only two of the Arduino boards included it:
```
~/github/zephyr/boards/adafruit/itsybitsy/adafruit_itsybitsy_nrf52840.dts:
  153  };
  154  
  155: #include <../boards/common/usb/cdc_acm_serial.dtsi>
  156  

~/github/zephyr/boards/arduino/opta/arduino_opta_stm32h747xx_m7.dts:
   29  };
   30  
   31: #include <../boards/common/usb/cdc_acm_serial.dtsi>
   32  
   33  &clk_hse {

~/github/zephyr/boards/arduino/portenta_h7/arduino_portenta_h7_stm32h747xx_m7.dts:
    9  #include <st/h7/stm32h747xihx-pinctrl.dtsi>
   10  #include "arduino_portenta_h7-common.dtsi"
   11: #include <../boards/common/usb/cdc_acm_serial.dtsi>
   12  
   13  / {

~/github/zephyr/boards/atmarktechno/degu_evk/degu_evk.dts:
  174  };
```

I also noted that the GIGA had added another chosen define that was in that looked related, so I added it as well
to the H7s in the overlay, like:
```
    chosen {
        zephyr,console = &usart6;
        zephyr,shell-uart = &usart6;
        zephyr,cdc-acm-uart0 = &usart6;
        zephyr,uart-mcumgr = &usart6;
    };
```
(The uart-mcumgr) line.

I removed it from the Portenta H7 board and rebuilt and reflashed and programmed and now have USB working again on the H7

mjs513, 2025-03-22:
@KurtE - @facchinm - @pillo79 

Just verified that the fix does work using zephyr 16.8.  Just one note

Have to add `zephyr,uart-mcumgr = &usart6;` otherwise nothing is printed to Serial monitor.

closed 2025-03-31 by mjs513