arduino/ArduinoCore-zephyr #86: Giga R1 - USB fails after flashing bootloader

timonsku, 2025-03-26:
After flashing the bootloader the board does not come back and I sometimes get very rapid USB connect and disconnect events in Windows.

Here the flash log:
`Sketch uses 110728 bytes (5%) of program storage space. Maximum is 1966080 bytes.
Global variables use 47496 bytes (9%) of dynamic memory, leaving 476128 bytes for local variables. Maximum is 523624 bytes.
dfu-util 0.10-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
Device ID 2341:0366
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash  2MB   "
Downloading element to address = 0x08040000, size = 112920

Erase   	[                         ]   0%            0 bytes
Erase   	[                         ]   0%            0 bytes
Erase   	[=                        ]   7%         8192 bytes
Erase   	[==                       ]  10%        12288 bytes
Erase   	[===                      ]  14%        16384 bytes
Erase   	[====                     ]  18%        20480 bytes
Erase   	[=====                    ]  21%        24576 bytes
Erase   	[======                   ]  25%        28672 bytes
Erase   	[=======                  ]  29%        32768 bytes
Erase   	[========                 ]  32%        36864 bytes
Erase   	[=========                ]  36%        40960 bytes
Erase   	[==========               ]  43%        49152 bytes
Erase   	[===========              ]  47%        53248 bytes
Erase   	[============             ]  50%        57344 bytes
Erase   	[=============            ]  54%        61440 bytes
Erase   	[==============           ]  58%        65536 bytes
Erase   	[===============          ]  61%        69632 bytes
Erase   	[================         ]  65%        73728 bytes
Erase   	[=================        ]  68%        77824 bytes
Erase   	[==================       ]  72%        81920 bytes
Erase   	[===================      ]  76%        86016 bytes
Erase   	[====================     ]  83%        94208 bytes
Erase   	[=====================    ]  87%        98304 bytes
Erase   	[======================   ]  90%       102400 bytes
Erase   	[=======================  ]  94%       106496 bytes
Erase   	[======================== ]  97%       110592 bytes
Erase   	[=========================] 100%       112920 bytes
Erase    done.

Download	[                         ]   0%            0 bytes
Download	[=                        ]   7%         8192 bytes
Download	[==                       ]  10%        12288 bytes
Download	[===                      ]  14%        16384 bytes
Download	[====                     ]  18%        20480 bytes
Download	[=====                    ]  21%        24576 bytes
Download	[======                   ]  25%        28672 bytes
Download	[=======                  ]  29%        32768 bytes
Download	[========                 ]  32%        36864 bytes
Download	[=========                ]  36%        40960 bytes
Download	[==========               ]  43%        49152 bytes
Download	[===========              ]  47%        53248 bytes
Download	[============             ]  50%        57344 bytes
Download	[=============            ]  54%        61440 bytes
Download	[==============           ]  58%        65536 bytes
Download	[===============          ]  61%        69632 bytes
Download	[================         ]  65%        73728 bytes
Download	[=================        ]  68%        77824 bytes
Download	[==================       ]  72%        81920 bytes
Download	[===================      ]  76%        86016 bytes
Download	[====================     ]  83%        94208 bytes
Download	[=====================    ]  87%        98304 bytes
Download	[======================   ]  90%       102400 bytes
Download	[=======================  ]  94%       106496 bytes
Download	[======================== ]  97%       110592 bytesWarning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release

Download	[=========================] 100%       112920 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state`

facchinm, 2025-03-26:
Hi @timonsku ,
this is expected, since the USB is handled by the user sketch at the moment. Please flash any sketch (a Blink is fine) with the board in bootloader mode to get back the serial port.

timonsku, 2025-03-26:
That does not work. I have no USB device at all after flashing the bootloader, not just a missing serial device. The serial device is just the only thing that popped up briefly.
This is what I get when uploading a sketch after flashing the bootloader:
```
No DFU capable USB device available
Failed uploading: uploading error: exit status 74
```
Is it expected to upload code via DFU atm? It does work to upload code that way but the instruction for uploading the bootloader made it sound like it there would be a new bootloader (MCUboot?) to support the Zephyr core to upload without entering DFU manually.

timonsku, 2025-03-26:
Ah, got it now what you meant. The bootloader works after there is a user sketch. The instructions are a bit confusing though or rather what is the point of the "burn bootloader" part if the first uploaded sketch would be needed to have it available?
It would be more clear if after that bootloader flashing step that there is an empty user sketch included so that uploading code without DFU works after that step (or phrase the instruction differently if uploading a sketch via DFU yields the same outcome of flashing all partitions)

closed 2025-03-26 by timonsku