arduino/ArduinoCore-zephyr #92: GIGA Shield: Display Libraries

KurtE, 2025-04-01:
Not sure if best to open new issues for this or add to others:

As mentioned in the WIKI version 0.3.0 release:

>Giga Display shield supported (no wrapper library yet)

Is there example code that demonstrates the usage of the current code? 

I see some information up at: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/shields/arduino_giga_display_shield/doc/index.rst

My quick look at the board data, it appears like maybe it is setup for LVGL?

That index page shows some setup but hoping there is something more Arduino centric.

mjs513, 2025-04-02:
Been digging a bit this morning on this and looking at @facchinm  [giga_display_shield branch ](https://github.com/facchinm/zephyr/blob/a4939539b1f7f7433afca15fd36bb4ffcd7b84a1/include/zephyr/drivers/auxdisplay.h) (easier to see changes) looks like:
1.  Display uses the [ST7701 display driver](https://github.com/facchinm/zephyr/blob/a4939539b1f7f7433afca15fd36bb4ffcd7b84a1/drivers/display/display_st7701.c)  and `[#include <zephyr/drivers/display.h>](https://github.com/facchinm/zephyr/blob/a4939539b1f7f7433afca15fd36bb4ffcd7b84a1/include/zephyr/drivers/display.h)` 

so the wrapper would have to get set up like the camera lib.  Guessing here

mjs513, 2025-04-02:
Right now working on a preliminary wrapper for the display shield.  Slow going for me as its a bit different than what I am use to but making progress hopefully by end of day.

KurtE, 2025-04-02:
Sounds good, let me know when you have something to play with.

mjs513, 2025-04-03:
@facchinm  -  @KurtE 

Been playing a bit more the driver for the display and am getting slightly confused.

In ST7701.c file I see the api defined as:
```
static DEVICE_API(display, st7701_api) = {
	.blanking_on = st7701_blanking_on,
	.blanking_off = st7701_blanking_off,
	.set_brightness = st7701_set_brightness,
	.get_capabilities = st7701_get_capabilities,
};
```

while the display.h has
```

/**
 * @brief Display driver API
 * API which a display driver should expose
 */
__subsystem struct display_driver_api {
	display_blanking_on_api blanking_on;
	display_blanking_off_api blanking_off;
	display_write_api write;
	display_read_api read;
	display_clear_api clear;
	display_get_framebuffer_api get_framebuffer;
	display_set_brightness_api set_brightness;
	display_set_contrast_api set_contrast;
	display_get_capabilities_api get_capabilities;
	display_set_pixel_format_api set_pixel_format;
	display_set_orientation_api set_orientation;
};
```
Think missing from the st7701 driver, but that leads me to something else, why is the display even showing anything when I write to it - not sure where its going.  Note not sure how init is getting called in the st7701 driver.....

mjs513, 2025-04-03:
@facchinm 
Need some help need I went ahead and started update  display_st7701.c but noticed on rebuild
```
CMake Warning at /home/my_new_zephyr_folder/zephyr/CMakeLists.txt:1011 (message):
  No SOURCES given to Zephyr library: drivers__display
```
If I moded the config for the giga to add CONFIG_ST7701=y you can image what happens:
```
Merged configuration '/home/my_new_zephyr_folder/ArduinoCore-zephyr/loader/../variants/arduino_giga_r1_stm32h747xx_m7/arduino_giga_r1_stm32h747xx_m7.conf'

error: Aborting due to Kconfig warnings

CMake Error at /home/my_new_zephyr_folder/zephyr/cmake/modules/kconfig.cmake:365 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/my_new_zephyr_folder/zephyr/cmake/modules/zephyr_default.cmake:131 (include)
  /home/my_new_zephyr_folder/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/my_new_zephyr_folder/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:13 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/my_new_zephyr_folder/ArduinoCore-zephyr/venv/bin/python3 -B/home/my_new_zephyr_folder/ArduinoCore-zephyr/build/arduino_giga_r1_stm32h747xx_m7 -GNinja -DBOARD=arduino_giga_r1//m7 -S/home/my_new_zephyr_folder/ArduinoCore-zephyr/loader
```

Looking at the giga overlay further I am not seeing where it is linking back to the display/shield so it knows about the st7701?

mjs513, 2025-04-04:
@facchinm - @KurtE 

Have been playing around with this some more and while I added somethings to the st7701.c driver I am begining to get the feeling that the display config is not using this driver nor display.h:

Hints:
1.  Tried to use blanking to turn display on an off:
```
[00:00:04.510,000] <err> display_stm32_ltdc: There is no panel controller to forward blanking_on call to
[00:00:05.235,000] <err> display_stm32_ltdc: There is no panel controller to forward blanking_on call to
uart:~$
```
showing using display_stm32_ltdc

2.  Tried to change orientation and put some debug printk and even tried log_err's but nothing showed when I ran the sketch - also tried just returning something different than a error message and still got a -134 which should have been what I hard coded it too.

3.  Doen't appear that the display is ever being initialized correctly.

And what I think should be working doesn't.  

SO CANT GET THERE FROM HERE since not sure what is begin used.

KurtE, 2025-04-05:
@facchinm @mjs513 - Mike, I played around a little with your first sketch/library you posted two days ago.
Wondered if maybe it was more or less working, except that it was not using it's buffer in 16 bit mode, so I modified your sketch slightly

[GIGA_Display_first-250404a.zip](https://github.com/user-attachments/files/19612297/GIGA_Display_first-250404a.zip)
I figured that the buffer was like an image data, so I drew it in a 3x5 pattern...

And looks like might be working:

[image]

Tomorrow, may play with converting this to work with the Adafruit_gfx library.

@facchinm - I have it setup using your most recent zephyr branch in case it fixed some things

mjs513, 2025-04-05:
Thanks @KurtE 

You made me get up to play around  with your  sketch and see what I messed up - still can't rotate screen or do blanking since its using the ltdc driver.

New function added to your sketch - fill screen with a specified color.
```
#include "Arduino.h"
#include "Arduino_Giga_display.h"

Display display;

uint16_t color_table[] = {
  RGB565_RED, RGB565_MAGENTA, RGB565_YELLOW, RGB565_BLACK, RGB565_GREEN,
  RGB565_WHITE, RGB565_ORANGE, RGB565_GREENYELLOW, RGB565_PINK
};
uint8_t color_index = 0;

uint8_t frameBuffer2[2 * 160 * 160] __attribute__((aligned(32)));
uint32_t sizeof_framebuffer2 = sizeof(frameBuffer2);

void fillScreen(uint16_t color) {
  display.setFrameStatus(false);
  for (int x = 0; x < 480; x += 160) {
    for (int y = 0; y < 800; y += 160) {
      uint16_t *pb = (uint16_t *)frameBuffer2;
      uint32_t count = sizeof_framebuffer2 / 2;
      while(count--) *pb++ = color;
      display.write8(x, y, frameBuffer2);
    }
  }

  display.setFrameStatus(true);
}

void setup() {
  while (!Serial && millis() < 5000)
    ;
  Serial.begin(115200);

  if (!display.begin()) {
    Serial.println("Failed to start display");
  };

  Serial.println("Display configured!!");

  //uint8_t color = 0x11u;
  //memset(frameBuffer2, color, sizeof_framebuffer2);

  //display.setFrameDesc(160, 160, 160, sizeof_framebuffer2);
  display.setFrameStatus(false);

  fillScreen(RGB565_DARKGREY);

  for (int x = 0; x < 320; x += 160) {
    for (int y = 0; y < 480; y += 160) {
      uint16_t *pb = (uint16_t *)frameBuffer2;
      uint32_t count = sizeof_framebuffer2 / 2;
      while(count--) *pb++ = color_table[color_index];
      display.write8(x, y, frameBuffer2);
      color_index++;
      if (color_index == (sizeof(color_table)/sizeof(color_table[0]))) color_index = 0;
    }
  }

  display.setFrameStatus(true);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

notice do not have to specify setFrameDesc for some reason.

mjs513, 2025-04-06:
@KurtE - @facchinm @pillo79 

Been playing with this some more and just added the a getBuffer function to use stm32_ltdc_get_framebuffer to display your checkboard.  Still using fillScreen using a stardard buffer approac.  And added a fillRect function to just draw a single rectangle after displaying the checkboard.

[GIGA_Display_first-250406a.zip](https://github.com/user-attachments/files/19621738/GIGA_Display_first-250406a.zip)

cheers

mjs513, 2025-04-13:
@KurtE and others
Just to close the loop what has been done so far with the display.  Modifing the graphics GFX library and generating a simple wrapper library we tested a few different sketches.

The standard graphics test sketch with the modified libraries embedded in the sketch and buffering when the whole display is finished updated:
```
Zephyr Giga Display Test!
0
Benchmark                Time (microseconds)
Screen fill              837157
Text                     162244
Lines                    180089
Horiz/Vert Lines         169809
Rectangles (outline)     165466
Rectangles (filled)      607311
Circles (filled)         193723
Circles (outline)        169495
Triangles (outline)      164681
Triangles (filled)       25825
Rounded rects (outline)  166603
Rounded rects (filled)   760426
Done!
```
Example:
[image]

The second complete sketch was modified @KurtE's picture viewer (displays bmps, jpgs, and pngs using @bitbang2 libraries) does work however thing its running out of room since can only the jpegdec and bmps to display.  Did have to modify SD library since there is a Freemem function that will cause an error since it is geared to AVR only.
BMP example
[image]

JPG example
[image]

I am pushing up the graphics test sketches to @KurtE's giga library in sub folder for zephyr giga display

KurtE, 2025-05-03:
Quick notes: It was unclear to me how I might adapt the Arduino_GigaDisplayTouch library to zephyr.  There is a PT911 object
defined in the overlay for the board GIGA under the Shield (GIGA Display Shield), that I know is defined and running. 

As as soon as I touch the display I get lots of these message in the monitor window:
```
[00:00:25.648,000] <wrn> input: Event dropped, queue full, not blocking in syswq.
[00:00:25.658,000] <wrn> input: Event dropped, queue full, not blocking in syswq.
[00:00:25.669,000] <wrn> input: Event dropped, queue full, not blocking in syswq.
[00:00:25.679,000] <wrn> input: Event dropped, queue full, not blocking in syswq.
```
I know that the PT911 class is looking for me to hook up a callback function. like some of the input samples show:
`INPUT_CALLBACK_DEFINE(touch_dev, touch_event_callback, NULL);
` 
But I know that adding this define to a library that gets loaded in our LLTEXT code won't work.   Was not sure if there are any 
Dynamic ways to setup the callback.  So I asked up on Zephyr Discussion:
https://github.com/zephyrproject-rtos/zephyr/discussions/89415

The response I received:
https://github.com/zephyrproject-rtos/zephyr/discussions/89415#discussioncomment-13017025
From fabiobaltieri,
>Can't be done at the moment, but it'd be very easy to a static callback that goes through a dynamic list. Or just implement the >whole thing in the subsystem and make it Kconfig optional, don't see a reason not to accept such a contribution.

My read of this is, that I would need to add the callback code into some place that is built into the loader, like maybe fixups.c
and maybe that code would receive the callbacks and save the touch information and maybe set a semaphore or the like
and then the library code could detect the semaphore and retrieve the last callback information.
Probably beyond my pay grade 😆, especially if your plans are to hard code the GIGA Display to always use LVGL.

mjs513, 2025-05-15:
Quick update: See https://github.com/arduino/ArduinoCore-zephyr/pull/117#issuecomment-2883919658

mjs513, 2025-06-05:
@KurtE - @pillo79 - @facchinm 

Started moving our graphics stuff to libraries so easier to test with and hopefully a starting point for when it gets officially incorported.

1.  Moved our zephyr changes into the arduino_GigaDisplay_GFX lib: https://github.com/mjs513/Arduino_GigaDisplay_GFX/tree/Arduino_GigaDisplay_GFX_zephyr
2. and the giga display api to Arduino_GigaDisplay - incorporated a new class file: https://github.com/mjs513/Arduino_GigaDisplay/tree/Arduino_GigaDisplay_Zephyr

tested with out graphics test sketch and the touch screen test.  As time permits have to test other sketches.

Also was looking at ArduinoGraphics lib but no clue how that gets linked to the display.  Going to look more - was late when I started looking at that.

KurtE, 2025-06-05:
I did do a PR for adding preliminary touch 
[#134](https://github.com/arduino/ArduinoCore-zephyr/pull/134)

I also pushed up our changes into the Arduino_GigaDisplayTouch library:
https://github.com/KurtE/Arduino_GigaDisplayTouch/tree/zephyr

@facchinm @pillo79 @iabdalkader - Should we create PR's yet for these three libraries that depend on some of the current
prs within this bithub project?  Or wait until you do so?

mjs513, 2025-06-08:
@facchinm - @pillo79 - @KurtE 

I managed to get ArduinoGraphics library working with Zephyr Giga Display:

[image]

Changes were made to gigadisplay.h in my version of Arduino_GigaDisplay library.   

Looking at Arduino_H7_video looks like these changes should be in that library but it is so focused on MBed trying to incorporate it as is would be challenging to say the least.  But at lease we know it works and operational.

leonardocavagnis, 2026-04-01:
@mjs513 Ah sorry, I didn’t notice. The repo is private and still a work in progress.

I fixed the LVGL memory issues by applying the same changes used [here](https://github.com/arduino/ArduinoCore-zephyr/pull/356) for the Arduino Portenta H7, and by updating [these lines](https://github.com/mjs513/Arduino_GigaDisplay/blob/23a01e13787e3105243047697c211bee2e0788dc/src/lv_conf_9.h#L38-L40) in `lv_conf.h`:

```
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_CLIB
#define LV_USE_STDLIB_STRING    LV_STDLIB_CLIB
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_CLIB
```

pennam, 2026-09-01:
Im going to close this since [Arduino_Video](https://github.com/arduino-libraries/Arduino_Video/releases) has been released together with [Arduino_GigaDisplay](https://github.com/arduino-libraries/Arduino_GigaDisplay/releases/tag/1.1.1) , [Arduino_GigaDisplay_GFX](https://github.com/arduino-libraries/Arduino_GigaDisplay_GFX/releases) and [Arduino_GigaDisplayTouch](https://github.com/arduino-libraries/Arduino_GigaDisplayTouch/releases)

closed 2026-09-01 by pennam