arduino/ArduinoCore-zephyr #100: Loosing Com Port on GIGA after reinstalling 3.1

mjs513, 2025-04-10:
After installing 2.3.6 (IDE) started having issues with the GIGA and loosing the com port while working on Display.

After several tries of rebuilding in ubuntu opted to reinstall 0.3.1 core via the board manager after removing the old core.  I did incorporate the USB fixes to make sure I had that covered. 

After that I closed the IDE and reopened it.  Burned the bootloader and ran the ASCIItable example sketch.  Again no com port showed after uploading sketch.

If I cycle power (unplug and replug it back it in) I get the Com port back and serial monitor prints out the table.  But if I hit restart I loose the com port again and nothich shows on serial1 for errors.

If I try loading the sketch in debug mode the sketch loads and runs correctly without loosing the com port.

reloading in standard mode it goes back to no comport until I cycle power.

No clue whats going on......

Dont believe @KurtE is having any issues with the giga.

Note. Tried teh C33 and not seeing this problem.  Havent tried the H7 yet.

mjs513, 2025-04-11:
Playing around again with the giga verified that I put in the USB Fixes I ran a different sketch, k_define test sketch.  Same thing happens.  To get Comport back have to remove power for about 10seconds and it comes back.  However, the sketch fails to run without any error messages in debug mode.

[image]

but I am not seeing the usual cdc enable message on Serial1.

ok finally got somethig on serial1 when I switched back to standard mode using the k_define test sketch
```
uart:~$
[00:00:01.326,000] <err> os: ***** BUS FAULT *****
[00:00:01.334,000] <err> os:   Imprecise data bus error
[00:00:01.342,000] <err> os: r0/a1:  0x2400fcf8  r1/a2:  0x000000aa  r2/a3:  0x480203a8
[00:00:01.353,000] <err> os: r3/a4:  0x24080007 r12/ip:  0x2400f089 r14/lr:  0x080570e3
[00:00:01.363,000] <err> os:  xpsr:  0x81000000
[00:00:01.371,000] <err> os: s[ 0]:  0xaaaaaaaa  s[ 1]:  0xaaaaaaaa  s[ 2]:  0xaaaaaaaa  s[ 3]:  0xaaaaaaaa
[00:00:01.383,000] <err> os: s[ 4]:  0xaaaaaaaa  s[ 5]:  0xaaaaaaaa  s[ 6]:  0xaaaaaaaa  s[ 7]:  0xaaaaaaaa
[00:00:01.396,000] <err> os: s[ 8]:  0xaaaaaaaa  s[ 9]:  0xaaaaaaaa  s[10]:  0xaaaaaaaa  s[11]:  0xaaaaaaaa
[00:00:01.409,000] <err> os: s[12]:  0xaaaaaaaa  s[13]:  0xaaaaaaaa  s[14]:  0xaaaaaaaa  s[15]:  0xaaaaaaaa
[00:00:01.421,000] <err> os: fpscr:  0xaaaaaaaa
[00:00:01.429,000] <err> os: Faulting instruction address (r15/pc): 0x08061360
[00:00:01.439,000] <err> os: >>> ZEPHYR FATAL ERROR 26: Unknown error on CPU 0
[00:00:01.449,000] <err> os: Current thread: 0x24003220 (main)
[00:00:01.457,000] <err> os: Halting system
```
this is the test sketch
```
/*
 * Copyright (c) 2017 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#include <zephyr/kernel.h>

#include "elapsedMillis.h"
elapsedMillis timer;

/* STEP 2 - Define stack size and scheduling priority used by each thread */
#define STACKSIZE 1024

#define THREAD0_PRIORITY 7
#define THREAD1_PRIORITY 7

void thread0(void)
{
  Serial.print("Hello Thread 0 Started\n");
  while (1) {
    /* STEP 3 - Call printk() to display a simple string "Hello, I am thread0" */
    Serial.print("Hello, I am thread0\n");
    /* STEP 6 - Make the thread yield */
    // k_yield();
    /* STEP 10 - Put the thread to sleep */
    k_msleep(5);
    /* Remember to comment out the line from STEP 6 */
  }
}

void thread1(void)
{
  Serial.println("Thread 1 Started...");
  while (1) {
    /* STEP 3 - Call printk() to display a simple string "Hello, I am thread1" */
    Serial.print("Hello, I am thread1\n");
    /* STEP 8 - Make the thread yield */
    // k_yield();
    /* STEP 10 - Put the thread to sleep */
    k_msleep(10);
    /* Remember to comment out the line from STEP 8 */
  }
}

/* STEP 4 - Define and initialize the two threads */
K_THREAD_DEFINE(thread0_id, STACKSIZE, thread0, NULL, NULL, NULL, THREAD0_PRIORITY, 0, 0);
K_THREAD_DEFINE(thread1_id, STACKSIZE, thread1, NULL, NULL, NULL, THREAD1_PRIORITY, 0, 0);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {};
  Serial.println("Threading sketch #2 started");

  delay(100);

k_tid_t our_tid = k_sched_current_thread_query();
  int main_pri = k_thread_priority_get(our_tid);
  Serial.print("main TID: ");
  Serial.print((uint32_t)our_tid, HEX);
  Serial.print(" pri: ");
  Serial.println(main_pri);
  printk("main TID:%x pri:%d\n", (uint32_t)our_tid, main_pri);
  k_thread_priority_set(our_tid, THREAD0_PRIORITY+1);
  main_pri = k_thread_priority_get(our_tid);
  Serial.print("\tupdated pri: ");
  Serial.println(main_pri);
  printk("main TID:%x pri:%d\n", (uint32_t)our_tid, main_pri);

  Serial.println("End Setup");
}

void loop() {
  // put your main code here, to run repeatedly:
  if(timer > 5000){
    Serial.println("called from loop");
    timer = 0;
  }
  k_msleep(100);
}
```

mjs513, 2025-04-11:
Another data point - if I run the graphics test sketch that we were working on:
```
- Capabilities:
  x_resolution [00:00:01.057,000] <inf> usb_cdc_acm: Device suspended
uart:~$ _resolution = 800
, supported_pixel_formats = 25
  current_pixel_format = 16, current_orientation = 0
[00:00:01.076,000] <err> os: ***** BUS FAULT *****
[00:00:01.083,000] <err> os:   Imprecise data bus error
[00:00:01.091,000] <err> os: r0/a1:  0xc0000000  r1/a2:  0xc0000000  r2/a3:  0xc00bb80c
[00:00:01.102,000] <err> os: r3/a4:  0xc0000000 r12/ip:  0x00000000 r14/lr:  0x240100a5
[00:00:01.113,000] <err> os:  xpsr:  0x01000000
[00:00:01.120,000] <err> os: s[ 0]:  0x24020ae4  s[ 1]:  0x0805a3a1  s[ 2]:  0x00000000  s[ 3]:  0x00000000
[00:00:01.133,000] <err> os: s[ 4]:  0x2401314b  s[ 5]:  0x24036824  s[ 6]:  0x240381b8  s[ 7]:  0x0805a0ad
[00:00:01.146,000] <err> os: s[ 8]:  0x2401314b  s[ 9]:  0x24036824  s[10]:  0x00000320  s[11]:  0x24013060
[00:00:01.158,000] <err> os: s[12]:  0x24013060  s[13]:  0x24413060  s[14]:  0xc0000000  s[15]:  0xc00bb80c
[00:00:01.171,000] <err> os: fpscr:  0x24013060
[00:00:01.178,000] <err> os: Faulting instruction address (r15/pc): 0x2400ff52
[00:00:01.188,000] <err> os: >>> ZEPHYR FATAL ERROR 26: Unknown error on CPU 0
[00:00:01.198,000] <err> os: Current thread: 0x24003220 (main)
[00:00:01.207,000] <err> os: Halting system
```
so maybe the bus fault is a related issues..

[graphicsTestGigaDisplay-250411a.zip](https://github.com/user-attachments/files/19708408/graphicsTestGigaDisplay-250411a.zip)

Note have same issues with a second giga I have.

mjs513, 2025-04-12:
@facchinm - @pillo79 
Any thoughts on this one because I am kind of dead in the water with using the giga and zephyr

mjs513, 2025-04-12:
@KurtE and others

I just re cloned and setup from the latest changes and included the USB Hot fixes.

After loading gigascanner still no serial (no comport) on serial1 all I see is
```
uart:~$
```
if I do the 10sec power off and power on I get the com port back and see
```
[00:00:01.057,000] <inf> usb_cdc_acm: Device suspended
[00:00:01.279,000] <inf> usb_cdc_acm: Device configured
uart:~$
```

If I back out the hot fixes and rebuild for the Giga same thing still happens

As you said something not getting set right or clear out.

KurtE, 2025-04-12:
Also wondering if there might have been something different between older and newer batches of the GIGA? 
The one that is working without typing sketch (I think newer bust dustier... I used canned air on the other 
 
[image]

The one I need to type sketch on:

[image]

Maybe some timing issue? Difference in voltage regulators, which take longer or shorter to get to stable voltage? 

Just drawing straws.

mjs513, 2025-04-12:
@KurtE 
Just for reference my old matches yours:
[image]

My new one (bougth about the same time as your new one)
[image]
has slightly different markings.

mjs513, 2025-04-12:
@KurtE and others
Using my old board where I can at least get USB back in debug mode I tested with several sketches
k_create:  works no issue in debug mode
k_define:  Gives an imprecise data bus fault !!!!
SDRAM_test: Works
graphicsTestGigaDisplay: Works no issue

Now if I run the graphicstest sketch on the new board:
```
[00:00:02.162,000] <err> os: ***** BUS FAULT *****
[00:00:02.169,000] <err> os:   Imprecise data bus error
[00:00:02.177,000] <err> os: r0/a1:  0xc0000000  r1/a2:  0xc0000000  r2/a3:  0x34813080
[00:00:02.188,000] <err> os: r3/a4:  0xc0000000 r12/ip:  0x00000000 r14/lr:  0x24010085
[00:00:02.199,000] <err> os:  xpsr:  0x01000000
[00:00:02.206,000] <err> os: s[ 0]:  0x24012965  s[ 1]:  0x00000000  s[ 2]:  0x00000010  s[ 3]:  0x00000000
[00:00:02.219,000] <err> os: s[ 4]:  0x24020ae4  s[ 5]:  0x0805a3a1  s[ 6]:  0x00000000  s[ 7]:  0x00000000
[00:00:02.232,000] <err> os: s[ 8]:  0x2401316b  s[ 9]:  0x24036824  s[10]:  0x240381b8  s[11]:  0x0805a0ad
[00:00:02.244,000] <err> os: s[12]:  0x2401316b  s[13]:  0x24036824  s[14]:  0x00000320  s[15]:  0x24013080
[00:00:02.257,000] <err> os: fpscr:  0x24013080
[00:00:02.264,000] <err> os: Faulting instruction address (r15/pc): 0x2400ff3e
[00:00:02.274,000] <err> os: >>> ZEPHYR FATAL ERROR 26: Unknown error on CPU 0
[00:00:02.284,000] <err> os: Current thread: 0x24003220 (main)
[00:00:02.293,000] <err> os: Halting system
```
bus fault is back.

and occurs with k_define and sdram_test as well.

So now have 2 giga boards behaving totally different with same firmware loaded!!!!!

mjs513, 2026-05-28:
Ok going to close this.  Just tried 0.55.2 on the misbehaving board and seems to be working.  Just wanted to close the loop

closed 2026-05-28 by mjs513