arduino/ArduinoCore-zephyr #106: Nano sense 33 BLE BMI270 failing to start again with latest changes as of 4/12/25

mjs513, 2025-04-13:
Rerunning BMI270 test sketches on the Sense is showing failing to start the BMI270 using the Sparkfun library so ran the Arduino BMI270-BMM150 library in debug mode.  

Now getting the following error:
```
[00:00:08.562,713] [1;31m<err> llext: Failed to allocate extension symbol table, ret -12[0m
```

mjs513, 2025-04-14:
Sorry didn't error didn't register yesterday - probably because GIGA stuff had me crazy.

Anyway changed llext_heap_size to 128 from 96 and that seems to have fixed it so that we don't get that error anymore.  Now sure what happened that not the BMI270 isn;t working with the sparkfun lib.  With the Arduino version it says started but not seeing any data so something is up.  Have to dig but may wait to dig into the libraries.

You want a PR for the change?

mjs513, 2025-04-15:
@pillo79 - @facchinm 

Even after making the the BMI270 was failing to start as I mentioned.  But found something interesting.  After uploading the sketech I will get Device not found error.  But if I re-burn the bootloader the sketch starts and runs correctly:

[image]

Also works if I upload the sketch after I do a double of the reset button???  

Of course if I power off and power on I get back to not finding the BMI270????

mjs513, 2025-04-15:
@pillo79 - @facchinm - @KurtE

Did finally get the Arduino_BMI270_BMM150 library working with Zephyr.  But again strange things happen.

1.  When I first upload the sketch I loose the com port/serial monitor.  To get it back I have to cycle power.  This happens fairly consistently and the resultant data is off - all accel axes are off.
```
-0.50	-0.50	0.50
-0.50	-0.50	0.50
-0.50	-0.50	0.50
```

If I cycle power I will get good data"
```
0.01	0.01	0.99
0.01	0.01	0.99
0.01	0.01	0.99
````

2.  If the sketch loads and I don't loose the com port its hit or miss getting good data with out cycling power as mentioned above.  Most of the time it takes several attempts at cycling power to get it working correctly.  Most of the time you have to wait a couple of minutes.

Here is the modified library if you want to try.

[Arduino_BMI270_BMM150.zip](https://github.com/user-attachments/files/19758955/Arduino_BMI270_BMM150.zip)

Just using the accelerometer sketch to test with.

mjs513, 2025-04-26:
Retested with the latest changes just posted and still seeing the same isssue in https://github.com/arduino/ArduinoCore-zephyr/issues/106#issuecomment-2809590311

facchinm, 2025-04-28:
@mjs513 I just retested and it looks fine after applying https://github.com/arduino-libraries/Arduino_BMI270_BMM150/pull/51 .
Will check on a clean installation this afternoon

mjs513, 2025-04-28:
@facchinm 
Pretty much made the same changes in my version of the library - but to be consistent I just made the same changes you made and reran my combined sensor (BMM150+BMI270).  Note: updated to lastest core changes less the USB Fixes PR which is still open.

on first upload I am seeing incorrect data from the BMM270:
```
Accel (x, y, z):	-0.50	-0.50	0.50
Mag (x, y, z):		-26.00	0.00	32.00
```
powering off for several seconds and the powering on I see the correct data
```
Accel (x, y, z):	0.02	0.01	0.99
Gyro (x, y, z):		-0.24	0.06	0.12
==================
Accel (x, y, z):	0.02	0.01	0.99
Gyro (x, y, z):		-0.18	0.06	0.06
Mag (x, y, z):		-28.00	-1.00	30.00
```

that is the problem I am really seeing.

closed 2025-05-15 by mjs513