arduino/ArduinoCore-zephyr #113: zephyr-sketch-tool not found

mjs513, 2025-04-25:
@facchinm - @pillo79 - @KurtE

Just tried with all the latest changes and received the error:
```
exec: "{runtime.tools.zephyr-sketch-tool.path}/zephyr-sketch-tool": executable file not found in %PATH%
```
so I went ahead and built the tool using the go build and it create a zephy-sketch-tool.  But didn't see the windows exe -  ALSO where do you put it!!!  I guessed but no luck - ps how do you build for windows?

mjs513, 2025-04-26:
Spent more time today trying to sort this issue out.  Did notice when looking at actual PR:
https://github.com/arduino/ArduinoCore-zephyr/pull/102/commits/98ec98da9bece1eafc02012e1147ecc230ab793c

there is a `build.sh` file that is not currently in the core.  Coping that file over to my local core extra/zephyr-sketch-tool directory an building it you will see the following errors:
```
# ./build.sh
mv: cannot stat 'zephyr-post-build-tool': No such file or directory
mv: cannot stat 'zephyr-post-build-tool': No such file or directory
mv: cannot stat 'zephyr-post-build-tool': No such file or directory
mv: cannot stat 'zephyr-post-build-tool*': No such file or directory
  adding: windows_386/ (stored 0%)
```
but it still builds the exe for windows as well as linux.
```
-rwxr-xr-x 1 root root     812 Apr 17 03:08 build.sh
drwxr-xr-x 6 root root    4096 Apr 26 09:58 distrib
-rw-r--r-- 1 root root      56 Apr 17 03:08 go.mod
-rw-r--r-- 1 root root    2473 Apr 17 03:08 main.go
-rwxr-xr-x 1 root root 2552864 Apr 26 09:58 zephyr-sketch-tool
-rwxr-xr-x 1 root root 2500608 Apr 26 09:58 zephyr-sketch-tool.exe
```
Now when I create a zephyr-sketch-tool directory in the same location as the as the old post build tool and mv the exe file to it I still receive the error:
```
exec: "{runtime.tools.zephyr-sketch-tool.path}/zephyr-sketch-tool": executable file not found in %PATH%

Compilation error: exec: "{runtime.tools.zephyr-sketch-tool.path}/zephyr-sketch-tool": executable file not found in %PATH%
```
so still no luck

KurtE, 2025-04-26:
On my Ubuntu machine I tried, creating new directory under the tools directory and copied the binary there and it appears to
now be happier.

```
kurte@kurte-750-435st:~/.arduino15/packages/arduino/tools/zephyr-sketch-tool/0.1.0$ ls -l
total 2472
-rwxrwxr-x 1 kurte kurte 2528618 Apr 26 07:06 zephyr-sketch-tool
kurte@kurte-750-435st:~/.arduino15/packages/arduino/tools/zephyr-sketch-tool/0.1.0$ 

```
End part of the build:
```
Archiving built core (caching) in: /home/kurte/.cache/arduino/cores/arduino-git_ArduinoCore-zephyr_portentac33_debug_true,mode_llext_7b383d2f1323a72b871bcc17d2cfe0f5/core.a
Linking everything together...
/home/kurte/.arduino15/packages/zephyr/tools/arm-zephyr-eabi/0.16.8/bin/arm-zephyr-eabi-g++ -L/home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B -r -e main -Wl,--gc-sections -mcpu=cortex-m33 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -std=c++17 -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fno-use-cxa-atexit -lstdc++ -lsupc++ -lnosys -nostdlib -T/home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/variants/_llext/linker_script.ld -Wl,-Map,/home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.map --specs=picolibc.specs --specs=nosys.specs -o /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino_debug.elf /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/sketch/SerialPassthroughX.ino.cpp.o /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/core/variant.cpp.o /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/core/core.a
/home/kurte/.arduino15/packages/zephyr/tools/arm-zephyr-eabi/0.16.8/bin/arm-zephyr-eabi-strip --strip-debug /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino_debug.elf -o /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf
/home/kurte/.arduino15/packages/zephyr/tools/arm-zephyr-eabi/0.16.8/bin/arm-zephyr-eabi-objcopy -O binary /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.bin
/home/kurte/.arduino15/packages/zephyr/tools/arm-zephyr-eabi/0.16.8/bin/arm-zephyr-eabi-objcopy -O ihex -R .eeprom /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.hex
/home/kurte/.arduino15/packages/arduino/tools/zephyr-sketch-tool/0.1.0/zephyr-sketch-tool -debug /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf
Bootloader file specified but missing: /home/kurte/Arduino/hardware/arduino-git/ArduinoCore-zephyr/bootloaders/zephyr-arduino_portenta_c33_r7fa6m5bh3cfc.bin
File /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf saved as /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf-zsk.bin
/home/kurte/.arduino15/packages/arduino/tools/zephyr-sketch-tool/0.1.0/zephyr-sketch-tool -debug /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.bin
File does not have an ELF header, adding empty space
File /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.bin saved as /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.bin-zsk.bin
/home/kurte/.arduino15/packages/zephyr/tools/arm-zephyr-eabi/0.16.8/bin/arm-zephyr-eabi-size -A /home/kurte/.cache/arduino/sketches/A314C4671ACAB9459BF7C9BFFF93640B/SerialPassthroughX.ino.elf
Sketch uses 3992 bytes (0%) of program storage space. Maximum is 1966080 bytes.
Global variables use 1652 bytes (0%) of dynamic memory, leaving 521972 bytes for local variables. Maximum is 523624 bytes.
```

KurtE, 2025-04-26:
On my Windows machine now, I created a new build.sh file.
I copied over the older one from the post build name and edited it to change names of directories and destinations:

My current version:

```
VERSION=$1

mkdir -p distrib/linux_amd64
GOOS=linux GOARCH=amd64 go build && mv zephyr-sketch-tool distrib/linux_amd64/zephyr-sketch-tool

mkdir -p distrib/linux_arm64
GOOS=linux GOARCH=arm64 go build && mv zephyr-sketch-tool distrib/linux_arm64/zephyr-sketch-tool

mkdir -p distrib/macos_amd64
GOOS=darwin GOARCH=amd64 go build && mv zephyr-sketch-tool distrib/macos_amd64/zephyr-sketch-tool

mkdir -p distrib/windows_386
GOOS=windows GOARCH=386 go build && mv zephyr-sketch-tool* distrib/windows_386/zephyr-sketch-tool.exe

cd distrib

tar -czf zephyr-sketch-tool-${VERSION}-linux_amd64.tar.gz linux_amd64
tar -czf zephyr-sketch-tool-${VERSION}-linux_arm64.tar.gz linux_arm64
tar -czf zephyr-sketch-tool-${VERSION}-macos_amd64.tar.gz macos_amd64
zip -r zephyr-sketch-tool-${VERSION}-windows_386.zip windows_386
```

I then ran it on my WSL  fork/branch of it on my Windows machine:
```
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool$ ls -l
total 16
-rwxr-xr-x 1 kurte kurte  812 Apr 26 09:42 build.sh
drwxr-xr-x 2 kurte kurte 4096 Apr 26 09:43 distrib
-rw-r--r-- 1 kurte kurte   56 Apr 26 09:39 go.mod
-rw-r--r-- 1 kurte kurte 2473 Apr 26 09:39 main.go
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool$ ./build 010
-bash: ./build: No such file or directory
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool$ ./build.sh 010
  adding: windows_386/ (stored 0%)
  adding: windows_386/zephyr-sketch-tool.exe (deflated 40%)
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool$ ls
build.sh  distrib  go.mod  main.go
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool$ cd distrib
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool/distrib$ ls
linux_amd64  macos_amd64  zephyr-sketch-tool-010-linux_amd64.tar.gz  zephyr-sketch-tool-010-macos_amd64.tar.gz
linux_arm64  windows_386  zephyr-sketch-tool-010-linux_arm64.tar.gz  zephyr-sketch-tool-010-windows_386.zip
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool/distrib$ cd windows_386/
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool/distrib/windows_386$ ls
zephyr-sketch-tool.exe
kurte@KurtXPS8950:~/git/ArduinoCore-zephyr/extra/zephyr-sketch-tool/distrib/windows_386$
```

Looks like it created an exe... will try it out soon.

mjs513, 2025-04-26:
@KurtE 

I moved the created a 0.1.0 directory under zephyr-sketch-tool director in tools and moved the exe to that directory and now sketches are loading to the Nano 33 ble sense that I am testing with.  Thanks

Also updated to your build.sh changes to avoid the errors.  Thanks for fixing that.

facchinm, 2025-04-28:
The tool was shipped with the "released" core 0.3.1, so updating from board manager should fix all the dangling links (of course, this doesn't impact the git core in any way)

mjs513, 2025-04-28:
@facchinm 
Interesting in the 3.1 version I had the tool was  not there?  Not sure why.  I just removed and reinstalled 0.3.1 and sure enough I am seeing it now.  Which is weird.

Anyways maybe should add build.sh back into the core just to be complete.

KurtE, 2025-04-28:
It was not there when I also installed... will try uninstall... reinstall as well

pillo79, 2025-04-28:
> Interesting in the 0.3.1 version I had the tool was not there? Not sure why. I just removed and reinstalled 0.3.1 and sure enough I am seeing it now. Which is weird.

Uh sorry guys for the confusion, my fault indeed. I updated the release adding a tool that was not used (yet) just for me to test, with the intention of starting to use it in the next version - but of course you had to reinstall the core to get it.