arduino/ArduinoCore-zephyr #124: Portenta H7: Why does blue LED blink...

KurtE, 2025-05-17:
**Describe the bug**
On every or almost every sketch I run on the Portenta H7, the BLUE led continuously blinks.
I believe it is probably connected to the charging circuit and there is no battery connected to my board.
And it do not blink like this when running on MBED.

Hopefully a simple one to resolve.

mjs513, 2025-05-17:
Just for the record I am seeing the same thing as @KurtE

KurtE, 2025-05-19:
Notes to self and probably only myself.....

When I boot up a simple sketch like blink on MBED: I see that the following stuff is output over I2C1, to device 0x08
```
write to 0x08 ack data: 0x4F 0x00 
write to 0x08 ack data: 0x50 0x0F 
write to 0x08 ack data: 0x4C 0x05 
write to 0x08 ack data: 0x4D 0x03 
write to 0x08 ack data: 0x52 0x09 
write to 0x08 ack data: 0x53 0x0F 
write to 0x08 ack data: 0x9C 0x80 
write to 0x08 ack data: 0x9E 0x20 
write to 0x08 ack data: 0x42 0x02 
write to 0x08 ack data: 0x94 0xA0 
write to 0x08 ack data: 0x3B 0x0F 
write to 0x08 ack data: 0x35 0x0F 
write to 0x08 ack data: 0x42 0x01 
```

And when I run it on Zephyr:
```
write to 0x08 ack data: 0x4F 0x00 
write to 0x08 ack data: 0x50 0x0F 
write to 0x08 ack data: 0x4C 0x05 
write to 0x08 ack data: 0x4D 0x03 
write to 0x08 ack data: 0x52 0x09 
write to 0x08 ack data: 0x53 0x0F 
write to 0x08 ack data: 0x9C 0x80 
write to 0x08 ack data: 0x9E 0x20 
write to 0x08 ack data: 0x42 0x02 
write to 0x08 ack data: 0x94 0xA0 
write to 0x08 ack data: 0x3B 0x0F 
write to 0x08 ack data: 0x35 0x0F 
write to 0x08 ack data: 0x00 
read to 0x08 ack data: 0x7C
write to 0x08 ack data: 0x32 0x06 
write to 0x08 ack data: 0x35 
read to 0x08 ack data: 0x0F
write to 0x08 ack data: 0x38 0x07 
write to 0x08 ack data: 0x3B 
read to 0x08 ack data: 0x0F
write to 0x08 ack data: 0x3E 0x0F 
write to 0x08 ack data: 0x42 
read to 0x08 ack data: 0x02
write to 0x08 ack data: 0x42 0x02 
write to 0x08 ack data: 0x41 
read to 0x08 ack data: 0x03
write to 0x08 ack data: 0x4C 0x05 
write to 0x08 ack data: 0x4D 
read to 0x08 ack data: 0x03
write to 0x08 ack data: 0x4F 0x00 
write to 0x08 ack data: 0x50 
read to 0x08 ack data: 0x0F
write to 0x08 ack data: 0x52 0x09 
write to 0x08 ack data: 0x53 
read to 0x08 ack data: 0x0F
```
Which I believe is the device that controls a lot of power managment:
https://www.nxp.com/docs/en/data-sheet/PF1550.pdf
Now to see if any of it makes sense

KurtE, 2025-05-21:
Quick update... The 2nd post was barking up the wrong tree ....
The LED that is blinking is not the charge LED, but instead the BLUE one part of the RGB LED stuff.

I believe it is probably related to the code in the MBED version where the
```
void initVariant() {
  RTCHandle.Instance = RTC;
  // Turn off LED from bootloader
  pinMode(PK_6, OUTPUT);
  digitalWrite(PK_6, HIGH);

```
I have tried to add similar code to an initVariant function for zephyr PortentaH7, but that did not stop blink, tried adding to
sketch as well did not stop it.  Guessing there is some timer or the like running ... still looking.

KurtE, 2025-05-21:
Thanks @facchinm will try it out later...

Which leads into another set of questions, on How you see things proceeding, like controlling the second core.
But I think it may be better for a discussion...

KurtE, 2025-05-22:
@facchinm - 
>Let me know if it works!

So far no luck.  If I understood correctly, I tried running this sketch on MBED:
```
#include <RPC.h>
void setup() {
  // put your setup code here, to run once:
  // If CM4 is already booted, disable auto-boot and reset.
  Serial.begin(115200);
  while (!Serial && millis() < 10000) {}
  FLASH_OBProgramInitTypeDef OBInit;

  OBInit.Banks = FLASH_BANK_1;
  HAL_FLASHEx_OBGetConfig(&OBInit);
  if (OBInit.USERConfig & FLASH_OPTSR_BCM4) {
    OBInit.OptionType = OPTIONBYTE_USER;
    OBInit.USERType = OB_USER_BCM4;
    OBInit.USERConfig = 0;
    check_return_status("HAL_FLASH_OB_Unlock ", HAL_FLASH_OB_Unlock());
    check_return_status("HAL_FLASH_Unlock ", HAL_FLASH_Unlock());
    check_return_status("HAL_FLASHEx_OBProgram ", HAL_FLASHEx_OBProgram(&OBInit));
    check_return_status("HAL_FLASH_OB_Launch ", HAL_FLASH_OB_Launch());
    check_return_status("HAL_FLASH_OB_Lock ", HAL_FLASH_OB_Lock());
    check_return_status("HAL_FLASH_Lock ", HAL_FLASH_Lock());
    Serial.println("CM4 autoboot disabled\n");
    NVIC_SystemReset();
    return;
  } else {
    Serial.println("FLASH_OPTSR_BCM4 not set");
  }
}

void check_return_status(const char *sz, HAL_StatusTypeDef ret) {
  if (ret == HAL_OK) return; 
  Serial.print(sz);
  Serial.print("failed: ");
  Serial.println(ret);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
When I ran it on MBED it simply printed that: FLASH_OPTSR_BCM4 not set

Soon as I burn bootloader for zephyr blue led is back... 
Will play some more.

facchinm, 2025-05-22:
Sorry for the delay, the config to be added to Portenta .conf is `CONFIG_STM32H7_BOOT_M4_AT_INIT=n` to complete the trick

PR #128 (2025-05-23): PortentaH7 - Stop the blue blink...
This was caused by the default sketch running on the M4 core when no other sketch was loaded on the M4.

This changes the config to not automatically start the M4 at boot and.

It appears like that left the Blue LED on, so added an InitVariant implementation for the PortentaH7 that turns it off.

This resolves #124
closed 2025-05-23 by facchinm