arduino/ArduinoCore-zephyr #137: Support for Nicla Vision?

KurtE, 2025-06-09:
As @mjs513 noticed in the issue #136 I was curious if there were any Zephyr native samples for the different
camera captures and noticed that there was an example samples/drivers/video/capture, which I thought I might
try.  Under this I saw that there is a board information for it to run on an Arduino Nicla Vision board.

Which I don't have but found that it is currently on sale so I ordered one... Yep I know...

But I see there is currently no support here in the ArduinoCore-zephyr project for this board?

Are there plans to include support for it?

josuah, 2025-06-29:
The Arduino Nicla Vision works well on Zephyr. Remains to get it working on the ArduinoCore I suppose...

Various hardware is tested from time to time and posts are put on that thread:
- https://github.com/zephyrproject-rtos/zephyr/discussions/90428

KurtE, 2025-06-29:
Thanks @josuah,

With the Nicla Vision, I was able to build the capture sketch directly on Zephyr, using the current code (actually now maybe synced two days ago, I may sync again as looks like there is a beta RC... 

But when I try run it I don't see anything in the terminal window, nor anything showed up in the Kernel monitor window.  It does show stuff when I run the Hello world sample as well as the Blink sample..

Any suggestions on how best to try it out?

Thanks again

josuah, 2025-06-29:
This is a good point, currently by default, the [capture sample](https://docs.zephyrproject.org/latest/samples/drivers/video/capture/README.html) is a bit crude without a display attached.
You might have gotten a `Capture started` message and then not much.

- You may add `-DCONFIG_SHELL=y -DCONFIG_VIDEO_SHELL=y` at the end of the `west build` command to enable the shell, then you would have a `video capture` command you can use that will show frame statistics.

- You may enable `-DCONFIG_VIDEO_LOG_LEVEL_DBG=y` to get a line of log for every frame.

- You may try the [USB Video webcam](https://docs.zephyrproject.org/latest/samples/subsys/usb/uvc/README.html) sample (UVC protocol), with these notes if on Windows: https://github.com/zephyrproject-rtos/zephyr/discussions/90428#discussioncomment-13597721

Do feel free to report on this thread how it went!

closed 2025-11-12 by KurtE