arduino/ArduinoCore-zephyr #142: LVGL on GIGA seems to be running out of memory - lvgl sketch doesnt fit.

mjs513, 2025-06-14:
We modified our gigadisplay library to include the LVGL code from the Arduino_H7_library, i.e., code wrapped in `#if __has_include("lvgl.h")`.  It now compiles the grid example:
```
#if __MBED__
#include "Arduino_H7_Video.h"
#include "lvgl.h"
#include "Arduino_GigaDisplayTouch.h"

Arduino_H7_Video Display(800, 480, GigaDisplayShield);
Arduino_GigaDisplayTouch TouchDetector;
#elif __ZEPHYR__
#include "lvgl.h"
#include "Arduino_GigaDisplay.h"

Display Display(800, 480);
#endif

void setup() {
  delay(3000);
  Display.begin();

  //Display & Grid Setup
  lv_obj_t* screen = lv_obj_create(lv_scr_act());
  lv_obj_set_size(screen, Display.width(), Display.height());

  static lv_coord_t col_dsc[] = { 370, 370, LV_GRID_TEMPLATE_LAST };
  static lv_coord_t row_dsc[] = { 215, 215, 215, 215, LV_GRID_TEMPLATE_LAST };

  lv_obj_t* grid = lv_obj_create(lv_scr_act());
  lv_obj_set_grid_dsc_array(grid, col_dsc, row_dsc);
  lv_obj_set_size(grid, Display.width(), Display.height());

  //top left
  lv_obj_t* obj;
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 0, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 0, 1);      //row

  //bottom left
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 0, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 1, 1);      //row
  //top right
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 1, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 0, 1);      //row

  //bottom right
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 1, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 1, 1);      //row
}

void loop() {
  lv_timer_handler();
  delay(5);

}
```
However, it fails to load the sketch with the following error:
```
uart:~$ sketch
[00:00:07.672,000] <err> llext: Failed allocating 499264 bytes 4-aligned for region 0
[00:00:07.680,000] <err> llext: Failed to copy regions, ret -12
Failed to load sketch, rc -12
```
so at this point not sure what can be done.

KurtE, 2025-06-20:
@mjs513 and all.

Wondering about the memory management and usage of memory.
For example, today I am playing with the Portenta H7, with an ST77XX SPI display driver which is starting to work...

But is so limited in memory, that I can not load a single ILI9341 like font in the different examples without the sketch failing
to load with errors like:
```
uart:~$
*** Booting Zephyr OS build v3.7.0-16361-gcd0993ba3220 ***
[00:00:00.207,000] <err> llext: Failed allocating 131072 bytes 131072-aligned for region 2
[00:00:00.218,000] <err> llext: Failed to copy regions, ret -12
Failed to load sketch, rc -12
uart:~$
```
Not sure what to do about it, but would be nice if:
a) the summary of a build that may say something like:
```
Sketch uses 28120 bytes (3%) of program storage space. Maximum is 786432 bytes.
Global variables use 2384 bytes (0%) of dynamic memory, leaving 521240 bytes for local variables. Maximum is 523624 bytes.
Performing 1200-bps touch reset on serial port COM23
```
That the sizes and summary would match up, with what actually can be loaded.

b) Wondering if it is possible with Arduino/Zephyr/ST77XX - to do like we can do with a Teensy and tell certain things like
font tables to not copy it down from flash to memory, but instead access it directly from flash... 

Back to playying

mjs513, 2025-06-21:
@KurtE - @facchinm - @pillo79 

Been playing a bit more and trying to debug whats going on with lvgl.

First think is that I dont understand why with MBed I can do
```
#include "Arduino_H7_Video.h"
#include "lvgl.h"
```
and it uses lvgl_config in the library folder but with Zephyr I have 
```
#include "lvgl.h"
#include "Arduino_GigaDisplay.h"
```
and put lvgl_config files in the arduino libraries folder next to where lvgl is installed?  Have a horrible feeling lvgl is not using sdram as memory

KurtE, 2025-11-03:
@mjs513 - I thought I would try this one again, to see if anything changed:

It builds:
```
File does not have an ELF header, adding empty space
File C:\Users\kurte\AppData\Local\arduino\sketches\06E307E82A70AF86124EF5AF18F26B5F/sketch_nov3a.ino.bin saved as C:\Users\kurte\AppData\Local\arduino\sketches\06E307E82A70AF86124EF5AF18F26B5F/sketch_nov3a.ino.bin-zsk.bin
Using library lvgl at version 9.4.0 in folder: C:\Users\kurte\Documents\Arduino\libraries\lvgl 
Using library Arduino_GigaDisplay at version 1.0.2 in folder: C:\Users\kurte\Documents\Arduino\libraries\Arduino_GigaDisplay 
Using library Wire in folder: C:\Users\kurte\Documents\Arduino\hardware\arduino-git\ArduinoCore-zephyr\libraries\Wire (legacy)
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\zephyr\\tools\\arm-zephyr-eabi\\0.16.8/bin/arm-zephyr-eabi-size" -A "C:\\Users\\kurte\\AppData\\Local\\arduino\\sketches\\06E307E82A70AF86124EF5AF18F26B5F/sketch_nov3a.ino.elf"
Sketch uses 545304 bytes (69%) of program storage space. Maximum is 786432 bytes.
Global variables use 69312 bytes (13%) of dynamic memory, leaving 454312 bytes for local variables. Maximum is 523624 bytes.
Performing 1200-bps touch reset on serial port COM7
Waiting for upload port...
No upload port found, using COM7 as fallback
"C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5/dfu-util" --device ,0x2341:0x0366 -D "C:\Users\kurte\AppData\Local\arduino\sketches\06E307E82A70AF86124EF5AF18F26B5F/sketch_nov3a.ino.elf-zsk.bin" -a0 --dfuse-address=0x080E0000:leave
dfu-util 0.11-arduino4
```
But does not load!
```
uart:~$ Timeout while waiting for high throughput clock
SDIO firmware download error, whd_bus_sdio_init failed at 559
Interface is not up/NULL and failed in function whd_deinit at line 203
Interface is not up/NULL and failed in function whd_wifi_get_mac_address at line                                                                                                                           3184

[00:00:04.490,000] <inf> flashdisk: Initialize device wlan
[00:00:04.499,000] <inf> flashdisk: offset 1000, sector size 4096, page size 409                                                                                                                          6, volume size 1044480
[00:00:04.511,000] <inf> flashdisk: Initialize device ota
[00:00:04.519,000] <inf> flashdisk: offset 100000, sector size 4096, page size 4                                                                                                                          096, volume size 5242880
[00:00:04.532,000] <inf> littlefs: littlefs partition at /storage
[00:00:04.540,000] <inf> littlefs: LittleFS version 2.11, disk version 2.1
[00:00:04.550,000] <inf> littlefs: FS at qspi-nor-flash@0:0x700000 is 1792 0x100                                                                                                                          0-byte blocks with 512 cycle
[00:00:04.562,000] <inf> littlefs: partition sizes: rd 16 ; pr 4096 ; ca 4096 ;                                                                                                                           la 32
[00:00:04.573,000] <inf> littlefs: Automount /storage succeeded
*** Booting Zephyr OS build v4.2.0-27-g09722075b043 ***
[00:00:04.589,000] <err> llext: Failed allocating 522244 bytes 4-aligned for reg                                                                                                                          ion 0
[00:00:04.599,000] <err> llext: Failed to copy regions, ret -12
Failed to load sketch, rc -12
uart:~$
Timeout while waiting for high throughput clock
SDIO firmware download error, whd_bus_sdio_init failed at 559
Interface is not up/NULL and failed in function whd_deinit at line 203
Interface is not up/NULL and failed in function whd_wifi_get_mac_address at line 3184

[00:00:04.490,000] <inf> flashdisk: Initialize device wlan
[00:00:04.499,000] <inf> flashdisk: offset 1000, sector size 4096, page size 4096, volume size 1044480
[00:00:04.511,000] <inf> flashdisk: Initialize device ota
[00:00:04.519,000] <inf> flashdisk: offset 100000, sector size 4096, page size 4096, volume size 5242880
[00:00:04.532,000] <inf> littlefs: littlefs partition at /storage
[00:00:04.540,000] <inf> littlefs: LittleFS version 2.11, disk version 2.1
[00:00:04.550,000] <inf> littlefs: FS at qspi-nor-flash@0:0x700000 is 1792 0x1000-byte blocks with 512 cycle
[00:00:04.562,000] <inf> littlefs: partition sizes: rd 16 ; pr 4096 ; ca 4096 ; la 32
[00:00:04.573,000] <inf> littlefs: Automount /storage succeeded
*** Booting Zephyr OS build v4.2.0-27-g09722075b043 ***
[00:00:04.589,000] <err> llext: Failed allocating 522244 bytes 4-aligned for region 0
[00:00:04.599,000] <err> llext: Failed to copy regions, ret -12
Failed to load sketch, rc -12
uart:~$
```
Wish list: - the information at the end of a build actually meant something that is:
it outputs:
`Global variables use 69312 bytes (13%) of dynamic memory, leaving 454312 bytes for local variables.
`
Yet it does not load?  Only way to know it did not load, was if you happen to know to have a USB to serial adapter connected up to the right pins on the GIGA and have a Terminal program running.

pennam, 2026-09-22:
I've updated the sketch to use released Arduino_Video library, it builds and load fine using latest 1.0.0 core.

[image]

```
Low memory available, stability problems may occur.
Sketch uses 363416 bytes (30%) of program storage space. Maximum is 1179648 bytes.
Global variables use 202212 bytes (77%) of dynamic memory, leaving 59932 bytes for local variables. Maximum is 262144 bytes.
```

```
#include "Arduino_Video.h"

#if defined(ARDUINO_GIGA)
  #define TOUCH_ENABLED 1
#endif

#if TOUCH_ENABLED
#include "Arduino_GigaDisplayTouch.h"
#endif

#include "lvgl.h"

Arduino_Video Display;
#if TOUCH_ENABLED
Arduino_GigaDisplayTouch TouchDetector;
#endif

void error() {
    while (true) {
        digitalWrite(LED_BUILTIN, LOW);
        delay(500);
        digitalWrite(LED_BUILTIN, HIGH);
        delay(500);
    }
}

void setup() {
  Serial.begin(115200);

  if (Display.begin()) {
      error();
  }

#if TOUCH_ENABLED
  TouchDetector.begin();
#endif

  //Display & Grid Setup
  lv_obj_t* screen = lv_obj_create(lv_scr_act());
  lv_obj_set_size(screen, Display.width(), Display.height());

  static lv_coord_t col_dsc[] = { 370, 370, LV_GRID_TEMPLATE_LAST };
  static lv_coord_t row_dsc[] = { 215, 215, 215, 215, LV_GRID_TEMPLATE_LAST };

  lv_obj_t* grid = lv_obj_create(lv_scr_act());
  lv_obj_set_grid_dsc_array(grid, col_dsc, row_dsc);
  lv_obj_set_size(grid, Display.width(), Display.height());

  //top left
  lv_obj_t* obj;
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 0, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 0, 1);      //row

  //bottom left
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 0, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 1, 1);      //row
  //top right
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 1, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 0, 1);      //row

  //bottom right
  obj = lv_obj_create(grid);
  lv_obj_set_grid_cell(obj, LV_GRID_ALIGN_STRETCH, 1, 1,  //column
                       LV_GRID_ALIGN_STRETCH, 1, 1);      //row
}

void loop() { 
  /* Feed LVGL engine */
  lv_timer_handler();
}

```

closed 2026-09-22 by pennam