arduino/ArduinoCore-zephyr #148: GIGA R1:  Wifi still not working after incorporation of PR#130 changes

mjs513, 2025-06-23:
@facchinm - @KurtE

Rested Wifi on the giga after making the latest set of changes to PR#130.  Am still receiving in the arduino monitor:
```
starting Bluetooth® Low Energy module failed!
```
and debug window
```
[00:01:05.769,000] <inf> usb_cdc_acm: Device suspended
[00:01:05.994,000] <inf> usb_cdc_acm: Device configured
[00:01:09.593,000] <dbg> bt_driver: h4_open:
[00:01:09.597,000] <dbg> bt_driver: rx_thread: started
[00:01:09.603,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:01:09.608,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:01:09.614,000] <inf> bt_hci_raw: Lower HCI transport: H:4
[00:01:09.620,000] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:01:10.777,000] <dbg> bt_driver: h4_send: buf 0x24039f90 type 1 len 4
[00:01:10.785,000] <dbg> bt_driver: process_rx: remaining 0 discard 0 have_hdr 0 rx.buf 0 len 0
[00:01:10.794,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 0 rx.buf 0 len 0
[00:01:10.803,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 0 rx.buf 0 len 0
[00:01:10.812,000] <dbg> bt_driver: get_evt_hdr: Got event header. Payload 4 bytes
[00:01:10.820,000] <dbg> bt_driver: process_rx: remaining 4 discard 0 have_hdr 1 rx.buf 0 len 0
[00:01:10.829,000] <dbg> bt_driver: get_rx: type 0x04, evt 0x0e
[00:01:10.835,000] <dbg> bt_driver: read_payload: Allocated rx.buf 0x2403aaa8
[00:01:10.843,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 3
[00:01:10.850,000] <dbg> bt_driver: read_payload: Payload (len 4): 040e0401
[00:01:10.857,000] <dbg> bt_driver: process_rx: remaining 3 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 4
[00:01:10.867,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 2
[00:01:10.875,000] <dbg> bt_driver: read_payload: Payload (len 5): 040e040103
[00:01:10.882,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 5
[00:01:10.892,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 1
[00:01:10.899,000] <dbg> bt_driver: read_payload: Payload (len 6): 040e0401030c
[00:01:10.907,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 6
[00:01:10.917,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 0
[00:01:10.924,000] <dbg> bt_driver: read_payload: Payload (len 7): 040e0401030c00
[00:01:10.932,000] <dbg> bt_driver: read_payload: Putting buf 0x2403aaa8 to rx fifo
[00:01:10.940,000] <dbg> bt_driver: rx_thread: Calling bt_recv(0x2403aaa8)
[00:01:10.948,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:01:10.963,000] <dbg> bt_driver: h4_send: buf 0x24039f90 type 1 len 4

uart:~$ sketch
[00:00:07.710,000] <inf> usb_cdc_acm: Device suspended
[00:00:07.934,000] <inf> usb_cdc_acm: Device configured
[00:00:11.496,000] <dbg> bt_driver: h4_open:
[00:00:11.500,000] <dbg> bt_driver: rx_thread: started
[00:00:11.506,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:11.511,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:11.517,000] <inf> bt_hci_raw: Lower HCI transport: H:4
[00:00:11.523,000] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:12.680,000] <dbg> bt_driver: h4_send: buf 0x24039f90 type 1 len 4
[00:00:12.688,000] <dbg> bt_driver: process_rx: remaining 0 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:12.697,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:12.706,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:12.715,000] <dbg> bt_driver: get_evt_hdr: Got event header. Payload 4 bytes
[00:00:12.723,000] <dbg> bt_driver: process_rx: remaining 4 discard 0 have_hdr 1 rx.buf 0 len 0
[00:00:12.732,000] <dbg> bt_driver: get_rx: type 0x04, evt 0x0e
[00:00:12.738,000] <dbg> bt_driver: read_payload: Allocated rx.buf 0x2403aaa8
[00:00:12.746,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 3
[00:00:12.753,000] <dbg> bt_driver: read_payload: Payload (len 4): 040e0401
[00:00:12.760,000] <dbg> bt_driver: process_rx: remaining 3 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 4
[00:00:12.770,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 2
[00:00:12.778,000] <dbg> bt_driver: read_payload: Payload (len 5): 040e040103
[00:00:12.785,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 5
[00:00:12.795,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 1
[00:00:12.802,000] <dbg> bt_driver: read_payload: Payload (len 6): 040e0401030c
[00:00:12.810,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 1 rx.buf 0x2403aaa8 len 6
[00:00:12.820,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 0
[00:00:12.827,000] <dbg> bt_driver: read_payload: Payload (len 7): 040e0401030c00
[00:00:12.835,000] <dbg> bt_driver: read_payload: Putting buf 0x2403aaa8 to rx fifo
[00:00:12.843,000] <dbg> bt_driver: rx_thread: Calling bt_recv(0x2403aaa8)
[00:00:12.851,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:12.866,000] <dbg> bt_driver: h4_send: buf 0x24039f90 type 1 len 4
```

Is the issue related to SDIO support in zephyr: 
https://github.com/zephyrproject-rtos/zephyr/pull/89776

KurtE, 2025-06-23:
Quick question: I tried using the BLE scanner, using @facchinm fork/branch for ArduinoBLE.

I am using my own/fork/branch: https://github.com/arduino/ArduinoCore-zephyr/pull/134
Which I rebased to be up to data with the arduino branch...

The example sketch built, but does not load:
 ```
uart:~$ sketch
[00:26:27.957,000] <err> llext: Undefined symbol with no entry in symbol table bt_enable_raw, offset 23824, link section 16
[00:26:27.968,000] <err> elf: Could not find symbol bt_enable_raw!
[00:26:27.975,000] <err> llext: Failed to link, ret -61
Failed to load sketch, rc -61
uart:~$
```

KurtE, 2025-06-23:
Edit: Disregard above as rebuilding the ArduinoCore-zephyr... flashing....
Now program sort of loads:  and fails the way @mjs513 mentioned:
```
$$$ main.cpp called
InitVaria[00:00:00.959,000] <inf> usb_cdc_acm: Device suspended
uart:~$ d
calling start_static_threads
static threads start_static_threads
[00:00:01.216,000] <inf> usb_cdc_acm: Device configured
[00:00:29.388,000] <dbg> bt_driver: h4_open:
[00:00:29.395,000] <dbg> bt_driver: rx_thread: started
[00:00:29.403,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:29.411,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:29.419,000] <inf> bt_hci_raw: Lower HCI transport: H:4
[00:00:29.427,000] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:30.586,000] <dbg> bt_driver: h4_send: buf 0x24039fc8 type 1 len 4
[00:00:30.596,000] <dbg> bt_driver: process_rx: remaining 0 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:30.607,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:30.618,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 0 rx.buf 0 len 0
[00:00:30.629,000] <dbg> bt_driver: get_evt_hdr: Got event header. Payload 4 bytes
[00:00:30.640,000] <dbg> bt_driver: process_rx: remaining 4 discard 0 have_hdr 1 rx.buf 0 len 0
[00:00:30.651,000] <dbg> bt_driver: get_rx: type 0x04, evt 0x0e
[00:00:30.659,000] <dbg> bt_driver: read_payload: Allocated rx.buf 0x2403aae0
[00:00:30.669,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 3
[00:00:30.679,000] <dbg> bt_driver: read_payload: Payload (len 4): 040e0401
[00:00:30.688,000] <dbg> bt_driver: process_rx: remaining 3 discard 0 have_hdr 1 rx.buf 0x2403aae0 len 4
[00:00:30.690,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 2
[00:00:30.700,000] <dbg> bt_driver: read_payload: Payload (len 5): 040e040103
[00:00:30.709,000] <dbg> bt_driver: process_rx: remaining 2 discard 0 have_hdr 1 rx.buf 0x2403aae0 len 5
[00:00:30.721,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 1
[00:00:30.731,000] <dbg> bt_driver: read_payload: Payload (len 6): 040e0401030c
[00:00:30.741,000] <dbg> bt_driver: process_rx: remaining 1 discard 0 have_hdr 1 rx.buf 0x2403aae0 len 6
[00:00:30.753,000] <dbg> bt_driver: read_payload: got 1 bytes, remaining 0
[00:00:30.762,000] <dbg> bt_driver: read_payload: Payload (len 7): 040e0401030c00
[00:00:30.772,000] <dbg> bt_driver: read_payload: Putting buf 0x2403aae0 to rx fifo
[00:00:30.783,000] <dbg> bt_driver: rx_thread: Calling bt_recv(0x2403aae0)
[00:00:30.792,000] <dbg> bt_driver: rx_thread: rx.buf 0
[00:00:30.810,000] <dbg> bt_driver: h4_send: buf 0x24039fc8 type 1 len 4
uart:~$

```
Serial Monitor:
`starting Bluetooth® Low Energy module failed!
`

pennam, 2025-08-07:
@KurtE @mjs513 the BLE is now working using release 3.2 and https://github.com/arduino-libraries/ArduinoBLE/pull/407

iabdalkader, 2025-08-07:
This should fix BLE: https://github.com/arduino-libraries/ArduinoBLE/pull/407