arduino/ArduinoCore-zephyr #151: Portenta H7:   Using SPI with camera on Mid-carrier - Camera FAILS TO START

mjs513, 2025-06-26:
@KurtE - @facchinm - @pillo79 

Ran into a surpising while trying to display images on a ST7796 SPI display attached via SPI with a GC2145 camera on the mid-carrier board.  Same sketch works on the GIGA without an issue.

PROBLEM:  Display starts normally but camera fails to start with:
```
call cam.begin
Failed to set video format
Camera begin failed
```

which typically idicates as clock issue.  Right now the GC2145 is set to 10Mhz.

Running camerarawbytes works with the visualizer anda the ST7796 using test sketches and does start and runs through its starting screens.  

Not sure what the issue could be.

A simple sketch to test with
```
#include <ST77XX_zephyr.h>
#include "camera.h"

Camera cam;
#include <SPI.h>
#include <ST77XX_zephyr.h>
#include <ST77XX_zephyr_font_Arial.h>

#define TFT_CS    7 //10  // CS & DC can use pins 2, 6, 9, 10, 15, 20, 21, 22, 23
#define TFT_DC    6 //8  //  but certain pairs must NOT be used: 2+10, 6+9, 20+23, 21+22
#define TFT_RST   5 //9  // RST can use any pin

#define CAMERA_WIDTH 320 //320
#define CAMERA_HEIGHT 240 // 240
// Use one or the other
//ST7735_zephyr tft = ST7735_zephyr(&SPI, TFT_CS, TFT_DC, TFT_RST);

// For 1.54" TFT with ST7789
//ST7789_zephyr tft = ST7789_zephyr(&SPI, TFT_CS, TFT_DC, TFT_RST);

// For 3.5" or 4.0" TFT with ST7796
ST7796_zephyr tft = ST7796_zephyr(&SPI, TFT_CS, TFT_DC, TFT_RST);

void fatal_error(const char *msg) {
  Serial.println(msg);
  pinMode(LED_BUILTIN, OUTPUT);
  while (1) {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
  }
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {}

  Serial.println("\n*** start display camera image on ILI9341 ***");
  // Use this initializer if you're using a 1.8" TFT 128x160 displays
  //tft.initR(INITR_BLACKTAB);

  // Or use this initializer (uncomment) if you're using a 1.44" TFT (128x128)
  //tft.initR(INITR_144GREENTAB);

  // Or use this initializer (uncomment) if you're using a .96" TFT(160x80)
  //tft.initR(INITR_MINI160x80);

  // Or use this initializer (uncomment) for Some 1.44" displays use different memory offsets
  // Try it if yours is not working properly
  // May need to tweek the offsets
  //tft.setRowColStart(32,0);

  // Or use this initializer (uncomment) if you're using a 1.54" 240x240 TFT
  //tft.init(240, 240);   // initialize a ST7789 chip, 240x240 pixels

  // OR use this initializer (uncomment) if using a 2.0" 320x240 TFT:
  //tft.init(240, 320);           // Init ST7789 320x240
  tft.init(320, 480);

  tft.setRotation(1);
  tft.fillScreen(ST77XX_BLACK);
  delay(500);
  tft.fillScreen(ST77XX_RED);
  delay(500);
  tft.fillScreen(ST77XX_GREEN);
  delay(500);
  tft.fillScreen(ST77XX_BLUE);
  delay(500);
  tft.fillScreen(ST77XX_BLACK);
  delay(500);

  Serial.println("call cam.begin");
  if (!cam.begin(CAMERA_WIDTH, CAMERA_HEIGHT, CAMERA_RGB565, true)) {
    fatal_error("Camera begin failed");
  }
  Serial.println("Camera started");
  cam.setVerticalFlip(false);
  cam.setHorizontalMirror(false);

}

void loop() {}

```

This uses @KurtE's ST7796 display library.

Any ideas?

mjs513, 2025-06-28:
Additional info:
1.  Tried ov7670 on H7 failed as welle

and

2,   Ran same sketch with the GIGA using SPI  instead of SPI1 and no issues on the giga.

mjs513, 2025-06-28:
@KurtE - @facchinm - @pillo79 

With help for @KurtE found the issue.  Seems that when using the camera on the h7 mid- carrier the following pins are not can not be used for anything else:

D11 - SDA - shared with camera
D12 -  SCL - shared with camera
D6   - looks like camera clock pin

KurtE, 2025-07-03:
@mjs513 @facchinm @pillo79 @iabdalkader ...

Quick update: I have been playing around with the GC2145 on the Mid-Carrier and I also have it currently hooked up to
a ILI9341 display.  

Over the last couple of days, I created created a zephyr sample like sketch that reads in frames from the GC2145 and outputs to the display without using ArduinoCore-zephyr.   Did it for a couple of reasons.  Makes it easier for people up on Zephyr to try it out...
And I wanted to get to the newer code base, where there there are new video APIS and am curious to see how hard it would be to
add in support for some of them at the camera level...

I pulled in bits and pieces of the code here, plus my port of the ILI9341 display driver...  Which I now have a version that is
reasonably working, BUT... 

More details up at: https://github.com/zephyrproject-rtos/zephyr/issues/92354

The way I was able to get it to work reasonably well, like read in and displayed over 3000 images on the display, was 
to hack up the code that feeds the buffers up the buffer list, and hack up the first buffer to use memory NOT in SDRAM:
```
	printk("Initialze video buffer list (%u)\n",  ARRAY_SIZE(buffers));
	for (i = 0; i < ARRAY_SIZE(buffers); i++) {
		buffers[i] = video_buffer_aligned_alloc(bsize, CONFIG_VIDEO_BUFFER_POOL_ALIGN,
							K_FOREVER);
		#ifdef CAMERA_USE_FIXED_BUFFER
		if ((i == 0) &&  (buffers[i] != NULL)) {
			// REAL hack change the buffer over to our fixed buffer...
			buffers[i]->buffer = (uint8_t *)frame_buffer;
		}
		#endif

		printk("  %d:%x\n", i, (uint32_t)buffers[i]);
		if (buffers[i] == NULL) {
			printk("ERROR: Unable to alloc video buffer\n") ;
			return 0;
		}
		buffers[i]->type = type;
		video_enqueue(video_dev, buffers[i]);
	}

```

Wondering about how doing this on MBED was able to run the sketch that converted the image from 320x240 to 640x480 without running into this... (or at least not as often)

Things maybe to look at include:
a) Are the SDRams clocked at the same speed?
b) Should retry again on GIGA both zephyr and MBED
c) Different cameras?
d) Different data paths for memory and how SDRam is used?  That is Currently on Zephyr:
1) Camera reads into SDRAM, when frame complete Interrupt:
2) Interrupt, grabs another buffer in SDRAM, and MEMCPY to it.
3) When I grab the frame the bytes are reversed, so it reads from SDRAM, swaps the bytes and writes back to SDRAM
4) Output the bytes to the display from SDRAM.
...

Note: @josuah on Zephyr issue, mentioned that there are other boards/shields that may have issues as well with
SDRAM.

Kurt

Another Side Note: : been fun playing with Zephyr and this sketch as I can build and run it on my Windows machine without
having to jump through hoops.  (ArduinoCore-zephyr setup: Do builds in WSL, then copy the stuff back to the Window FS, to then run it in Arduino IDE...)