arduino/ArduinoCore-zephyr #153: UVC support: USB Video Class to turn Arduino's into webcams

josuah, 2025-07-13:
Through [Zephyr introduction of UVC](https://docs.zephyrproject.org/latest/samples/subsys/usb/uvc/README.html), it is now possible to turn a video driver into a webcam.

Arduino-side, I suspect users might want to do things slightly differently, maybe have Arduino sketch define the pixel format and resolution themselves without having to write an "adapter" driver just for this purpose.

This means the Zephyr UVC API has to be extended to support the typical Arduino use-cases.

What would be the ideal Arduino API for UVC? This way it is possible to plan the features Zephyr-side, and likely this will help non-Arduino users too in the process.

KurtE, 2025-07-13:
Sound interesting, Hopefully one of the main Arduino developers will be able to provide better insight on this.

Thinking off the top of my head wondering if this breaks down to several different sub-topics or capabilities.

Like Camera Library support, USB support (USB Video Device Class support), and the glue... Which as your post suggests
the glue might be contained in one or both of these...

On the Zephyr side, I see that you have an example using the Nicla Vision, so we know it is doable.  

But how best to do this on Arduino?  Hopefully others with more understanding of how the released Zephyr version of boards
will be packaged will maybe try to address things like:

How to change from one camera to another, or change something like, how many
buffers, or clock speed, etc.  Currently the Arduino user has to install the Arduino version of a full zephyr install,  
make the changes there, build the updated firmware, install with new firmware and related files into their Arduino install, 
flash the board with the new firmware and then finally build and upload their Arduino sketch.  
Which is a challenge on windows, as the Arduino build tools for generating the new firmware does not run natively on windows

Sorry I know some of that last part might be slightly off topic, but maybe parts of could ripple back into the zephyr classes.

josuah, 2025-07-14:
> Camera Library support

The UVC device is listed as yet another video device, so it should help with that.
It can be accessed like any camera device, with a name such as `uvc` or `uvc0` depending on how it is configured.

Then it can receive frames sent with the usual API `video_enqueue()`.

> How to change from one camera to another

One way would be to expose one UVC device per camera in case there are multiple on the same system? Or wait that the "selector unit" that permits to switch the video source at runtime gets implemented...

If you meant to change for a camera model for another (i.e. plug a different camera module): once a camera is defined on Zephyr-side, the camera driver is "scanned" by UVC and there is no need to adapt the UVC configuration for the camera.

> how many buffers

By exposing a video API, UVC will let the application manage the buffers, so this removes the duty out of UVC.

> or clock speed

The XCLK or PCLK image sensor clock speed is currently configured via devicetree, so different topic from UVC. I take note, however as interesting to be able to tune that without rebuilding.

> Which is a challenge on windows, as the Arduino build tools for generating the new firmware does not run natively on windows

Jumping to https://github.com/arduino/ArduinoCore-zephyr/discussions/127

closed 2025-10-25 by josuah