arduino/ArduinoCore-zephyr #202: __stm32_sdram2_section is no longer defined in Zephyr with their current sources.

KurtE, 2025-09-21:
**Describe the bug**
Zephyr removed these defines from secttion_tags.h.
I found this out when I rebased my Pull request to the current sources:
https://github.com/zephyrproject-rtos/zephyr/pull/93797

I received a reply to see:
https://github.com/zephyrproject-rtos/zephyr/pull/95299

which was pulled in 2 days ago.

So the loader/fixups.c code:
```
#if defined(CONFIG_SHARED_MULTI_HEAP)
#include <zephyr/kernel.h>
#include <zephyr/devicetree.h>
#include <zephyr/multi_heap/shared_multi_heap.h>

__stm32_sdram1_section static uint8_t __aligned(32) smh_pool[4 * 1024 * 1024];

int smh_init(void) {
	int ret = 0;
	ret = shared_multi_heap_pool_init();
	if (ret != 0) {
		return ret;
	}

	struct shared_multi_heap_region smh_sdram = {
		.addr = (uintptr_t)smh_pool,
		.size = sizeof(smh_pool),
		.attr = SMH_REG_ATTR_EXTERNAL,
	};

	ret = shared_multi_heap_add(&smh_sdram, NULL);
	if (ret != 0) {
		return ret;
	}
	return 0;
}

SYS_INIT(smh_init, POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEFAULT);
#endif
```
no longer builds.

**Target board + cli verbose compilation output**
This will impact both: Arduino GIGA as well as Arduino Portenta H7

**Full verbose** compilation output, ideally with `arduino-cli` invocation or from IDE 2.3.3+
My zephyr test sketches builds fail with:
```
D:/zephyrproject/zephyr_test_sketches/camera_to_tft/src/main.cpp:790:1: error: '__stm32_sdram1_section' does not name a type
  790 | __stm32_sdram1_section static uint8_t __aligned(32) smh_pool[4*1024*1024];
      | ^~~~~~~~~~~~~~~~~~~~~~
D:/zephyrproject/zephyr_test_sketches/camera_to_tft/src/main.cpp: In function 'int smh_init()':
D:/zephyrproject/zephyr_test_sketches/camera_to_tft/src/main.cpp:801:29: error: 'smh_pool' was not declared in this scope
  801 |         .addr = (uintptr_t) smh_pool,
      |                             ^~~~~~~~
[68/273] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/shell/modules/kernel_service/thread/resume.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\zephyrproject\zephyr_test_sketches\camera_to_tft\build'
```

KurtE, 2025-09-22:
Update: it actually works, if I put in the right size 
Z_GENERIC_SECTION(SDRAM1) static uint8_t __aligned(32) smh_pool[410241024];
was probably typed in as:
`Z_GENERIC_SECTION(SDRAM1) static uint8_t __aligned(32) smh_pool[4*1024*1024];`
But posting took the * * as bold or the like and as such requested the pool to be too large.

pennam, 2025-11-20:
@KurtE can we close this? If I understood correctly this is not an issue anymore.

KurtE, 2025-11-20:
> [@KurtE](https://github.com/KurtE) can we close this? If I understood correctly this is not an issue anymore.

I don't know... That is, this applies to the zephyr changes AFTER the Arduino fork of it.  I have not checked if any of the changes needed to make it work were made locally or not. 

That is did Arduino change over to use the Z_GENERIC_SECTION for this?

KurtE, 2025-11-20:
@pennam - Looks like Arduino code has not been updated:
https://github.com/arduino/ArduinoCore-zephyr/blob/main/loader/fixups.c#L114
`__stm32_sdram1_section static uint8_t __aligned(32) smh_pool[4 * 1024 * 1024];
`
The fix at the time was:
`Z_GENERIC_SECTION(SDRAM1) static uint8_t __aligned(32) smh_pool[4*1024*1024];`

Note: I have not verified this with the Zephyr current sources to see if the old way works again or not...

But hopefully when someone starts the moving ArduinoCore-zephry to the new zephyr release 4.3 we will know.
Looks like @pillo79 may have started?
https://github.com/pillo79/zephyr/tree/zephyr-arduino-v4.3.0

pennam, 2025-11-21:
> Looks like Arduino code has not been updated:

we are currently using zephyr 4.2.0 so there is no need to update it. We will update the code when we switch to zephyr 4.3.0.

pennam, 2026-09-01:
I will close this since loader has been updated and builds using zephyr 4.4.x

closed 2026-09-01 by pennam